openwrt/openwrt #18523: Regular sysupgrade from 24.10.0 to 24.10.1 fails on Zyxel GS1900

pboguslawski, 2025-04-18:
### Describe the bug

Upgrading Zyxel GS1900-24 v1 and Zyxel GS1900-24HP v2 Switch from 24.10.0 to 24.10.1 using Attended Sysupgrade fails with error

```
Fri Apr 18 08:05:42 2025 user.info upgrade: The device is supported, but this image is incompatible for sysupgrade based on the image version (1.0->2.0).
Fri Apr 18 08:05:42 2025 user.info upgrade: Dual firmware paritition merged due to size constraints. Upgrade requires a new factory install. Regular sysupgrade is not possible.
```

Changed partition layout was already applied on previous upgrade 23.05  -> 24.10.0 and according to [release notes](https://openwrt.org/releases/24.10/notes-24.10.1) partition layout change is required only when upgrading from 23.05-.

### OpenWrt version

OpenWrt 24.10.0 r28427-6df0e3d02a / LuCI openwrt-24.10 branch 25.037.68331~42f464c

### OpenWrt release

OpenWrt 24.10.0 r28427-6df0e3d02a / LuCI openwrt-24.10 branch 25.037.68331~42f464c

### OpenWrt target/subtarget

OpenWrt 24.10.0 r28427-6df0e3d02a / LuCI openwrt-24.10 branch 25.037.68331~42f464c

### Device

realtek/rtl838x

### Image kind

Official downloaded image

### Diffconfig

```text

```

### Terms

- [x] I am reporting an issue for OpenWrt, not an unsupported fork.

pboguslawski, 2025-04-18:
> Is this from a clean repository?

It's Attended Sysupgrade.

> Is this from a supported device?

Both devices are supported according to error message above.

robimarko, 2025-04-18:
So, somehow compat version is not being updated via:
https://github.com/openwrt/openwrt/blob/main/target/linux/realtek/base-files/etc/board.d/05_compat-version

aimacintyre, 2025-04-19:
> So, somehow compat version is not being updated

There have been several forum threads where this has been a result of users restoring a configuration backup after installing a version with a compat_version change and then having the compat_version check fail at the next upgrade (e.g. config restored after upgrade from 23.05.x to 24.10.0 then having the check fail while upgrading 24.10.1).

The config restoration code would appear to require logic to check compat_version during restoration to catch this case and preserve the updated compat_version even if a user forced the config restoration.

robimarko, 2025-04-19:
That would explain the issue, in this case you must then manually update the compat version in the config.

Config restoration is pretty simple and does not check anything but import the config, but it would be nice to see a warning at least about this

paulastronaut, 2025-04-19:
I fell into this trap as well. Adding the line `option compat_version '2.0'` to /etc/config/system let my apply the sysupdate to 24.10.1. Pay attention, I'm pretty sure that you can brick a device if you add this line in cases where it does not apply!

I would also have wished for some information about this somewhere. A remark in the release notes would have helped, for example.

pboguslawski, 2025-04-20:
> There have been several forum threads where this has been a result of users restoring a configuration backup after installing a version with a compat_version change and then having the [...]

Yes, config was restored from backup made in 23.05 after upgrade to 24.10.0 and `/etc/config/system` contained
```
option compat_version '1.0'
```
Manual replacing
```
root@mygs1900:~# sed -i "s/option compat_version '1.0'/option compat_version '2.0'/" /etc/config/system
```
fixed this issue. Thank you for hint.

Why not to remove `compat_version` from configuration and hardcode it into firmware (to avoid such surprises after restoring backup)?

Neustradamus, 2025-04-22:
@svanheule: Have you seen this ticket?

svanheule, 2025-04-22:
I've added a bit of info to the release notes noting this caveat with restoring configs. I'm afraid that's the best I can do for now, as the uci-defaults script that sets the compat_version is only run on the first boot after an upgrade.

For what it's worth, this is consistent with other compat_version changes, which normally imply "upgrade without keeping config" (minor version) or "fresh install without keeping config" (major version). Usually _most_ of the config can actually be restored, as is the case here, but the user must be aware of what they are doing. I hope the amended release notes help with that.

closed 2025-04-22 by svanheule