openwrt/openwrt #18527: apk doesn't create a user defined using USERID on upgrade

GeorgeSapkin, 2025-04-18:
### Describe the bug

When using apk, a user and a group defined using `USERID` in a package Makefile doesn't get created when it's defined in a _new_ version of a package and that version of a package is installed over an _existing_ version (i.e. upgraded) without `USERID` defined. It work when a new package with `USERID` defined is installed. I have successfully reproduced this on a clean VM with a snapshot (r29252-5d26507a73) and apk.

Both installing a new package and upgrading on a clean VM with 24.10 also works as expected. This seems like a regression compared to opkg/ipk, unless something is messed up with the Makefile or this is working as expected with apk.

Package PR with this issue: https://github.com/openwrt/packages/pull/26225

### OpenWrt version

r29252-5d26507a73

### OpenWrt release

SNAPSHOT

### OpenWrt target/subtarget

x86/64

### Device

QEMU Standard PC (Q35 + ICH9, 2009)

### Image kind

Official downloaded image

### Steps to reproduce

- Install a package _without_ a `USERID` defined
- Check `/etc/passwd` to ensure the target user doesn't exist
- Define `USERID` and rebuild a package
- Upgrade the package
- Check `/etc/passwd` to ensure the target user exists, but it doesn't

### Actual behaviour

User and group defined using `USERID` is _not_ created when it's defined in a new version of a package on upgrade. But is created when a new package is installed.

### Expected behaviour

User and group defined using `USERID` is created both when a new package is installed and on upgrade, same as with opkg/ipk.

### Additional info

Example define that works with opkg/ipk on upgrade in 24.10, but doesn't work with apk:

```Makefile
define Package/adguardhome
	SECTION:=net
	CATEGORY:=Network
	TITLE:=Network-wide ads and trackers blocking DNS server
	URL:=https://github.com/AdguardTeam/AdGuardHome
	DEPENDS:=$(GO_ARCH_DEPENDS) +ca-bundle
	USERID:=adguardhome=853:adguardhome=853
endef
```

### Diffconfig

```text

```

### Terms

- [x] I am reporting an issue for OpenWrt, not an unsupported fork.

efahl, 2025-04-18:
@Ansuel is this something you've seen before?

GeorgeSapkin, 2025-04-18:
I looked into it a bit: looks like there is only a `post-install` script defined `package-pack.mk`, which doesn't seem to run on upgrade. I tested adding the same script to `post-upgrade` [1] and it seems to resolve the issue. I'll make a PR with this fix.

[1]: https://wiki.alpinelinux.org/wiki/Creating_an_Alpine_package

commit 03880e25 (2025-06-13): include/package-pack: add missing apk package lifecycle events

- pre-install calls preinst
- pre-upgrade calls preinst with PKG_UPGRADE=1
- post-upgrade calls postinst with PKG_UPGRADE=1
- prerm and postrm from the previous version of a package are NOT
  executed on upgrade, so packages are expected to handle their own
  migrations in the new versions

Link: https://wiki.alpinelinux.org/wiki/Creating_an_Alpine_package
Fixes: #18527
Reported-by: Dobroslaw Kijowski <dobo90@gmail.com>
Suggested-by: Eric Fahlgren <ericfahlgren@gmail.com>
Suggested-by: Jonas Gorski <jonas.gorski@gmail.com>
Suggest [... cut]
PR #18531 (2025-06-13): include/package-pack: add missing apk package lifecycle events
Maintainer: @Ansuel, @aparcar, @efahl, @robimarko
Compile tested: snapshot SDK with this PR applied
Run tested: snapshot r29252-5d26507a73, x86/64 on QEMU Standard PC (Q35 + ICH9, 2009)  

When using apk, a user and a group defined using `USERID` in a package Makefile doesn't get created when it's defined in a new version of a package and that version of a package is installed over an existing version (i.e. upgraded) without `USERID` defined. It only works when a new package with `USERID` defined is installed due to `post-install`  [... cut]
closed 2025-06-13 by openwrt-bot