openwrt/openwrt #18528: Disable/Enable PoE Command is not working with 24.10.1 and 24.10.0 on the Zyxel GS1900-24HP v1

Wupsman, 2025-04-18:
### Describe the bug

The Disable/Enable PoE Command `ubus call poe manage '{ "port" : "lan1", "enable" : false }'` no longer works with the update to 24.10.1/24.10.0 and as far as I could see, nothing has changed in realtek-poe.

### OpenWrt version

r28597-0425664679

### OpenWrt release

24.10.1

### OpenWrt target/subtarget

realtek/rtl838x

### Device

Zyxel GS1900-24HP v1

### Image kind

Official downloaded image

### Steps to reproduce

Update the switch to 24.10.1 and try to deactivate PoE with `ubus call poe manage ‘{ “port” : “lan1”, “enable” : false }’`. 

### Actual behaviour

It is the same behaviour as when the command is accepted, except that PoE remains Enable.

### Expected behaviour

For the corresponding port, the PoE status changes from Delivering power/Searching to Disabled.

### Additional info

With version 23.05.5 it works fine on the Zyxel GS1900-24HP v1.

`root@OpenWrtSwitchTechnikraum:~# ubus call poe info
{
        "firmware": "v17.1",
        "mcu": "ST Micro ST32F100 Microcontroller",
        "budget": 170.000000,
        "consumption": 10.700000,
        "ports": {
                "lan1": {
                        "priority": 2,
                        "mode": "PoE+",
                        "status": "Searching"
...
`

Version 24.10.1 works on a GS100-10HP without any problems:

`root@OpenWrt:~# ubus call poe info
{
        "firmware": "v22.4",
        "mcu": "ST Micro ST32F100 Microcontroller",
        "budget": 77.000000,
        "consumption": 0.000000,
        "ports": {
                "lan1": {
                        "priority": 2,
                        "mode": "PoE+",
                        "status": "Disabled"
...
`

### Diffconfig

```text

```

### Terms

- [x] I am reporting an issue for OpenWrt, not an unsupported fork.

Neustradamus, 2025-04-22:
@svanheule: Have you seen this ticket?

Wupsman, 2025-04-25:
Looks like it's my fault.

I tried again on my GS1900-24HP v1 with 24.10.1 and did not keep the configuration, with the result that it now works!

closed 2025-04-25 by Wupsman