openwrt/openwrt #18541: bananapi,bpi-r4 24.10.0 -> 24.10.1 update file due to compat_version

adrianschroeter, 2025-04-20:
### Describe the bug

Using official sysupgrade on a Banana BPi-r4 I get

`the device is supported, but the config is incompatible to the new image (1.1->1.0)
`

I suppose an adaption similar to r3 is missing here:

`target/linux/mediatek/filogic/base-files/etc/uci-defaults/05_fix-compat-version
`

I have not tried to adapt/upgrade/sending a pull request yet, feel free to ask for adaption and retry.
thanks! 

### OpenWrt version

r28427-6df0e3d02a

### OpenWrt release

24.10.0

### OpenWrt target/subtarget

mediatek/filogic

### Device

Bananapi BPI-R4

### Image kind

Official downloaded image

### Steps to reproduce

```
# sysupgrade -i https://downloads.openwrt.org/releases/24.10.1/targets/mediatek/filogic/openwrt-24.10.1-mediatek-filogic-bananapi_bpi-r4-squashfs-sysupgrade.itb
Downloading 'https://downloads.openwrt.org/releases/24.10.1/targets/mediatek/filogic/openwrt-24.10.1-mediatek-filogic-bananapi_bpi-r4-squashfs-sysupgrade.itb'
Connecting to 2a04:4e42:6f::644:443
Writing to '/tmp/sysupgrade.img'
/tmp/sysupgrade.img  100% |*******************************| 16640k  0:00:00 ETA
Download completed (17039668 bytes)
Sun Apr 20 09:21:16 UTC 2025 upgrade: The device is supported, but the config is incompatible to the new image (1.1->1.0). Please upgrade without keeping config (sysupgrade -n).
Image check failed.

```

### Expected behaviour

Being able to update a patchlevel increment without loosing config.

### Diffconfig

```text

```

### Terms

- [x] I am reporting an issue for OpenWrt, not an unsupported fork.

aimacintyre, 2025-04-20:
Likely to be similar to #18523

adrianschroeter, 2025-04-22:
yes, I imported an older configuration as well. (it complained about the config, not the partitioning though).

Setting the compat_version in /etc/config/system back to 1.0 leads to a smooth update.

aimacintyre, 2025-04-23:
See [resolution of #18523](https://github.com/openwrt/openwrt/issues/18523#issuecomment-2822403438) - suggest closing on that basis.

adrianschroeter, 2025-04-24:
If you say an import off back'd up config always can lead to this, sure.

IMHO it would make sense to validate the version at import time instead though. 
Because you are already running at that point of time for quite some time using this config. So, it is either a problem at that time already or the upgrade, which happens way later, should just work from my POV.

So the warning/error should popup way earlier IMHO. Later on you need to remember that you actually imported the config ...

But feel free to close the issue, if you disagree. Thanks a lot in any case.