openwrt/openwrt #18544: 5ghz wifi does not start if country code set to india

olumolu, 2025-04-20:
### Describe the bug

I find this issue for some release now and it only happen to india specific and happens on my all devices if i set 5ghz wifi as india country code the wifi does not  work as wifi does not start.
Where as if i set this as uk us or jp or another country wifi works
Devices
Xiaomi gigabit 4a 1st 2 device
Tp link ax23 

### OpenWrt version

24.10.1

### OpenWrt release

OpenWrt 24.10.1 r28597-0425664679 

### OpenWrt target/subtarget

ramips/mt7621

### Device

Xiaomi Mi Router 4A Gigabit Edition

### Image kind

Official downloaded image

### Steps to reproduce

Go to
Network -> wireless -> edit openwrt 5ghz wifi
Set country code as india -IN
AND after selecting save changes and wait but 5ghz will never start it will stay off forever until unless you select country code as some other country or driver default.

### Expected behaviour

It should work and switch between the allowed wireless range from the country i am using other country so it works.

### Additional info

If needed i can provide other debug info thanks.

### Diffconfig

```text

```

### Terms

- [x] I am reporting an issue for OpenWrt, not an unsupported fork.

brada4, 2025-04-21:
Power device on with 5ghz disabled but country set, then check that `iw list` shows adequate channel map, and enable 5ghz. Changing DFS region (etsi/fcc/jp) on the fly usually disables 5ghz dfs until poweroff (reboot is not sufficient).

olumolu, 2025-04-21:
> Power device on with 5ghz disabled but country set, then check that `iw list` shows adequate channel map, and enable 5ghz. Changing DFS region (etsi/fcc/jp) on the fly usually disables 5ghz dfs until poweroff (reboot is not sufficient).

after enabling 5ghz wifi does not turn on as channel does not appear if selection set to auto.

brada4, 2025-04-21:
Wait 1-10min to settle with radars.
https://git.kernel.org/pub/scm/linux/kernel/git/wens/wireless-regdb.git/tree/db.txt#n931

olumolu, 2025-04-21:
No it did not work

[image]

olumolu, 2025-04-21:
And what is your point wifi need 10min to turn on look if it does not turn on immediately it won't there is a bug

brada4, 2025-04-21:
> what is your point

Indian government radio regulations, you can not boss yourself out of them.

Please check `logread -f -e hostap` to see ACS completes.

olumolu, 2025-04-22:
```
Tue Apr 22 12:51:30 2025 kern.info kernel: [    1.320134] pci 0000:01:00.0: BAR 6: assigned [mem 0x60100000-0x6010ffff pref]
Tue Apr 22 12:51:30 2025 kern.info kernel: [    1.327288] pci 0000:00:00.0: PCI bridge to [bus 01]
Tue Apr 22 12:51:30 2025 kern.info kernel: [    1.332181] pci 0000:00:00.0:   bridge window [io  0x0000-0x0fff]
Tue Apr 22 12:51:30 2025 kern.info kernel: [    1.338263] pci 0000:00:00.0:   bridge window [mem 0x60000000-0x600fffff]
Tue Apr 22 12:51:30 2025 kern.info kernel: [    1.344968] pci 0000:00:00.0:   bridge window [mem 0x60100000-0x601fffff pref]
Tue Apr 22 12:51:30 2025 kern.info kernel: [    1.352175] pci 0000:02:00.0: BAR 0: assigned [mem 0x60200000-0x602fffff]
Tue Apr 22 12:51:30 2025 kern.info kernel: [    1.358916] pci 0000:00:01.0: PCI bridge to [bus 02]
Tue Apr 22 12:51:30 2025 kern.info kernel: [    1.363803] pci 0000:00:01.0:   bridge window [io  0x1000-0x1fff]
Tue Apr 22 12:51:30 2025 kern.info kernel: [    1.369881] pci 0000:00:01.0:   bridge window [mem 0x60200000-0x602fffff]
Tue Apr 22 12:51:30 2025 kern.info kernel: [    1.376639] pci 0000:00:01.0:   bridge window [mem 0x60300000-0x603fffff pref]
Tue Apr 22 12:51:30 2025 kern.info kernel: [    1.385411] Serial: 8250/16550 driver, 16 ports, IRQ sharing enabled
Tue Apr 22 12:51:30 2025 kern.info kernel: [    1.401463] printk: console [ttyS0] disabled
Tue Apr 22 12:51:30 2025 kern.info kernel: [    1.406553] 1e000c00.uartlite: ttyS0 at MMIO 0x1e000c00 (irq = 18, base_baud = 3125000) is a 16550A
Tue Apr 22 12:51:30 2025 kern.info kernel: [    1.415594] printk: console [ttyS0] enabled
Tue Apr 22 12:51:30 2025 kern.info kernel: [    1.423911] printk: bootconsole [early0] disabled
Tue Apr 22 12:51:30 2025 kern.info kernel: [    1.438614] spi-mt7621 1e000b00.spi: sys_freq: 220000000
Tue Apr 22 12:51:30 2025 kern.info kernel: [    1.445440] spi-nor spi0.0: gd25q128 (16384 Kbytes)
Tue Apr 22 12:51:30 2025 kern.notice kernel: [    1.450513] 8 fixed-partitions partitions found on MTD device spi0.0
Tue Apr 22 12:51:30 2025 kern.err kernel: [    1.456975] OF: Bad cell count for /palmbus@1e000000/spi@b00/flash@0/partitions
Tue Apr 22 12:51:30 2025 kern.err kernel: [    1.464287] OF: Bad cell count for /palmbus@1e000000/spi@b00/flash@0/partitions
Tue Apr 22 12:51:30 2025 kern.notice kernel: [    1.471894] Creating 8 MTD partitions on "spi0.0":
Tue Apr 22 12:51:30 2025 kern.notice kernel: [    1.476761] 0x000000000000-0x000000030000 : "u-boot"
Tue Apr 22 12:51:30 2025 kern.notice kernel: [    1.483556] 0x000000030000-0x000000040000 : "u-boot-env"
Tue Apr 22 12:51:30 2025 kern.notice kernel: [    1.490300] 0x000000040000-0x000000050000 : "Bdata"
Tue Apr 22 12:51:30 2025 kern.notice kernel: [    1.496440] 0x000000050000-0x000000060000 : "factory"
Tue Apr 22 12:51:30 2025 kern.err kernel: [    1.502867] OF: Bad cell count for /palmbus@1e000000/spi@b00/flash@0/partitions
Tue Apr 22 12:51:30 2025 kern.notice kernel: [    1.510672] 0x000000060000-0x000000070000 : "crash"
Tue Apr 22 12:51:30 2025 kern.notice kernel: [    1.517246] 0x000000070000-0x000000080000 : "cfg_bak"
Tue Apr 22 12:51:30 2025 kern.notice kernel: [    1.523532] 0x000000080000-0x000000180000 : "overlay"
Tue Apr 22 12:51:30 2025 kern.notice kernel: [    1.529897] 0x000000180000-0x000001000000 : "firmware"
Tue Apr 22 12:51:30 2025 kern.notice kernel: [    1.536553] 2 uimage-fw partitions found on MTD device firmware
Tue Apr 22 12:51:30 2025 kern.notice kernel: [    1.542489] Creating 2 MTD partitions on "firmware":
Tue Apr 22 12:51:30 2025 kern.notice kernel: [    1.547473] 0x000000000000-0x000000309266 : "kernel"
Tue Apr 22 12:51:30 2025 kern.warn kernel: [    1.552421] mtd: partition "kernel" doesn't end on an erase/write block -- force read-only
Tue Apr 22 12:51:30 2025 kern.notice kernel: [    1.561915] 0x000000309266-0x000000e80000 : "rootfs"
Tue Apr 22 12:51:30 2025 kern.warn kernel: [    1.566961] mtd: partition "rootfs" doesn't start on an erase/write block boundary -- force read-only
Tue Apr 22 12:51:30 2025 kern.info kernel: [    1.577385] mtd: setting mtd9 (rootfs) as root device
Tue Apr 22 12:51:30 2025 kern.notice kernel: [    1.582520] 1 squashfs-split partitions found on MTD device rootfs
Tue Apr 22 12:51:30 2025 kern.notice kernel: [    1.588731] 0x0000006e0000-0x000000e80000 : "rootfs_data"
Tue Apr 22 12:51:30 2025 kern.info kernel: [    1.739533] mt7530-mdio mdio-bus:1f: MT7530 adapts as multi-chip module
Tue Apr 22 12:51:30 2025 kern.info kernel: [    1.754348] mtk_soc_eth 1e100000.ethernet eth0: mediatek frame engine at 0xbe100000, irq 19
Tue Apr 22 12:51:30 2025 kern.info kernel: [    1.765846] mtk_soc_eth 1e100000.ethernet eth1: mediatek frame engine at 0xbe100000, irq 19
Tue Apr 22 12:51:30 2025 kern.info kernel: [    1.775963] i2c_dev: i2c /dev entries driver
Tue Apr 22 12:51:30 2025 kern.info kernel: [    1.784462] NET: Registered PF_INET6 protocol family
Tue Apr 22 12:51:30 2025 kern.info kernel: [    1.792693] Segment Routing with IPv6
Tue Apr 22 12:51:30 2025 kern.info kernel: [    1.796532] In-situ OAM (IOAM) with IPv6
Tue Apr 22 12:51:30 2025 kern.info kernel: [    1.800595] NET: Registered PF_PACKET protocol family
Tue Apr 22 12:51:30 2025 kern.info kernel: [    1.805725] bridge: filtering via arp/ip/ip6tables is no longer available by default. Update your scripts to load br_netfilter if you need this.
Tue Apr 22 12:51:30 2025 kern.info kernel: [    1.819192] 8021q: 802.1Q VLAN Support v1.8
[... cut ...]
Tue Apr 22 12:55:09 2025 daemon.info dnsmasq-dhcp[1]: DHCPACK(br-lan) 192.168.1.162 de:b1:81:cf:06:11
Tue Apr 22 12:55:11 2025 daemon.warn odhcpd[1639]: No default route present, overriding ra_lifetime to 0!
Tue Apr 22 12:55:12 2025 daemon.info dnsmasq[1]: read /etc/hosts - 12 names
Tue Apr 22 12:55:12 2025 daemon.info dnsmasq[1]: read /tmp/hosts/dhcp.cfg01411c - 4 names
Tue Apr 22 12:55:12 2025 daemon.info dnsmasq[1]: read /tmp/hosts/odhcpd - 0 names
Tue Apr 22 12:55:12 2025 daemon.info dnsmasq-dhcp[1]: read /etc/ethers - 0 addresses
```

brada4, 2025-04-22:
Looks dfs scan failed( ie radar on channel), normal modus operandi would be to ACS other channel, ultimately with all else failing oneof 2 non-dfs options.
Try to restart 5ghz wifi and wait few minutes staring at logread i mentioned before. Network card has ch100 blocked for next hour, nothing works better than 5 5ghz wifi restart ritual against a bug.

olumolu, 2025-04-22:
> Looks dfs scan failed( ie radar on channel), normal modus operandi would be to ACS other channel, ultimately with all else failing oneof 2 non-dfs options. Try to restart 5ghz wifi and wait few [...]

it used to work in past when 22.03.0X version was there but after that it no longer working i think there is bug in the firmware.

olumolu, 2025-04-22:
> How the five restarts worked out?
> 
> Workaround(disable DFS channels) https://w1.fi/cgit/hostap/tree/hostapd/hostapd.conf#n216 Ie add to 5ghz radio section
> 
> ```
> list hostapd_options 'acs_exclude_dfs=1'
> ```

this workaround was also suggested by deepseek but i have switch to another country uk and it works with it. when it fixes i will switch to it.

brada4, 2025-04-22:
etsi dfs does not conform with indian law.

brada4, 2025-04-22:
Would be nice to close llm-inspired non-issue. Youre so busy with it you keep poisoning dfs circuitry and have not completed 5 restart enchantment. ie you gotta prove issue is not YOU jumping continents for no reason.

olumolu, 2025-04-22:
> Would be nice to close llm-inspired non-issue

No this is not a llm inspired. Dont make a statement like this. And 

> 5 restart enchantment

yes i will check.

brada4, 2025-04-22:
You spent 2 days making no change at all untl you refered to AI hallucination as your source.

olumolu, 2025-04-24:
> How the five restarts worked out?

does not work but switching channels manually works on all channels.

brada4, 2025-04-24:
Please do all the testing powering device on with disabled radio in driver default or india country code. Does not sound serious that every time you brag about changing country code on powered device. That alone is nearly impossible in real world besides unsuspecringly breakinh the law and the driver and firmware.

olumolu, 2025-04-24:
i would like to ask you if you have a running openwrt router running can you test this if this works on your router
country india
channel mode auto
wifi 5ghz

brada4, 2025-04-24:
Cannot test, eu mandates etsi.