openwrt/openwrt #18551: uhttpd fails to listen on specific IP address when network is slow to configure

adelton, 2025-04-21:
### Describe the bug

To be able to have additional services listen under their own hostnames and thus under their own IP addresses on ports 80 and 443, I need to move uhttpd from its default
```
config uhttpd 'main'
	list listen_http '0.0.0.0:80'
	list listen_https '0.0.0.0:443'
```
to a more specific
```
config uhttpd 'main'
	list listen_http '192.168.1.1:80'
	list listen_https '192.168.1.1:443'
```
(let's focus on the IPv4 addresses only for a start).

However, when the network is slow to get configured, I get
```
daemon.err uhttpd[2062]: bind(): Address not available
```
and uhttpd fails to listen, making the admin interface potentially not available.

### OpenWrt version

r28597-0425664679

### OpenWrt release

24.10.1

### OpenWrt target/subtarget

x86/64

### Device

QEMU Standard PC (Q35 + ICH9, 2009)

### Image kind

Official downloaded image

### Steps to reproduce

We will simulate slow network by creating a script which will sleep for a while before running the actual `/sbin/netifd`:
```
root@OpenWrt:~# cat > /sbin/netifd-x <<'EOF'
#!/bin/sh
logger -s -t "network" -p daemon.crit "Starting network delay ..."
sleep 5
logger -s -t "network" -p daemon.crit "Starting network now ..."
exec /sbin/netifd "$@"
EOF
root@OpenWrt:~# chmod a+x /sbin/netifd-x
```
and configure the `network` service to use it:
```
root@OpenWrt:~# sed -i 's/netifd/netifd-x/' /etc/init.d/network
```

We reboot the device and afterwards we log in, check with `logread | grep Starting` that the delayed network setup is enabled, and with `netstat` that uhttpd listens everywhere just fine:
```
root@OpenWrt:~# netstat -tlnp | grep uhttpd
tcp        0      0 0.0.0.0:80              0.0.0.0:*               LISTEN      2061/uhttpd
tcp        0      0 0.0.0.0:443             0.0.0.0:*               LISTEN      2061/uhttpd
tcp        0      0 :::80                   :::*                    LISTEN      2061/uhttpd
tcp        0      0 :::443                  :::*                    LISTEN      2061/uhttpd
```

Then we configure uhttpd to listen on the specific IP address:
```
root@OpenWrt:~# sed -i 's/0\.0\.0\.0/192.168.1.1/' /etc/config/uhttpd
```
and reboot.

After the device reboots, we check that uhttpd now only listens on the IPv6 addresses:
```
root@OpenWrt:~# netstat -tlnp | grep uhttpd
tcp        0      0 :::80                   :::*                    LISTEN      2065/uhttpd
tcp        0      0 :::443                  :::*                    LISTEN      2065/uhttpd
```
and the log shows the reason:
```
root@OpenWrt:~# logread | grep -E 'Starting|uhttpd'
Mon Apr 21 13:08:47 2025 daemon.crit network: Starting network delay ...
Mon Apr 21 13:08:47 2025 daemon.err uhttpd[2065]: bind(): Address not available
Mon Apr 21 13:08:47 2025 daemon.err uhttpd[2065]: bind(): Address not available
Mon Apr 21 13:08:52 2025 daemon.crit network: Starting network now ...
```

### Actual behaviour

uhttpd does not listen on addresses that are not configured yet when it starts.

### Expected behaviour

uhttpd can listen on all IP addresses that eventually get configured.

### Additional info

Possible solutions might include: use of IP_FREEBIND by uhttpd; restart of the uhttpd triggered after network configuration finishes; ... Some other approach?

Is there a quick workaround?

### Diffconfig

```text

```

### Terms

- [x] I am reporting an issue for OpenWrt, not an unsupported fork.

adelton, 2025-04-21:
It seems the solution I used in https://github.com/openwrt/packages/issues/25795 for collectd-mod-network works for uhttpd as well: create `/etc/init.d/uhttpd_network_ready` with
```
#!/bin/sh /etc/rc.common

USE_PROCD=1
START=78
STOP=99

start_service() {
	procd_open_instance
	procd_set_param command /bin/true
	procd_close_instance
}
service_triggers() {
	procd_add_interface_trigger "interface.*.up" "lan" /etc/init.d/uhttpd restart
}
```
run
```
root@OpenWrt:~# chmod a+x /etc/init.d/uhttpd_network_ready
root@OpenWrt:~# /etc/init.d/uhttpd_network_ready enable
root@OpenWrt:~# reboot
```

After reboot, enjoy LuCI available on http://192.168.1.1/ again, and verify with
```
root@OpenWrt:~# netstat -tlnp | grep uhttpd
tcp        0      0 192.168.1.1:443         0.0.0.0:*               LISTEN      2692/uhttpd
tcp        0      0 192.168.1.1:80          0.0.0.0:*               LISTEN      2692/uhttpd
tcp        0      0 :::80                   :::*                    LISTEN      2692/uhttpd
tcp        0      0 :::443                  :::*                    LISTEN      2692/uhttpd
```
that indeed uhttpd listens just where we wanted it to listen.

leomeinel, 2025-05-27:
I can confirm this. Only started happening after upgrading from 23.05

koshkamau, 2025-06-13:
I just got this behavior on 23.05.5 r24106-10cc5fcd00 after adding two more wireguard interfaces in my config.
Strangely enough it didn't disappear when I removed those interfaces and accompanying rules.
```
Fri Jun 13 22:38:45 2025 daemon.err uhttpd[1820]: bind(): Address not available
Fri Jun 13 22:38:45 2025 daemon.err uhttpd[1820]: bind(): Address not available
```
Anyway, I am glad to find this bugreport because I feared some hardware problem.
I have Xiaomi 4A Gigabit Edition router where I resoldered flash memory and built Openwrt system myself.
I am glad it is not my fault :D

leomeinel, 2026-05-01:
## Update

This still seems to be an issue. I just encountered it again on: `OpenWrt 25.12.2, r32802-f50512027` on Zyxel NWA50AX Pro.