openwrt/openwrt #18561: ERROR: luci-app-openvpn-server-3.0-r0: trying to overwrite etc/config/openvpn owned by openvpn-openssl-2.6.14-r1.

seaks, 2025-04-22:
ERROR: luci-app-openvpn-server-3.0-r0: trying to overwrite etc/config/openvpn owned by openvpn-openssl-2.6.14-r1.
1 error; 294 MiB in 424 packages
make[2]: *** [package/Makefile:99: package/install] Error 1
make[1]: *** [package/Makefile:184: /home/seaks/immortalwrt/staging_dir/target-x86_64_musl/stamp/.package_install] Error 2
seaks@debian:~/immortalwrt$ 
seaks@debian:~/immortalwrt$ 
seaks@debian:~/immortalwrt$

LGA1150, 2025-04-30:
`luci-app-openvpn-server` is not an officially maintained app. Please complain to the app's author.

1715173329, 2025-05-15:
The source you used is from immortalwrt, not openwrt. Upstream has nothing to do with it.