openwrt/openwrt #18563: SNR information shows 0/111db in gui and in command line on Gli.net B3000

matrix200, 2025-04-22:
### Describe the bug

Wrong SNR values are reported in both LUCI and with command line utils
[image]

### OpenWrt version

r29212-b5b170e155

### OpenWrt release

SNAPSHOT

### OpenWrt target/subtarget

qualcommax/ipq50xx

### Device

GL.iNet B3000, Inc. IPQ5018/AP-MP03.5-C1

### Image kind

Official downloaded image

### Steps to reproduce

1. Associate any wireless device to the router's SSID
2. Login to LUCI web interface
3. Check the SNR ratio in the list of associated devices.

### Additional info

This happens both in Luci and in command line.
This also happens when the router is used in WDS client mode.

### Diffconfig

```text

```

### Terms

- [x] I am reporting an issue for OpenWrt, not an unsupported fork.

TheRootED24, 2025-04-22:
just to add on some more info..

its something iwinfo-ubus ...its not harvesting the signal or noise properly for what ever reason, it may run deeper. 

- Signal populates properly during/ after scan in the found devices window.
- Signal and Noise populate when the interface first comes up.
- Signal and Noise remain correct until the first update, it then goes blank and propagates up the list from clients to station.
- Both Signal and Noise populate intermittent and seemly randomly, usually lasting for 2-5min before blank again 
-  During the later condition, the graphs working when the S n R are populating properly.
 
I had some success tinkering with macs, but it seemed to be totally random and not something that really shouldn't have an  impact this anyway.  Certain mac combinations seems to affect this condition for whatever reason, in using certain (random) macs for the phy's, I was able to get usually 3/4 interfaces populating correctly, never all 4 at the same time, and i can't overstate how random it seemed

matrix200, 2025-04-22:
Just some output of iwinfo in two consecutive runs:
root@OpenWrt:~# iwinfo
phy1-sta0 ESSID: "SonosWIFI"
          Access Point: 94:83:C4:A4:9F:D7
          Mode: Client  Channel: 36 (5.180 GHz)  HT Mode: VHT80
          Center Channel 1: 42 2: unknown
          Tx-Power: 26 dBm  Link Quality: 0/70
          Signal: unknown  Noise: -111 dBm
          Bit Rate: 780.0 MBit/s
          Encryption: WPA2 PSK (CCMP)
          Type: nl80211  HW Mode(s): 802.11ac/ax/n
          Hardware: embedded [Qualcomm Atheros QCN6102/QCN6122]
          TX power offset: none
          Frequency offset: none
          Supports VAPs: yes  PHY name: phy1

root@OpenWrt:~# iwinfo phy1 info
phy1      ESSID: "SonosWIFI"
          Access Point: 94:83:C4:A4:9F:D7
          Mode: Client  Channel: 36 (5.180 GHz)  HT Mode: VHT80
          Center Channel 1: 42 2: unknown
          Tx-Power: 26 dBm  Link Quality: unknown/70
          Signal: unknown  Noise: unknown
          Bit Rate: unknown
          Encryption: unknown
          Type: nl80211  HW Mode(s): 802.11ac/ax/n
          Hardware: embedded [Qualcomm Atheros QCN6102/QCN6122]
          TX power offset: none
          Frequency offset: none
          Supports VAPs: yes  PHY name: phy

TheRootED24, 2025-04-22:
just a another example 
```
phy0      ESSID: unknown                                                        
          Access Point: 94:83:C4:A4:A7:24                                       
          Mode: Master  Channel: 1 (2.412 GHz)  HT Mode: HT40                   
          Center Channel 1: 3 2: unknown                                        
          Tx-Power: 27 dBm  Link Quality: unknown/70                            
          Signal: unknown  Noise: unknown                                       
          Bit Rate: unknown                                                     
          Encryption: unknown                                                   
          Type: nl80211  HW Mode(s): 802.11ax/b/g/n                             
          Hardware: embedded [Qualcomm Atheros IPQ5018]                         
          TX power offset: none                                                 
          Frequency offset: none                                                
          Supports VAPs: yes  PHY name: phy0
root@OpenWrt:~# iwinfo phy1 info                                                
phy1      ESSID: unknown                                                        
          Access Point: 94:83:C4:A4:A7:25                                       
          Mode: Master  Channel: 157 (5.785 GHz)  HT Mode: VHT80                
          Center Channel 1: 155 2: unknown                                      
          Tx-Power: 26 dBm  Link Quality: unknown/70                            
          Signal: unknown  Noise: unknown                                       
          Bit Rate: unknown                                                     
          Encryption: unknown                                                   
          Type: nl80211  HW Mode(s): 802.11ac/ax/n                              
          Hardware: embedded [Qualcomm Atheros QCN6102/QCN6122]                 
          TX power offset: none                                                 
          Frequency offset: none                                                
          Supports VAPs: yes  PHY name: phy1                                    
root@OpenWrt:~# 
```

psyke83, 2025-04-24:
I noticed the same issue with my GL-B3000 router. I intended to provision it as a AP-STA repeater via travelmate, but it refuses to recognize a connection to the uplink network (via either radio); I suspect it's due to the signal not being reported correctly. Setting up the AP manually instead of using travelmate works OK and will stay connected, though.

Possibly OT, but I also noticed that the OpenWrt firmware has much poorer performance and range on the 2.4Ghz radio. In the same test conditions, my client can struggle to get 70Mbps on OpenWrt but the OEM firmware reaches ~150Mbps (even if I disable NSS to make the comparison fairer). Additionally, the IPQ5018 radio supports a max tx-power of 30dBm on the OEM firmware compared to 20dBm on OpenWrt.

TheRootED24, 2025-04-25:
>Additionally, the IPQ5018 radio supports a max tx-power of 30dBm on the OEM firmware compared to 20dBm on OpenWrt.

country code ( US ) allows 30dbm

psyke83, 2025-04-27:
> country code ( US ) allows 30dbm

Thanks, that worked, but I'm still almost 100% certain that (only) the 2.4G radio's signal strength is not being applied correctly.

When set to US on both radios, I can select up to 27dBm for 2.4G and 24dBm for 5G. Measuring with a stationary client via ìw wl2sp0 link`:

| Radio  | Signal at 0dBm | Signal at max setting (27/24dBm) |
| ------------- | ------------- | ------------- | 
| 2.4G  | -61  | -61 |
| 5G | -84  | -54  |

So it seems that the 5G radio is setting the power correctly, but the 2.4G radio is not. In fact, at max power, the 5G signal is stronger at the far side of the house from where the router is situated, which should not be the case given 2.4G's higher power on the router and better penetration of that band in general.

I suspect that whatever default the 2.4G radio is really stuck on is definitely not max power, as the signal is much worse compared to the OEM at max power. It's really unfortunate because this is the only thing keeping me from switching for now.

TheRootED24, 2025-04-27:
i agree something is not right with the 5018 (2.4g) radio but i am not convinced its directly to do with  txpower, as seems to be assigned correctly..i think the problem runs deeper. I am looking into this currently, I found a couple mismatched variables types , the driver was sending u8 type rssi .. firmware sends s8  .. iwinfo expects s8. This has fixed the signal on 2.4g station, but ap need more investigation   

>root@OpenWrt:~# iwinfo phy0-ap0 info
phy0-ap0  ESSID: "OpenWrt (2.4ghz)"
          Access Point: 94:83:C4:A4:A7:24
          Mode: Master  Channel: 1 (2.412 GHz)  HT Mode: HT20
          Center Channel 1: 1 2: unknown
          Tx-Power: 27 dBm  Link Quality: unknown/70
          Signal: unknown  Noise: -111 dBm
          Bit Rate: unknown
          Encryption: WPA2 PSK (CCMP)
          Type: nl80211  HW Mode(s): 802.11ax/b/g/n
          Hardware: embedded [Qualcomm Atheros IPQ5018]
          TX power offset: none
          Frequency offset: none
          Supports VAPs: yes  PHY name: phy0
>

>2.4g station 
root@OpenWrt:~# iwinfo phy0-sta0 info
phy0-sta0 ESSID: "GL-X750-db1"
          Access Point: 96:83:C4:A4:A7:24
          Mode: Client  Channel: 1 (2.412 GHz)  HT Mode: HT20
          Center Channel 1: 1 2: unknown
          Tx-Power: 27 dBm  Link Quality: 54/70
          Signal: -56 dBm  Noise: -108 dBm
          Bit Rate: 39.0 MBit/s
          Encryption: unknown
          Type: nl80211  HW Mode(s): 802.11ax/b/g/n
          Hardware: embedded [Qualcomm Atheros IPQ5018]
          TX power offset: none
          Frequency offset: none
          Supports VAPs: yes  PHY name: phy0
>

TheRootED24, 2025-11-15:
I have found the SNR issue is between iw and the ath11k driver. iw is not parsing signal and avg_signal correctly. I am currently using a workaround method, using the ack_signal and ack_signal_avg, which seem to accurate as the oem runn ing next to it in terms of SNR.

you can try this patch to iwinfo 

```
diff --git a/iwinfo_nl80211.c b/iwinfo_nl80211.c
index ba7b954..a005740 100644
--- a/iwinfo_nl80211.c
+++ b/iwinfo_nl80211.c
@@ -1607,7 +1607,7 @@ static int nl80211_fill_signal_cb(struct nl_msg *msg, void *arg)
                {
                        if (sinfo[NL80211_STA_INFO_SIGNAL])
                        {
-                               dbm = nla_get_u8(sinfo[NL80211_STA_INFO_SIGNAL]);
+                               dbm = nla_get_u8(sinfo[NL80211_STA_INFO_ACK_SIGNAL]);
                                rr->rssi = (rr->rssi * rr->rssi_samples + dbm) / (rr->rssi_samples + 1);
                                rr->rssi_samples++;
                        }
@@ -2306,10 +2306,10 @@ static int nl80211_get_assoclist_cb(struct nl_msg *msg, void *arg)
                              attr[NL80211_ATTR_STA_INFO], stats_policy))
        {
                if (sinfo[NL80211_STA_INFO_SIGNAL])
-                       e->signal = nla_get_u8(sinfo[NL80211_STA_INFO_SIGNAL]);
+                       e->signal = nla_get_u8(sinfo[NL80211_STA_INFO_ACK_SIGNAL]);
 
                if (sinfo[NL80211_STA_INFO_SIGNAL_AVG])
-                       e->signal_avg = nla_get_u8(sinfo[NL80211_STA_INFO_SIGNAL_AVG]);
+                       e->signal_avg = nla_get_u8(sinfo[NL80211_STA_INFO_ACK_SIGNAL_AVG]);
 
                if (sinfo[NL80211_STA_INFO_INACTIVE_TIME])
                        e->inactive = nla_get_u32(sinfo[NL80211_STA_INFO_INACTIVE_TIME]);
```

give it a name and put it in packages/network/utils/iwinfo/patches/101-ath11k-ipq50xx-iwinfo-SNR-workaround.patch.

Also you'll need the correct iwinfo, I am currently using this patch 
>Makefile
```
PKG_SOURCE_PROTO:=git
PKG_SOURCE_URL=$(PROJECT_GIT)/project/iwinfo.git
PKG_SOURCE_DATE:=2025-11-07
PKG_SOURCE_VERSION:=5f4c213fc59f1a82d9f0140642d668ae2969531b
PKG_MIRROR_HASH:=363e245821d9bc6ffe2cf20f4b51670f1f91928df22a2ffe73a06767f737fb6d
PKG_MAINTAINER:=Jo-Philipp Wich <jo@mein.io>
PKG_LICENSE:=GPL-2.0

```

[image]

TheRootED24, 2025-11-15:
just to be clear on what we're populating and the difference ..

>info_signal: This is the received signal strength indication (RSSI) of data frames from the connected wireless client (or access point, depending on the context). It indicates how strong the signal from the other device is as measured by your device's receiver.

>info_ack_signal: This is the received signal strength indication (RSSI) of the acknowledgment (ACK) frames sent back by the other device. In Wi-Fi, after a device sends a unicast data frame, it expects an ACK frame in return to confirm successful receipt. The info_ack_signal measures the strength of that specific return signal as received by your device.

psyke83, 2025-11-16:
I will investigate your patches soon - thanks. But of course that seems like a workaround rather than a proper fix.

I notice this (old) commit which might be relevant: https://github.com/torvalds/linux/commit/b488c766442f7d9c07ea0708aa2a1a6bff1baef5

If the description is still accurate for the driver today, it means that the kernel needs to be built with CONFIG_ATH11K_DEBUGFS in order for RSSI stats to be reported by supported chips. This might not fix the issue as-is but might be a starting step? I'm not an expert on the openwrt build process, but I can't find any reference to this config being set from a cursory grep of the source.