openwrt/openwrt #18565: NTP sync / Clock problem with MT6000 - OpenWRT 24.10.x

cryptage21, 2025-04-22:
### Describe the bug

Hello,

I have a problem with NTP on OpenWRT 24.10.x with GL.iNet MT6000. 
NTP sync is totally unstable (1-3-17-77-177-377-0-1-3-17....). 

I've disabled sysntpd, checked dmesg, checked logread, ran ntpd in debug mode but nothing helped.

```    
 remote           refid      st t when poll reach   delay   offset  jitter
==============================================================================
*ntp-server      172.16.5.249     2 u    -   64   17    0.257  +148.03  72.326
*ntp-server      172.16.5.249     2 u   64   64   17    0.257  +148.03  72.326
*ntp-server      172.16.5.249     2 u   32   64   37    0.257  +148.03  73.808
*ntp-server      172.16.5.249     2 u    -   64   77    0.257  +148.03  88.328
*ntp-server      172.16.5.249     2 u   64   64   77    0.257  +148.03  88.328
*ntp-server      172.16.5.249     2 u   32   64  177    0.257  +148.03 117.920
*ntp-server      172.16.5.249     2 u    -   64  377    0.257  +148.03 154.872
*ntp-server      172.16.5.249     2 u   64   64  377    0.257  +148.03 154.872
*ntp-server      172.16.5.249     2 u   41   64  377    0.257  +148.03 195.517
*ntp-server      172.16.5.249     2 u   36   64  377    0.257  +148.03 237.222
 ntp-server      .STEP.          16 u  406   64    0    0.000   +0.000   0.000
 ntp-server      172.16.5.249     2 u   29   64    1    0.782  +164.63   0.000
 ntp-server      172.16.5.249     2 u   49   64    1    0.397  +188.25  16.722
*ntp-server      172.16.5.249     2 u   58   64    3    0.375  +219.86  28.802
*ntp-server      172.16.5.249     2 u   26   64    7    0.375  +219.86  33.771
*ntp-server      172.16.5.249     2 u   90   64    7    0.375  +219.86  33.771
*ntp-server      172.16.5.249     2 u   58   64   17    0.375  +219.86  49.059
*ntp-server      172.16.5.249     2 u   26   64   37    0.375  +219.86  77.997
*ntp-server      172.16.5.249     2 u   90   64   37    0.375  +219.86  77.997
*ntp-server      172.16.5.249     2 u   58   64   77    0.375  +219.86 112.834
*ntp-server      172.16.5.249     2 u   26   64  177    0.375  +219.86 152.720
*ntp-server      172.16.5.249     2 u   91   64  177    0.375  +219.86 152.720
*ntp-server      172.16.5.249     2 u   59   64  377    0.375  +219.86 196.306
 ntp-server      .STEP.          16 u  200   64    0    0.000   +0.000   0.000
 ntp-server      172.16.5.249     2 u   11   64    1    0.444  +44.830   0.000
 ntp-server      172.16.5.249     2 u   13   64    1    0.380  +79.623  24.616
*ntp-server      172.16.5.249     2 u   40   64    3    0.332  +100.52  25.917
*ntp-server      172.16.5.249     2 u    8   64    7    0.332  +100.52  25.526
*ntp-server      172.16.5.249     2 u   72   64    7    0.332  +100.52  25.526
*ntp-server      172.16.5.249     2 u   40   64   17    0.332  +100.52  25.478
*ntp-server      172.16.5.249     2 u    8   64   37    0.332  +100.52  43.217
*ntp-server      172.16.5.249     2 u   72   64   37    0.332  +100.52  43.217
*ntp-server      172.16.5.249     2 u   40   64   77    0.332  +100.52  67.498
*ntp-server      172.16.5.249     2 u    8   64  177    0.332  +100.52  97.485
*ntp-server      172.16.5.249     2 u   72   64  177    0.332  +100.52  97.485
*ntp-server      172.16.5.249     2 u   40   64  377    0.332  +100.52 131.986
*ntp-server      172.16.5.249     2 u    8   64  377    0.332  +100.52 170.694
*ntp-server      172.16.5.249     2 u   72   64  377    0.332  +100.52 170.694
*ntp-server      172.16.5.249     2 u   40   64  377    0.389  +438.91 212.152
*ntp-server      172.16.5.249     2 u   38   64  377    0.389  +438.91 176.036
 ntp-server      .STEP.          16 u   59   64    0    0.000   +0.000   0.000
 ntp-server      172.16.5.249     2 u   34   64    1    0.327  +37.867   0.000
 ntp-server      172.16.5.249     2 u    1   64    1    0.303  +64.954  23.663
 ntp-server      172.16.5.249     2 u   63   64    3    0.303  +64.954  25.396
 ntp-server      172.16.5.249     2 u   31   64    7    0.303  +64.954  39.224
 ntp-server      172.16.5.249     2 u   95   64    7    0.303  +64.954  39.224
 ntp-server      172.16.5.249     2 u   63   64   17    0.303  +64.954  64.250
*ntp-server      172.16.5.249     2 u   31   64   37    0.304  +66.102  95.914
*ntp-server      172.16.5.249     2 u   95   64   37    0.304  +66.102  95.914
*ntp-server      172.16.5.249     2 u   63   64   77    0.346  +93.744 106.782
*ntp-server      172.16.5.249     2 u   31   64  177    0.330  +315.08 165.766
*ntp-server      172.16.5.249     2 u   95   64  177    0.330  +315.08 165.766
*ntp-server      172.16.5.249     2 u   63   64  377    0.330  +315.08 143.454
```

Same configuration on Xiaomi AX3000T (same version) : 

```
[... 100 lines cut ...]
        option enabled '1'
        option enable_server '0'
        list server 'ntp-server'
```

Check NTP sync. 

### Actual behaviour

NTP is unstable and "restarts".

### Expected behaviour

Should be stable. 
A second router with same version does not have the problem. 

### Diffconfig

```text

```

### Terms

- [x] I am reporting an issue for OpenWrt, not an unsupported fork.

cryptage21, 2025-04-22:
I've tried with Chrony and the problem doesn't seem to exist.

mlichvar, 2025-04-23:
What frequency does `chronyc tracking` report? If it's larger than 500 ppm, ntpd wouldn't be able to compensate for it. It would most likely be a kernel misconfiguration/bug, causing the clocksource to run too slow or fast.

cryptage21, 2025-04-23:
Thanks. 

``` 
chronyc tracking
Frequency       : 1065.248 ppm slow
```

stefanct, 2025-12-03:
I did not invest too much more time yet but there is a MT7986 application note "MT7986A/B/C HW Design Notice" online that recommends 40 MHz XTALs with <20 PPM over their full temperature range (in relation to the 20°C value). I'd doubt the both board designers completely fucked that up.
I presume this external clock is used to derive the system clock via some PLL in the MT part of the CPU and eventually fed to the ARM "arch timer" instances available to the 4 CPU cores (which I presume is used by Linux for the local clock). All of that has to be verified and we should try to get some measurements going as well. I have looked through the upstream linux changes related to the MT7986's device tree and if that's any indication I would not be surprised if the culprit is within the kernel code from MT. with more documentation (especially a schematic of the routers' PCBs) it would be much easier as always. :/

PalebloodSky, 2026-05-12:
Interesting solution, I never considered Chrony, going to give that a try. There is an extensive feature comparison on their site and the table seems to confirm what you're saying too (e.g. drift tracking):  https://chrony-project.org/comparison.html

It seems many Linux distros use chrony, perhaps OpenWrt devs could consider dropping ntpd and switching to chrony for 26.xx?