openwrt/openwrt #18569: bpf-headers: failed to build

pesa1234, 2025-04-22:
### Describe the bug

Latest snapshot failed to build bpf-headers.

Output of verborse V=s:

`make[3]: Entering directory '/home/ubundesk/owrt/MT6000/private/owrtBuildTree/openwrt/package/kernel/bpf-headers'
Makefile:18: /home/ubundesk/owrt/MT6000/private/owrtBuildTree/openwrt/include/kernel-6.6: No such file or directory
make[3]: *** No rule to make target '/home/ubundesk/owrt/MT6000/private/owrtBuildTree/openwrt/include/kernel-6.6'.  Stop.
make[3]: Leaving directory '/home/ubundesk/owrt/MT6000/private/owrtBuildTree/openwrt/package/kernel/bpf-headers'
time: package/kernel/bpf-headers/compile#0.54#0.05#0.59
    ERROR: package/kernel/bpf-headers failed to build.
make[2]: *** [package/Makefile:189: package/kernel/bpf-headers/compile] Error 1`

### OpenWrt version

r29312-d89f5f3ab0

### OpenWrt release

SNAPSHOT

### OpenWrt target/subtarget

mediatek/filogic

### Device

GL.iNet GL-MT6000

### Image kind

Self-built image

### Steps to reproduce

`git pull` on latest main
make -j1 defconfig download clean world V=s

### Actual behaviour

ERROR: package/kernel/bpf-headers failed to build.

### Expected behaviour

it should compile as before

### Additional info

@Ansuel

I'm sorry, but I think that this commit: https://github.com/openwrt/openwrt/commit/8865dadea7b94e7859b416d3b1931b897ea43c48
is responsible of failing.

`git revert 8865dadea7b94e7859b416d3b1931b897ea43c48`
and a new make solve the issue

Thanks

### Diffconfig

```text

```

### Terms

- [x] I am reporting an issue for OpenWrt, not an unsupported fork.

openips, 2025-04-22:
cat  logs/package/kernel/bpf-headers/dump.txt

```
Package: bpf-headers
Version: 6.6
Depends: +libc
Conflicts:
Menu-Depends:
Provides:
Section: kernel
Category: Kernel modules
Repository: base
Title: eBPF kernel headers
Maintainer:
Source: linux-6.6.tar.xz
Type: ipkg
Build-Only: 1
Hidden: 1
Description: eBPF kernel headers
@@

Makefile:18: include/kernel-6.6: No such file or directory

```

ptpt52, 2025-04-23:
```diff
diff --git a/package/kernel/bpf-headers/Makefile b/package/kernel/bpf-headers/Makefile
index a644f47ab72..bc2bbdc170b 100644
--- a/package/kernel/bpf-headers/Makefile
+++ b/package/kernel/bpf-headers/Makefile
@@ -15,7 +15,7 @@ include $(INCLUDE_DIR)/kernel.mk
 PKG_NAME:=linux
 PKG_PATCHVER:=6.6
 # Manually include kernel version and hash from kernel details file
-include $(INCLUDE_DIR)/kernel-$(PKG_PATCHVER)
+include $(INCLUDE_DIR)/../target/linux/generic/kernel-$(PKG_PATCHVER)
 
 PKG_VERSION:=$(PKG_PATCHVER)$(strip $(LINUX_VERSION-$(PKG_PATCHVER)))
 PKG_SOURCE:=$(PKG_NAME)-$(PKG_VERSION).tar.xz
```

namiltd, 2025-04-23:
Try this: https://github.com/openwrt/openwrt/pull/18568

robimarko, 2025-04-23:
Fixed via 0e05cd6a153921c16bd79128d06125d87ca3d968

closed 2025-04-23 by robimarko