openwrt/openwrt #18577: FYI: upcoming TLS certificate behaviour changes

systemcrash, 2025-04-23:
Affects Public CA issued certificates - but it seems this spec may be mandated by browsers in the future.

The implication is to move to automated renewal (ACME)

https://www.digicert.com/blog/tls-certificate-lifetimes-will-officially-reduce-to-47-days

The maximum certificate lifetime is going down:

-   From today until March 15, 2026, the maximum lifetime for a [TLS certificate](https://www.digicert.com/tls-ssl/tls-ssl-certificates) is 398 days.
*  As of March 15, 2026, the maximum lifetime for a TLS certificate will be 200 days.
*  As of March 15, 2027, the maximum lifetime for a TLS certificate will be 100 days.
*  As of March 15, 2029, the maximum lifetime for a TLS certificate will be 47 days.

brada4, 2025-04-26:
Back-date all certs to march 14?

Djfe, 2025-04-30:
While this is useful to know I don't think there is anything that needs to be changed within OpenWrt, is there?
Or should self-signed tls-certificates by OpenWrt also be adjusted according to these specifications to adhere to browser specs?

Djfe, 2025-04-30:
> Back-date all certs to march 14?

I don't think so. It just means, that certificates issued after day x will have a shorter life span