openwrt/openwrt #18583: SDK has a non fixed base feed

thom24, 2025-04-24:
Using the SDK v24.10.1 I noticed that the base feed is not set to a specific tag/commit, only the branch is defined.
```
$ cat feeds.conf.default
src-git-full base https://git.openwrt.org/openwrt/openwrt.git;openwrt-24.10
src-git packages https://git.openwrt.org/feed/packages.git^d8cd30f4e281d6853b3de134c4f147a807583e43
src-git luci https://git.openwrt.org/project/luci.git^2ac26e56cc55102cb10e7b0867c2b78e0f6d5fd8
src-git routing https://git.openwrt.org/feed/routing.git^c9b636698881059a3c981032770968f5a98ff201
src-git telephony https://git.openwrt.org/feed/telephony.git^fd605af7143165a2490681ec1752f259873b9147
```
The code looks ok
https://github.com/openwrt/openwrt/blob/27adf03f703319f4a6900b07be5f6786045a6dad/target/sdk/Makefile#L41-L50
Then I built myself the SDK, and in this case the tag v24.10.1 is set
```
$ cat build_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/openwrt-sdk-stm32-stm32mp1_gcc-13.3.0_musl_eabi.Linux-x86_64/feeds.conf.default
src-git-full base git://git.openwrt.org/openwrt/openwrt.git;v24.10.1
src-git packages https://git.openwrt.org/feed/packages.git^d8cd30f4e281d6853b3de134c4f147a807583e43
src-git luci https://git.openwrt.org/project/luci.git^2ac26e56cc55102cb10e7b0867c2b78e0f6d5fd8
src-git routing https://git.openwrt.org/feed/routing.git^c9b636698881059a3c981032770968f5a98ff201
src-git telephony https://git.openwrt.org/feed/telephony.git^fd605af7143165a2490681ec1752f259873b9147
```

thom24, 2025-04-24:
Just checked, 23.05 and 22.03 do this too.

thom24, 2025-05-12:
The problem is that I'm not able to build the exact same SDK. 
And it is weird to have non fixed base feed whereas other feeds are fixed.

@hauke @Ansuel @aparcar, do you have an idea ?

Ansuel, 2025-05-12:
Honestly it's something I also notice but this made the benefits of permitting to fix bugs in stable branch without having to retire the entire SDK. (there was one case with hostapd broken that was fixed with this trick)

It's something we need to discuss whether it was originally intended or it's a very old bug nobody notice.

ynezz, 2025-05-13:
> Just checked, 23.05 and 22.03 do this too.

In 21.02 it was OK, so its regression

```
src-git base https://git.openwrt.org/openwrt/openwrt.git;v21.02.7
src-git-full packages https://git.openwrt.org/feed/packages.git^48242ee7a190db7740f7b9b3ef1debfa4d5857f6
src-git-full luci https://git.openwrt.org/project/luci.git^e98243ef9eb838cca80cdd6d1bd0cf69a509d103
src-git-full routing https://git.openwrt.org/feed/routing.git^8071852b4556a02533cacb7a0f6a432df3507302
src-git-full telephony https://git.openwrt.org/feed/telephony.git^920fbc5c0a2e4badf51bceff42e9a1e3eb693462
```

>  Then I built myself the SDK, and in this case the tag v24.10.1 is set

It might be caused by changes in buildbot itself (`Git()` source step) or by something https://github.com/openwrt/buildbot/blob/main/phase1/master.cfg.

> The code looks ok

Does it? Shouldn't the logic be switched, prefer the `GIT_TAGNAME` if available? Now the `GIT_BRANCH` has precedence.

```diff
-  BASE_FEED:=$(if $(GIT_URL),src-git-full base $(GIT_URL)$(if $(GIT_BRANCH),;$(GIT_BRANCH),$(if $(GIT_TAGNAME),;$(GIT_TAGNAME)))) 
+  BASE_FEED:=$(if $(GIT_URL),src-git-full base $(GIT_URL)$(if $(GIT_TAGNAME),;$(GIT_TAGNAME),$(if $(GIT_BRANCH),;$(GIT_BRANCH)))) 
```

Ideally, avoid a bit vague Git tags and use the `GIT_COMMIT` for base feed

```diff
-  BASE_FEED:=$(if $(GIT_URL),src-git-full base $(GIT_URL)$(if $(GIT_BRANCH),;$(GIT_BRANCH),$(if $(GIT_TAGNAME),;$(GIT_TAGNAME)))) 
+  BASE_FEED:=$(if $(GIT_URL),src-git-full base $(GIT_URL)$(if $(GIT_TAGNAME),;$(GIT_COMMIT),$(if $(GIT_BRANCH),;$(GIT_BRANCH)))) 
```

Ansuel, 2025-05-13:
Thanks for checking 21.02, so my suspicious was right that this is a very old bug.

closed 2025-12-04 by openwrt-bot