openwrt/openwrt #18586: Cannot Select vLAN Interfaces in Radio Wi-Fi Access Point Settings

HexKernel, 2025-04-24:
### Describe the bug

In the newest release (24.10.1), I’m unable to select any of my vLAN network interfaces under the Radio Wi-Fi Access Point settings, even though this worked correctly in version 23.05.5. The save button does not work. See the attached screenshot for details. Any assistance would be appreciated.

### OpenWrt version

r28597-0425664679

### OpenWrt release

24.10.1

### OpenWrt target/subtarget

bcm27xx/bcm2711

### Device

Raspberry Pi 4 Model B Rev 1.5

### Image kind

Official downloaded image

### Steps to reproduce

Selecting a vLAN interface under the Radio Wi-Fi Access Point settings.

[image]

### Diffconfig

```text

```

### Terms

- [x] I am reporting an issue for OpenWrt, not an unsupported fork.

HexKernel, 2025-04-24:
Temporary Workaround Implemented:

1. Created a bridge adapter named br-wireless with vlan40 as its member interface
2. Configured the br-wireless interface in the network interface management section
3. Set the bridge protocol to unmanaged mode (since it functions solely as a Layer 2 pass-through)
4. Assigned the br-wireless interface as the upstream interface for the WiFi network

mattymuc, 2025-04-24:
I can confirm this bug on OpenWrt One and some Zyxel Router. It was working in 24.10.0, so this bug is new in 24.10.1. I downgraded to 24.10.0.

HexKernel, 2025-04-25:
I would like to add that even on version 23.05.5, with the ability to assign a VLAN network interface and make it functional, it would only work in unencrypted mode—meaning, when there was no password on the WiFi access point, clients could connect. However, using any supported encryption mode would bring up the WiFi AP, but clients would get stuck during password authentication: they would briefly connect, get stuck at authentication, and then disconnect, repeating the cycle. I tried every available encryption method on the list, and none worked except unencrypted mode. However, once I created a bridge with that VLAN and assigned the bridge to the WiFi AP, all supported encryption modes worked smoothly without any issues.

HexKernel, 2025-04-25:
Although this method temporarily works, as soon as a client connects, the IP address of the interface stops responding — I can no longer ping it or access LuCI — even though the client remains connected. After a reboot, the same VLAN interface (configured as a DHCP client) fails to obtain an IP address and becomes non-functional.

In this scenario, VLAN40 is set as a DHCP client. I then changed its protocol to static (i.e., assigning it a static IP and running a DHCP server). After testing, Wi-Fi clients were able to connect and authenticate, but they got stuck during DHCP IP assignment and couldn't obtain an IP. Assigning the Wi-Fi AP directly to the static, DHCP-enabled network caused authentication issues again for the Wi-Fi clients.

HexKernel, 2025-04-25:
If anyone has a solution to this, I would greatly appreciate it if you could share a fix or workaround. I urgently need to get the Wi-Fi interface up and running, and I've already tried many different methods—none of which have worked so far.

mattymuc, 2025-04-25:
As a workaround, downgrade to 24.10.0 as it works there without problems.

HexKernel, 2025-04-25:
Just to confirm — you were able to assign a VLAN network interface to your Wi-Fi AP and tested it with both static (DHCP server) and DHCP client configurations? And in both cases, the AP came up and functioned properly, Wi-Fi clients were able to connect and authenticate without issues, and the OpenWrt interface IP remained accessible and pingable from the Wi-Fi clients?

Could this difference in behavior be due to your device being a Zyxel router and mine a Raspberry Pi? Maybe proper Wi-Fi support is still lacking for the bcm27xx/bcm2711 image, despite OpenWrt’s documentation stating the issue was resolved in version 21.02 and later

HexKernel, 2025-04-25:
I tried the following releases, and in all of them this issue is present:

24.10.1
24.10.0
23.05.5
23.05.0