openwrt/openwrt #18588: Vr. 24.10.1: PPTP Protocol Missing After Upgrading luci-proto-ppp

HexKernel, 2025-04-24:
### Describe the bug

In version 23.05.5, upgrading the luci-proto-ppp package immediately added “PPTP” to the Add New Interface → Protocol list in LuCI. On the latest 24.10.1 release, however, the PPTP option does not appear—even after installing or upgrading luci-proto-ppp, logging out/in, or rebooting. Screenshots attached.

### OpenWrt version

r28597-0425664679

### OpenWrt release

24.10.1

### OpenWrt target/subtarget

bcm27xx/bcm2711

### Device

Raspberry Pi 4 Model B Rev 1.5

### Image kind

Official downloaded image

### Steps to reproduce

1. Fresh install of OpenWrt 24.10.1.
2. Go to System → Software, click Update lists, then install or upgrade luci-proto-ppp.
3. Navigate to Network → Interfaces → Add new interface → Protocol.
4. Observe that “PPTP” is not listed.

[image]
[image]

### Diffconfig

```text

```

### Terms

- [x] I am reporting an issue for OpenWrt, not an unsupported fork.

brada4, 2025-04-27:
Do proper owut upgrade, luci part  versions are out of sync

HexKernel, 2025-04-29:
I was only able to resolve this issue by downgrading to the 23.05.5 version.

brada4, 2025-04-29:
You need rather massive ppp-mod-pptp package which is not dependency pf luci-proto package.

HexKernel, 2025-04-29:
Technically, if I install the package and run it from the CLI, there wouldn’t be any issue. However, in that case, I wouldn’t be able to monitor it through the LuCI interface. I would like to monitor it through LuCI as well, especially for troubleshooting connectivity issues or for basic users with limited backend knowledge, so they can identify and resolve problems without needing CLI access.

brada4, 2025-04-29:
You can install packages from luci.

HexKernel, 2025-04-29:
Yes, I understand what you mean. I tried that approach as well by installing all the dependencies from LuCI software section as instructed. However, I wasn't able to see the PPTP protocol in the interface section when adding a new interface on version 24.10.1. I tried logging out and back in, refreshing the page, and even rebooting multiple times, but none of these helped. That’s why I saw no other choice but to downgrade, especially since there were other issues with this version, including the Wi-Fi AP 'Save' button not working.

closed 2025-04-30 by HexKernel

Arnie97, 2025-10-08:
Still not working as of 24.10.3