openwrt/openwrt #18589: Hardware flow offloading is not working for WLAN<->LAN

XCas13, 2025-04-24:
### Describe the bug

Hardware flow offloading is not working for WLAN<->LAN, but working very well for WLAN<->WAN and LAN<->WAN, both IPv4 and IPv6.

### OpenWrt version

r28597-0425664679

### OpenWrt release

24.10.1

### OpenWrt target/subtarget

mediatek/filogic

### Device

GL-MT6000

### Image kind

Official downloaded image

### Steps to reproduce

Switch both Hardware offloading and WED on. Run iperf3 between a laptop/pc/whatever connected by WiFi and another one connected by ethernet, e.g lan2 port.

### Actual behaviour

CPU load is about 24% for LAN->WLAN and about 15% in case for LAN<-WLAN. 

### Expected behaviour

CPU load should be close to 0 as if running iperf for LAN/WLAN<->WAN

### Diffconfig

```text

```

### Terms

- [x] I am reporting an issue for OpenWrt, not an unsupported fork.

brada4, 2025-04-25:
There is a `bridger` package to allow offloading non-routed traffic.

LearZhou, 2025-04-25:
Will it work if WLAN and lan are in different private subnet?

brada4, 2025-04-25:
Precisely. There is hardware offload for bridge firewall pending in mainline kernel, so 2-3 years wait ahead.

XCas13, 2025-04-25:
OK, it's not implemented in yet. Thanks for the info.

closed 2025-04-25 by XCas13

brada4, 2025-04-25:
bridger package works in the meantime.

geldot, 2025-05-15:
Linux bridge-fastpath patch for reference:

https://lore.kernel.org/netdev/20250227180913.6248bbd3@kernel.org/T/

https://github.com/ericwoud/linux/commits/bpir-nftflow-net-next