openwrt/openwrt #18590: [qualcommax] Broken image for GL.iNet GL-B3000

danpawlik, 2025-04-25:
### Describe the bug

Hi,
Since 2-3 days there is a broken image created for GL.iNet GL-B3000.
Last known and working image is comming with commit https://github.com/openwrt/openwrt/commit/c3b8108a2be4df3a114ac6d964c664dffd2d5bbf.

I raised discussion on forum.openwrt.org in this [thread](https://forum.openwrt.org/t/ipq5018-glinet-b3000-info/210771/651) and @TheRootED24 check the issue via UART, below logs from booting the device:

```
[    1.507934] mtd: setting mtd11 (rootfs) as root device                                                                                             
[    1.509149] mtdsplit: error occured while reading from "rootfs"                                                                                    
[    1.523918] remoteproc remoteproc0: cd00000.remoteproc is available                                                                                
[    1.524164] qcom-q6-mpd cd00000.remoteproc: pd-1 node found                                                                                        
[    1.531023] remoteproc remoteproc1: pd-1 is available                                                                                              
[    1.534787] qcom-q6-mpd cd00000.remoteproc: pd-3 node found                                                                                        
[    1.541589] remoteproc remoteproc2: pd-3 is available                                                                                              
[    1.551618] ubi0: attaching mtd11                                                                                                                  
[    1.552331] ubi0 warning: ubi_io_read: error -74 (ECC error) while reading 64 bytes from PEB 0:0, read only 64 bytes, retry                        
[    1.554455] ubi0 warning: ubi_io_read: error -74 (ECC error) while reading 64 bytes from PEB 0:0, read only 64 bytes, retry                        
[    1.565931] ubi0 warning: ubi_io_read: error -74 (ECC error) while reading 64 bytes from PEB 0:0, read only 64 bytes, retry                        
[    1.576568] ubi0 error: ubi_io_read: error -74 (ECC error) while reading 64 bytes from PEB 0:0, read 64 bytes                                      
[    1.587146] CPU: 0 PID: 1 Comm: swapper/0 Not tainted 6.6.87 #0                                                                                    
[    1.597153] Hardware name: GL.iNet GL-B3000 (DT)                                                                                                   
[    1.602881] Call trace:                                                                                                                            
[    1.607739]  dump_backtrace+0xa0/0xe0                                                                                                              
[    1.609910]  show_stack+0x18/0x24                                                                                                                  
[    1.613728]  dump_stack_lvl+0x48/0x60                                                                                                              
[    1.617027]  dump_stack+0x18/0x24                                                                                                                  
[    1.620672]  ubi_io_read+0x11c/0x32c                                                                                                               
[    1.623972]  ubi_io_read_ec_hdr+0x50/0x1f0                                                                                                         
[    1.627618]  ubi_attach+0x35c/0x133c                                                                                                               
[    1.631524]  ubi_attach_mtd_dev+0x494/0xaf4                                                                                                        
[    1.635257]  ubi_init_attach+0xac/0x2e4                                                                                                            
[    1.639163]  do_one_initcall+0x6c/0x1fc                                                                                                            
[    1.642983]  kernel_init_freeable+0x204/0x2e4                                                                                                      
[    1.646803]  kernel_init+0x28/0x1dc                                                                                                                
[    1.651316]  ret_from_fork+0x10/0x20       
```

### OpenWrt version

r29281-c3b8108a2b (this works, newer does not)

### OpenWrt release

SNAPSHOT

### OpenWrt target/subtarget

qualcommax/ipq50xx

### Device

GL.iNet GL-B3000

### Image kind

Official downloaded image

### Steps to reproduce

Flash sysupgrade/factory image and after reboot it does not start.

### Diffconfig

```text

```

### Terms

- [x] I am reporting an issue for OpenWrt, not an unsupported fork.

danpawlik, 2025-04-25:
Taken from forum:
@musashino205 pointed, that:

```
The spare size (oobsize) of Winbond W25N01GW used on the GL-B3000 is 64 and you need to set the nand-ecc-strength property in the dts to 4 instead of 8.
This property was not used before that commit and calculated from the registered spare size of the Linux Kernel.
```

More info in dts [file](https://github.com/openwrt/openwrt/pull/17903/files#diff-f77c3c75068ef20f3e5e6ffe2abfb64d60991a885cadc56360abe56fefc4679cR229)

robimarko, 2025-04-25:
Fixed via: https://github.com/openwrt/openwrt/commit/aa6d245e399112d5498344e6432ff197e7bb63df

closed 2025-04-25 by robimarko

TheRootED24, 2025-04-25:
that was quick!

Thanks @musashino205  @robimarko 👍