openwrt/openwrt #18605: IPQ8074A: Add support for Ubiquiti AmpliFi Alien (AFI-ALN-R)

ege-art, 2025-04-26:
### Device Information

- **Model**: Ubiquiti AmpliFi Alien (AFI-ALN-R)
- **SoC**: Qualcomm IPQ8074A (ARM Cortex-A53, quad-core @ 2.2 GHz)
- **RAM**: 1 GB DDR3
- **Storage**: NAND flash-256 MB
- **Wi-Fi**: Tri-band Wi-Fi 6 (Qualcomm-based, possibly with Broadcom 3rd radio)
- **Firmware**: Ubiquiti's customized OpenWrt/LEDE-based system
- **UART Access**: Full access with root shell confirmed

### Collected Data

I have gathered a comprehensive system information dump, including:

- Bootlog (via UART)
- `/proc/cpuinfo`, `/proc/meminfo`, `/proc/mtd`
- `dmesg`, loaded modules, MAC addresses
- MTD partition layout and filesystem details
- Device tree references

📎 **Download Full Dump**: (https://drive.google.com/file/d/1Ll3lGWGWGbYM5mRTYsh8OVa5OJOxdxiw/view?usp=drive_link)

📎 **Download (bootlog + system info)**: (https://drive.google.com/file/d/1CpReDxdF1QgpBUdRtHWOvH6_6qaBbTS7/view?usp=drive_link)

📎 **maybe some usefull infos from the firmware extracted with Binwalk (.bin)**: (https://drive.google.com/drive/folders/1AakD95KbanqsrVuBvXNEOGNFGBr8EBXa?usp=drive_link)

I'm new to OpenWrt development and collected this data with assistance from ChatGPT. While I may not be able to assist with patch creation, I'm ready to test and provide feedback as needed.

This device shares the IPQ807x platform with other devices already in development (e.g., TP-Link Deco X80, Xiaomi AX9000), so I believe support is feasible.

Thank you in advance!

musashino205, 2025-04-26:
This is a bug tracker, please post to the Forum.

https://forum.openwrt.org/

ege-art, 2025-04-26:
> This is a bug tracker, please post to the Forum.
> 
oh sorry do i need to close this issue? and i already did post there:  https://forum.openwrt.org/t/unifi-ubiquiti-ubnt-device-support-ipq8074-ipq807x-ipq8074a-add-support-for-ubiquiti-amplifi-alien-router/230912?u=a-somebody_8272

musashino205, 2025-04-26:
> do i need to close this issue?

Yes, please.

closed 2025-04-26 by ege-art