openwrt/openwrt #18608: 6.6: NAT66 breaks parallel glibc DNS lookups to IPv6 server

skbeh, 2025-04-26:
### Describe the bug

When a client device using glibc performs standard parallel IPv4 (A) and IPv6 (AAAA) UDP DNS lookups to an IPv6 DNS server, the lookups partially fail if there is an active NAT66 rule (either SNAT via Masquerading or a specific DNAT rule) on the OpenWrt router matching the DNS server's traffic.

Specifically, only the first DNS request packet (either A or AAAA) appears to be correctly processed by the NAT66 rule and receives a reply. The second concurrent request packet times out after the default glibc DNS timeout (5 seconds). After the timeout, glibc automatically re-attempts the failed lookup, which then typically succeeds.

This results in a ~5-second delay for any application relying on glibc's `getaddrinfo` for hostname resolution when an IPv6 DNS server is used and NAT66 is active.

### OpenWrt version

r28597-0425664679

### OpenWrt release

24.10.1

### OpenWrt target/subtarget

mediatek/filogic

### Device

H3C Magic NX30 Pro

### Image kind

Official downloaded image

### Steps to reproduce

1.  Configure the client device to use an **IPv6** address for DNS resolution in resolv.conf.
2.  Ensure the client device uses glibc for DNS resolution.
3.  Implement **one** of the following NAT66 rules on the OpenWrt router:
    *   **Scenario A (SNAT):** Enable IPv6 Masquerading on the wan interface.
    *   **Scenario B (DNAT):** Add a specific DNAT rule targeting an IPv6 address and redirecting traffic to another IPv6 address.
4.  From a client device connected through the router, perform a DNS lookup that triggers parallel A/AAAA requests, for example:
    ```bash
    ping www.google.com
    ```
    or
    ```bash
    getent ahosts www.google.com # `getent ahostsv4 www.google.com` and `getent ahostsv6 www.google.com` works correctly
    ```

### Actual behaviour

The command (`ping`, `getent`, etc.) hangs for approximately 5 seconds before resolving the hostname.

Packet captures and observations indicate:
*   Glibc sends two UDP packets concurrently (one for A, one for AAAA) to the IPv6 DNS server address.
*   Only the *first* packet seems to be correctly handled by the NAT66 rule:
    *   In Scenario A (Masquerading/SNAT): The destination DNS server receives only the first packet. The second packet might be sent with the original client source IP instead of the router's masqueraded IP, or dropped.
    *   In Scenario B (DNAT): The first packet is correctly redirected according to the DNAT rule. The second packet is *not* redirected and is sent to the original destination address specified in the packet, bypassing the DNAT rule.
*   The client receives a reply for the first request promptly.
*   The client receives no reply for the second request.
*   After ~5 seconds (glibc default timeout), glibc re-sends the request for the timed-out query.
*   This re-attempted query succeeds, and the application finally receives the resolved address.

### Expected behaviour

Both the parallel A and AAAA UDP DNS requests sent by the glibc client should be correctly and concurrently processed by the configured NAT66 rule (SNAT or DNAT) on the router.

### Additional info

*   The issue specifically affects the parallel UDP DNS lookup mechanism used by default in glibc.
*   The bug can be avoided by forcing glibc to not use parallel lookups or to use TCP. Adding `options use-vc` (force TCP) or `options single-request` / `options single-request-reopen` (force sequential A/AAAA lookups) to the client's `/etc/resolv.conf` prevents the bug from triggering.
*   Tools like `dig` work correctly (e.g., `dig A www.google.com; dig AAAA www.google.com`), presumably because they don't perform lookups in the same parallel manner as glibc.
*   The bug was **not** present in OpenWrt 23.05.5.
*   This issue might be related to the NAT66 regressions observed with **Linux kernel 6.6**, potentially similar to or caused by the same underlying issue as reported in [#15857](https://github.com/openwrt/openwrt/issues/15857).
*   Mentioning @nbd168 due to his significant work on OpenWrt's network stack implementation, particularly around recent kernel changes.

### Diffconfig

```text

```

### Terms

- [x] I am reporting an issue for OpenWrt, not an unsupported fork.

tmcqueen-materials, 2025-04-26:
@skbeh : I believe this is the same issue as in #18387 . Can you please disable rx-gro-list (see that bug for how with `ethtool`) and see if this problem goes away.

nbd168, 2025-04-26:
I just found that the IPv6 UDP codepath is missing handling of NAT66 for gro-list packets. Could you please try this kernel patch to see if it resolves the issue? https://nbd.name/p/ffc4c0d7

tmcqueen-materials, 2025-04-26:
@nbd168 : Thanks! I am currently building 24.10.1 with this patch to see if it fixes the issues in #18387 , which I believe are also this issue. While that is happening, are similar modifications needed for other IP protocols (beyond TCP where a patch is already landed, and UDP, which is the patch you just posted)? In other words, have all GRO offload paths for this kind of case been appropriately modified?

skbeh, 2025-04-26:
@tmcqueen-materials The issue was resolved after disabling `rx-gro-list` on the interface my client was connected to.

tmcqueen-materials, 2025-04-26:
@skbeh : yeah, I'm sure its the same underlying issue. With `rx-gro-list` on and @nbd168 's patch above, it should also work.

skbeh, 2025-04-26:
@tmcqueen-materials @nbd168 Yeah, the issue is also resolved after applying the patch.

closed 2025-04-26 by openwrt-bot