openwrt/openwrt #18615: [VLAN] Save button not working when adding a VLAN device as device to wireless configuration

codiflow, 2025-04-26:
### Describe the bug

Model: `GL.iNet GL-MT6000`
Architecture: `ARMv8 Processor rev 4`
Target Platform: `mediatek/filogic`
Firmware Version: `OpenWrt 24.10.1 r28597-0425664679 / LuCI (HEAD detached at 2ac26e56) branch 25.103.51521~2ac26e5`
Kernel Version: `6.6.86`

While tracking down this non-issue (https://github.com/openwrt/openwrt/issues/17131) I was setting up WIFI **on my new GL.iNet Router** and found a bug.

Any interface which has a VLAN device set (Type `VLAN (802.1q)`) as device **cannot be set as network from within the wireless configuration**.

The "Save" button **just doesn't work** – no error message and no browser console output – just no saving of the new settings possible (no appearing spinner). As soon as I select another interface which has **no VLAN device assigned** the "Save" button works just fine.

Here's a short video:

https://github.com/user-attachments/assets/669696f8-c741-4d1a-8206-1dbc650ba8fe

All interfaces:

[image]

The workaround:
- **Disable** the wireless network
- **Create a new interface within the network selection box** during the wireless configuration by entering a name and hitting Enter afterwards
- Hit "Save"
- Now go to `Network` => `Interfaces` and change the device for the newly setup interface to your VLAN device, set the firewall zone to `lan` (or whatever you want it to be) and hit "Save"
- "Save & Apply"
- Now reenable the WIFI network
- Voila – WIFI uses the VLAN device and works

I did **not** experience this problem with a TP-Link EAP615-Wall v1 and OpenWRT 24.10

### OpenWrt version

r28597-0425664679

### OpenWrt release

24.10.1

### OpenWrt target/subtarget

mediatek/filogic

### Device

GL.iNet GL-MT6000

### Image kind

Official downloaded image

### Diffconfig

```text

```

### Terms

- [x] I am reporting an issue for OpenWrt, not an unsupported fork.

shtefko, 2025-04-28:
You should only be able to add bridge devices to wireless configuration. Adding them to pure VLAN devices is unsupported, nor it make much sense. You should first create bridges for all of your VLANs, then assign those bridge devices to your interfaces, then add such interface to your wireless configuration.

codiflow, 2025-04-28:
OK seems to be a misunderstanding here. I never added the VLAN directly to the wireless network.

There always is a `bridge` (device) which has `hardware interfaces` as members:

[image]

Then I add a device type "VLAN (802.1q)" with the `bridge` as base device:

[image]

This device is then used as the device for the `interface`:

[image]

So to clarify the term "bridged VLAN": Is this a **valid** VLAN configuration which should work with the wireless configuration?

---

I filed a bug because the exact same behaviour DOES WORK on another OpenWRT device with 24.10. So no unresponsive "Save" button there.

If the input would be wrong there should be an **error message** or otherwise the bug should be fixed 😉

If I can help with more information just let me know.

HexKernel, 2025-04-29:
I was experiencing the same issue I reported earlier at https://github.com/openwrt/openwrt/issues/18586. In version 23.05.5, I was able to add VLAN interfaces directly to the Wi-Fi AP without any problems. However, in version 24.10.1, this is no longer possible because the "Save" button does not function.

Even in versions 24.10.0, 23.05.5, and 23.05.0, adding the VLAN interface directly to the Wi-Fi AP causes Wi-Fi clients to fail authentication — they get stuck during password verification. However, when no encryption (no password) is used, the VLAN works fine when added directly.

I also tried the suggested approach of creating a bridge with the intended VLAN interface and adding that bridge to the Wi-Fi AP. This allowed Wi-Fi clients to connect, but as soon as the first client connects, access to OpenWrt's IP address and LuCI interface is lost — it becomes unresponsive to pings and inaccessible via any interface.

HexKernel, 2025-04-29:
My device is a Raspberry Pi 4 Model B, which has only one LAN port and a single Wi-Fi chip. Therefore, I have no choice but to convert the single LAN port into multiple VLANs and connect it to a managed switch to connect all my devices and use its Wi-Fi. If anyone has suggestions or solutions to bypass these limitations/issues, that would be greatly appreciated.

shtefko, 2025-04-29:
You're assigning VLAN device to an interface, which is more or less naming of that device, and allows to configure IP on it. When you try adding that interface to the wireless configuration, you are basically trying to add another device to that interface, but only interfaces that are bridge devices can do that.
What you should do is create a new bridge device, let's call it br-mng. In the configuration of the bridge, add your VLAN 10 to bridged ports. Then create interface which uses br-mng as its device. Configure IP, etc. And finally use that interface in your wireless configuration.

codiflow, 2025-04-29:
Thanks a lot for clarifying this.

I wonder if this is reflected within the "official guides" as I never read of this kind of requirement for wireless connections until now. I totally get your explanation technically but most guides I can find on the internet don't mention this kind of "double assignment" in a VLAN environment. So this is somehow confusing 😅

Examples of guides saying the opposite:
https://reddit.com/r/openwrt/comments/1ebu234/vlan_setup_for_wifi/levrgx4/?context=3#levrgx4
https://forum.openwrt.org/t/wireless-access-point-and-vlans/73292/3

The first guide I found which describes what you were saying:
https://lowtek.ca/roo/2023/openwrt-vlans-for-guest-and-iot-networks/

But apart from knowing the valid way to create those VLAN interfaces now, there are still **two questions left**:

1. Why is there no error message in the GUI if the selected device in the wireless configuration is wrong? So the field should get a a red border like with other invalid fields instead of an inresponsive Save button, shouldn't it?
2. Why does the "wrong way" (assigning the VLAN device directly) work (since years)? For example like I mentioned earlier in OpenWRT 24.10 on a TP-Link EAP615-Wall v1?

HexKernel, 2025-04-29:
> You're assigning VLAN device to an interface, which is more or less naming of that device, and allows to configure IP on it. When you try adding that interface to the wireless configuration, you are [...]

Thank you very much for that, I tried this approach as well, however, even though at first it allowed Wi-Fi clients to connect, but as soon as the first client connects, access to the OpenWrt's IP address and LuCI interface is lost — it becomes unresponsive to pings and inaccessible via any interface.

HexKernel, 2025-04-29:
> Thanks a lot for clarifying this.
>
> I wonder if this is reflected within the "official guides" as I never read of this kind of requirement for wireless connections until now. I totally get your [...]

Regarding the direct method of assigning a VLAN interface to your Wi-Fi access point, may I ask if it was working successfully for you in previous versions, such as 24.10.0, without any connectivity issues? If not, could you please share your OpenWRT setup—perhaps a screenshot of the Interfaces and Bridge Devices sections—so I can figure out where I might have gone wrong? Even though this method worked for me in earlier versions, it caused all interfaces to lose their IP addresses, making it impossible to ping or access LuCI, which left me stuck for a while. I would really appreciate any help or solution you could provide.

mattymuc, 2025-04-30:
Same issue here on all my OpenWrt One devices, save button does not work in 24.10.1, nothing happens. With 24.10.0 I can save without issue.

shtefko, 2025-04-30:
> Thank you very much for that, I tried this approach as well, however, even though at first it allowed Wi-Fi clients to connect, but as soon as the first client connects, access to the OpenWrt's IP [...]

Make sure that a firewall zone that allows connections is assigned to your interface. Seems that OpenWrt will use default firewall rules for interfaces that are not in any zone. If this is a dumb AP you are configuring You may set your default rules to accept.

shtefko, 2025-04-30:
> Thanks a lot for clarifying this.
>
> I wonder if this is reflected within the "official guides" as I never read of this kind of requirement for wireless connections until now. I totally get your [...]

You are welcome. I agree with you on all points. There are almost no good guides on the web that correctly explain setting up VLANs. Many of them are even misleading in one way or another. There are many nuances that one needs to take into account when configuring VLANs. For example, if the device that you are configuring is a switch, then you need to use Network->Switch in Luci to configure individual ports of the switch. You would not want to create a bridge over that swich, and use Bridge VLAN filtering, because it would create trunk ports on all of switch' ports. On the other hand, if the device is individual NIC, (which is a port), you must create a bridge over that port and then use Bridge VLAN filtering on it.
When dealing with network stuff, always think in terms of what is it the thing you're dealing with. Is this a port type device or is it a bridge type device. So, for example, software VLANs you defined are port type devices. You can't add more things to a port, so if you want to make a grouping, define a bridge that will hold individual ports you want to group.

Regarding the point no. 1, I believe that the UI should only present interfaces that are bridges in wireless settings. Then there would be no need for error message and perhaps less confusion.
Regarding 2. I would not want to speculate. Is it a bug or perhaps a feature?

codiflow, 2025-04-30:
**100% ACK** with what you were saying. Thanks for the in detail explanation. 💪🙏

> For example, if the device that you are configuring is a switch, then you need to use Network->Switch in Luci to configure individual ports of the switch. You would not want to create a bridge over [...]

I think this is ESSENTIAL knowledge everyone should have before dealing with OpenWRT and VLANs. Thanks 😎

I see that this topic is also part of the (unfortunately not very comprehensive and intuitive) [official guide](https://openwrt.org/docs/guide-user/network/vlan/switch_configuration). 

> I believe that the UI should only present interfaces that are bridges in wireless settings. Then there would be no need for error message and perhaps less confusion.

Sounds like a really good solution. This would also prevent people from writing "misleading guides" because you cannot select values which are not being offered in LuCI 😉

Is there something we can do to help bringing this into production?

> I would not want to speculate. Is it a bug or perhaps a feature?

Maybe it's both 😅😁

HexKernel, 2025-05-01:
> Make sure that a firewall zone that allows connections is assigned to your interface. Seems that OpenWrt will use default firewall rules for interfaces that are not in any zone. If this is a dumb AP [...]

To do that, I created a LAN firewall zone and added all the interfaces I wanted to follow the same policies to it. Is that acceptable, or could it be causing the issue I’ve been experiencing? Should I create a separate firewall zone for each interface, even if the policies are the same?

shtefko, 2025-05-06:
> So, we're supposed to have the firewall enabled on "dumb" APs now? What is the point if there is no routing going on at all?

No one claimed that. Just be aware if you disable your firewall that setting may be lost after sysupgrade.

pabl0, 2025-05-06:
> No one claimed that. Just be aware if you disable your firewall that setting may be lost after sysupgrade.

Right. But on 24.10.1, it is not possible to add new wireless networks using LuCI in dumb-AP use case without a firewall (not an uncommon scenario). Of course, it works from the command line just fine, but I still spend quite some time figuring out why the "save" button in the web interface suddenly does not react in any way (unlike in 24.10.0 with identical configuration and hardware).

mattymuc, 2025-05-06:
> Right. But on 24.10.1, it is not possible to add new wireless networks using LuCI in dumb-AP use case without a firewall 
I tested a new Snapshot version and the save button is working again.... So I hope this problem is fixed in 24.10.2 :-)

MrEagle123, 2025-05-08:
For fix this problem, go to "System" - "Software" - click "Update lists". Then open "Updates" tab and click "Install" on ALL packages. Then reboot router.

mikec24, 2025-05-27:
I can confirm that MrEagles advise works

rvandegrift, 2025-07-07:
I ran into this bug.  I didn't try upgrading to a snapshot, but found a workaround for 24.10.1.

Create your interface with a dummy bridge - luci will let you associate a wireless SSID with it.  Then edit the interface to change it's device to the DSA dot1q subinterface.