openwrt/openwrt #18620: Regression related to switch driver in 24.10.1 realtek/rtl838x target mainly GS1900-24E (possibly others from that series)

AreYouLoco, 2025-04-27:
### Describe the bug

Continuing discussion from here: https://github.com/maurerr/openwrt/commit/ff4ec132c9b04697912f9cc3a9328411404aa00f#commitcomment-155667935

Seems like older hardware revision of GS1900-24E has problems with switch driver (possibly to GPIO pin changes).
I have two such a switches (different hardware revisions) and problem is related only to the older one:

Mentioning @svanheule as he asked for it.

And sending U-Boot logs from 24.10.0 (still working tag). May also dump info from v24.10.1 if needed as I already have the switch taken apart.

Here it goes:
```
RTL838x# printenv
baudrate=115200
boardmodel=ZyXEL_GS1900_24E
bootargs=console=ttyS0,115200 mem=64M quiet
bootcmd=cst fcTest; boota
bootdelay=0
ethact=rtl8380#0
ethaddr=B8:EC:A3:E5:05:10
ipaddr=192.168.1.1
netmask=255.255.255.0
serverip=192.168.1.111
stderr=serial
stdin=serial
stdout=serial
Environment size: 301/1020 bytes

RTL838x# printsys
resetdefault=0
dualfname0=openwrt-22.03.3-zyxel_gs1900-24e-initramfs-kernel.bin
mac_end=B8:EC:A3:E5:05:28
mac_start=B8:EC:A3:E5:05:10
sn=S173C17000180
bootpartition=0
dualfname1=runtime-GS1900-24Ev2-V2.70(AAHK.5).bix
System information size: 217/4096 bytes
```

Just after boot:
```
RTL838x# md.l 0xbb00a09c
bb00a09c: 00030002 00019000 0003e800 0007d000    ................
bb00a0ac: 00000000 00000000 00000000 00000000    ................
bb00a0bc: 00000000 00000000 00000000 00000000    ................
bb00a0cc: 00000000 00000000 00000000 00000000    ................
bb00a0dc: 00699041 00000181 00000000 00000000    .i.A............
bb00a0ec: 00000000 00000000 00000000 00000000    ................
bb00a0fc: 00000000 000480a8 00d71eec 00d71eec    ................
bb00a10c: 00d71eec 00d71eec 00d71eec 00d71eec    ................
bb00a11c: 00d71eec 00d71eec 00d71eec 00d71eec    ................
bb00a12c: 00d71eec 00d71eec 00d71eec 00d71eec    ................
bb00a13c: 00d71eec 00d71eec 00d71eec 00d71eec    ................
bb00a14c: 00d71eec 00d71eec 00d71eec 00d71eec    ................
bb00a15c: 00d71eec 00d71eec 00d71eec 00d71eec    ................
bb00a16c: 00d71eec 00d71eec 00d71eec 00ffffff    ................
bb00a17c: 00ffffff 00000000 00000000 00000000    ................
bb00a18c: 05000000 00000000 00220000 00000000    ........."......
```

Network on:
```
RTL838x# rtk network on
Enable network
Force port28 link up 1G
Please wait for PHY init-time ...

RTL838x# md.l 0xbb00a09c
bb00a09c: 00030002 00019000 0003e800 0007d000    ................
bb00a0ac: 00000000 00007000 00000fc7 00000000    ......p.........
bb00a0bc: 00000000 00000000 00000000 00007000    ..............p.
bb00a0cc: 00000fc7 00000000 00000000 00000000    ................
bb00a0dc: 00699041 00000181 00000000 00000000    .i.A............
bb00a0ec: 00000000 00000000 00000000 00000000    ................
bb00a0fc: 00000000 000480a8 00d71eec 00d71eec    ................
bb00a10c: 00d71eec 00d71eec 00d71eec 00d71eec    ................
bb00a11c: 00d71eec 00d71eec 00d71eec 00d71eec    ................
bb00a12c: 00d71eec 00d71eec 00d71eec 00d71eec    ................
bb00a13c: 00d71eec 00d71eec 00d71eec 00d71eec    ................
bb00a14c: 00d71eec 00d71eec 00d71eec 00d71eec    ................
bb00a15c: 00d71eec 00d71eec 00d71eec 00d71eec    ................
bb00a16c: 00d71eec 00d71eec 0006192f 00ffffff    .........../....
bb00a17c: 00ffffff 00000000 00000000 00000000    ................
bb00a18c: 05000000 20000000 0022a200 00000000    .... ...."......
```

After establishing link:
```
RTL838x# md.l 0xbb00a09c
bb00a09c: 00030002 00019000 0003e800 0007d000    ................
bb00a0ac: 00000000 00007000 00000fc7 00000000    ......p.........
bb00a0bc: 00000000 00000000 00000000 00007000    ..............p.
bb00a0cc: 00000fc7 00000000 00000000 00000000    ................
bb00a0dc: 00699041 00000181 00000000 00000000    .i.A............
bb00a0ec: 00000000 00000000 00000000 00000000    ................
bb00a0fc: 00000000 000480a8 00d71eec 00d71eec    ................
bb00a10c: 00d71eec 00d71eec 00d71eec 00d71eec    ................
bb00a11c: 00d71eec 00d71eec 00d71eec 00d71eec    ................
bb00a12c: 00d71eec 00d71eec 00d71eec 00d71eec    ................
bb00a13c: 00d71eec 00d71eec 00d71eec 00d71eec    ................
bb00a14c: 00d71eec 00d71eec 00d71eec 00d71eec    ................
bb00a15c: 00d71eec 00d71eec 00d71eec 00d71eec    ................
bb00a16c: 00d71eec 00d71eec 0006192f 00ffffff    .........../....
bb00a17c: 00ffffff 00000000 00000000 00d04000    ..............@.
bb00a18c: 05000000 20000000 0022a200 00000000    .... ...."......
```

```
RTL838x# version
U-Boot Version: 2.0.0.59413 (Jul 08 2015 - 10:00:04)
mips-linux-uclibc-gcc (GCC) 3.4.4 mipssde-6.03.00-20051020
GNU ld version 2.15.94 mipssde-6.03.00-20051020
```

### OpenWrt version

r28597-0425664679

### OpenWrt release

24.10.1

### OpenWrt target/subtarget

realtek/rtl838x

### Device

Zyxel GS1900-24E

### Image kind

Official downloaded image

### Steps to reproduce

Upgrade using owut mechanism from v24.10.0 to v.24.10.1

### Actual behaviour

Tagged ports seem isolated. The only access to switch is via UART or untagged port + static IP. All tagged ports RX packet count stays 0. TX seems to try to transmit data but they never leave port (checked with tcp-dump).

### Expected behaviour

Switch ports functioning of course.

### Additional info

Workaround is to stay on v24.10.0 tag for that older hardware revision of a switch.

### Diffconfig

```text

```

### Terms

- [x] I am reporting an issue for OpenWrt, not an unsupported fork.

AreYouLoco, 2025-04-27:
Also @svanheule I checked what you asked other user here:
https://github.com/openwrt/openwrt/issues/9534#issuecomment-2605452217

And I do get the same ...h buuu.... :D value (exact same output as @es86de so its SoC rev A for me as well.

svanheule, 2025-04-27:
> I do get the same ...h buuu.... :D value (exact same output as [@es86de](https://github.com/es86de))

Can you provide the full output of the `md.l` command (for both devices)? The ASCII-decoded output doesn't really say much and is identical for all SoC revisions.

AreYouLoco, 2025-04-27:
For the rev A it will take me a few minutes. But for the newer one I need to take it out of a rack unfortunately. But today luckily I have time for this. And thank you for interest in fixing that.

AreYouLoco, 2025-04-27:
```
md.l
```
Asks me for read address

AreYouLoco, 2025-04-27:
[image]
Newer board marking 37ZY-G724**D3** +412 V1.2 and model name

[image]
Older board marking 37ZY-G724**D0**+412 V1.2

AreYouLoco, 2025-04-27:
I am reading docs https://docs.u-boot.org/en/latest/usage/cmd/md.html

AreYouLoco, 2025-04-27:
Now I've connected via UART to the newer board and there is much newer U-Boot there if that makes a difference:
```
RTL838x# version

U-Boot Version: 2.0.0.1 (Aug 14 2020 - 18:20:13)
mips-linux-uclibc-gcc (GCC) 3.4.4 mipssde-6.03.00-20051020
GNU ld version 2.15.94 mipssde-6.03.00-20051020
```

AreYouLoco, 2025-04-27:
Ok i got something:
```
RTL838x# bdinfo                    
[...]
memstart    = 0x80000000
memsize     = 0x08000000
[...]
```
@svanheule So seems like ```md.l 0x80000000 0x02000000``` in 32-bit words (/4) would be the whole memory. Do you need that all?

svanheule, 2025-04-27:
No, just the output of 
```
RTL838x# mw.l bb000058 3
RTL838x# mw.l bb0000d8 a0000000
RTL838x# md.l bb0000d0 4
```

AreYouLoco, 2025-04-27:
Here is from the working (newer-one):
```
RTL838x# mw.l bb000058 3     
RTL838x# mw.l bb0000d8 a0000000
RTL838x# md.l bb0000d0 4
bb0000d0: 00000003 83826800 a0036275 00000017    ......h...bu....
```

Will edit in <2min when I get to the other.... Hang on

svanheule, 2025-04-27:
Then both devices have the same SoC revision, so that can't be the issue here. Assuming you mean you want to run these commands after installing 24.10.1, that won't make a difference. These values are IDs baked into the chip.

svanheule, 2025-04-27:
Do you get functional networking on the older device if you run `rtk network on` in u-boot before booting (with `boota`) 24.10.1?

AreYouLoco, 2025-04-27:
Damn I just reverted to 24.10.0 to fix my network... Going back on spot doing owut upgrade to 24.10.1 and will try networking in u-boot. Also the uplink is tagged port. Will that make a difference? Is U-Boot networking VLAN aware?

Also in non working state (v24.10.1) if I set IP from the range statically on laptop and connect to untagged port I get access to the older switch. Like so I downgraded.. The other three ports that I use on that switch to connect upstairs with downstairs and one more lets say are all VLAN tagged.

AreYouLoco, 2025-04-27:
Ok so the answer won't be simple yes/no unfortunately.

Older hardware revision switch OK 24.10.0 -> then 24.10.1 flashed via owut -> it rebooted. All seemed to work after it came up. Connectivity ok. Same thing when I checked initially the release. Then I did normal reboot from console and after reboot no connectivity.

Then I rebooted to U-boot. Did ```rtk networking on``` Connected a network cable and it worked for a while as I got IP from the router when in U-Boot so there must have been IP connectivity. Let the ping ran to the router and after some time I got Destination Host Unreachable. Then I did ```boota``` it booted into v24.10.1 but still no network.

Another reboot. Still no network.

**So there must be some state somewhere lost with reboot**. It's the same weirdness. I discovered the problem initially just after power went down when the device rebooted itself must have been. Because I did upgrade to v.24.10.1 back then I checked all fine like just now. It came up. So no problem it seemed.

AreYouLoco, 2025-04-27:
I was looking at the logs via UART. And there was no difference when it worked and not. Ports going into promiscuous then disabled, learning, propagating. So its not STP to blame here for sure. And reverting 24.10.0 fixes everything. And after each reboot it works each time. I've checked that many times already.

AreYouLoco, 2025-04-27:
Idea I have for now is to dump /dev/mtd0 from newer one which is U-Boot and flash it to older. So we are for sure on the same U-Boot version.

Or maybe we should stick to that:
https://github.com/openwrt/openwrt/compare/v24.10.0...v24.10.1
One of commits introduced here is breaking one.

svanheule, 2025-04-27:
You could try using the newer bootloader, but that wouldn't fix the problem for other users with an older device.

Assuming the issue is indeed caused by GPIO driver change (60cae0e08fd2b4809b8189b171397ae6b6360993 "realtek: rtl838x: Switch GS1900 rtl8231 driver"), you could try compiling an overview of the state of the RTL8231 pins:
on 24.10.0 and 24.10.1, both on the new and old device.

Alternatively, if you don't mind flashing a few builds (or running initramfs images, if you can reproduce it there), you could bisect the commits between these two releases. You'll need about 7 or 8 builds to pinpoint the commit that breaks networking. There are some kernel updates in there too, so I hope you have a good build machine in that case :-)

AreYouLoco, 2025-04-27:
> but that wouldn't fix the problem for other users with an older device.

Exactly.

Ok how do I dump running GPIO pins states then? I will indeed build a few images. To bad my servers' fan broke down and its off for now. I would have some workhorse like that. But have to build on laptop. But even kernel alone takes like 30-45mins there.....

Anyway. For today it's enough debugging. Will need to get more familiar with build system. As I built only single packages so far and kernel.

AreYouLoco, 2025-04-27:
I will use some other fan and run the server machine. Plan is to start from the commit before https://github.com/openwrt/openwrt/commit/60cae0e08fd2b4809b8189b171397ae6b6360993 and narrow down breaking change. And then check if upgrading U-Boot will do any better.

I have dumps of 2015-one and newer. So may be restored for further debugging if needed.

Thank you for your time so far @svanheule

plappermaul, 2025-06-17:
As Sander is otherwise busy these days maybe I can help here. To sum the discussion up: You have a working system (VLAN) with v24.10.0 and issues with v24.10.1. 

The only solution to nail it down is to bisect this. If you do not now how to do it you should create an image from v24.10 branch but only upto (and including) commit [realtek: drop old RTL8231 driver](https://git.openwrt.org/?p=openwrt/openwrt.git;a=commit;h=0ae0766f29e537484748ce6681d78d33ecea8ec6). Why this:

- the realtek mdio driver is included
- a lot of kernel bumps from v24.10.1 are left out.

AreYouLoco, 2025-07-05:
So for now its one of this changes that break stuff:
https://github.com/openwrt/openwrt/compare/6df0e3d02a...0ae0766f29e537484748ce6681d78d33ecea8ec6

plappermaul, 2025-07-05:
Thanks for taking time on this. To nail it down to the new driver I would expect that two more builds should give the answer.

- One with the last commit before it changed in your DTS
- One with the commit that activates it in your DTS

AreYouLoco, 2025-07-20:
So I've build three images to test out (sorted in commit order):
```
openwrt-24.10.0-c62bafb223b5bb861ce44c9436c348eae5109cf4-realtek-rtl838x-zyxel_gs1900-24e-squashfs-sysupgrade.bin
openwrt-24.10.0-8e539afa957d849401997c5c6decb249f8c7aec8-realtek-rtl838x-zyxel_gs1900-24e-squashfs-sysupgrade.bin
openwrt-24.10.0-f5b66f765163dfd5ef066f6e93700908916084f1-realtek-rtl838x-zyxel_gs1900-24e-squashfs-sysupgrade.bin
```

And its already broken the same on first one:
https://github.com/openwrt/openwrt/commit/c62bafb223b5bb861ce44c9436c348eae5109cf4

Which was unexpected. Will build one commit before

AreYouLoco, 2025-07-20:
I will dedicate time today. To finish git bisecting this. Anyway that I started

AreYouLoco, 2025-07-20:
Now I wonder. I am using https://downloads.openwrt.org/releases/24.10.0/targets/realtek/rtl838x/config.buildinfo as base for .config file and just change version name and add owut and ca-certificates. Is it even good test (config file)?

plappermaul, 2025-07-20:
I dont't know exactly how config files work. I'm only working with common commands for local build environment. That boils down to

```
git checkout -b new-branch-name commit-hash
make clean
make -j9 world V=s
```

AreYouLoco, 2025-07-20:
My workflow for tests was:
```
make clean
git checkout COMMITHASH
feed update & feed install
wget https://downloads.openwrt.org/releases/24.10.0/targets/realtek/rtl838x/config.buildinfo -O .config
make menuconfig (add version string and owut & ca-certificates and choose only profile for 24E model)
make -j16 Vs
```

plappermaul, 2025-07-20:
But I do not know what the wget really changes. Because .config is auto generated.

AreYouLoco, 2025-07-31:
Well. I am umable to verify it differently than build newer 24.10.1+ with GPIO reset line commented out

SimonFischer04, 2025-08-03:
I also have an older 1900-24E with physical power switch.
After a long time questioning if I did just configure something wrong, I found this.
After testing around a bit more, I found that passing tagged traffic works fine for me (VLAN 50 tagged in at port 24 and out tagged at port 2 to my laptop).
Just skimmed through - haven't read every message 100% - have you tried this as well?

The only thing not working seems to be accessing the switch itself tagged. 
(The switch is connected through lan24 to my other switch / network and static ip configured on switch.50)
If I now configure the vlan filtering on the bridge:
lan24 TAGGED 50 -> does not work (can't access the switch - but passes traffic to other ports correctly)
lan24 UNTAGGED +PVID 50 (same on my other switch of course) -> works (can access the switch)

@AreYouLoco what's your status. Is there anything I can help with?

As a first step, I attached the kernel log, if that helps.
[dmesg.log](https://github.com/user-attachments/files/21567888/dmesg.log)

plappermaul, 2025-08-05:
Made a little test config to verify this VLAN management. 

```
config bridge-vlan 'lan_vlan'
        option device 'switch'
        option vlan '1'
        list ports 'lan9'
        list ports 'lan10'
        list ports 'lan11'
        list ports 'lan12'
        list ports 'lan13'
        list ports 'lan14'
        list ports 'lan15'
        list ports 'lan16'
        list ports 'lan17'
        list ports 'lan18'
        list ports 'lan19'
        list ports 'lan20'
        list ports 'lan21'
        list ports 'lan22'
        list ports 'lan23'
        list ports 'lan24'
        list ports 'lan25'
        list ports 'lan26'
        list ports 'lan27'
        list ports 'lan28'

config device
        option name 'switch.1'
        option macaddr '00:ad:24:3c:92:a0'

config interface 'lan'
        option device 'switch.1'
        option proto 'static'
        list ipaddr '192.168.2.71/24'
        option ip6assign '60'

config bridge-vlan
        option device 'switch'
        option vlan '22'
        list ports 'eth0:t'
        list ports 'lan1:t'
        list ports 'lan2:t'
        list ports 'lan3:t'
        list ports 'lan4:t'
        list ports 'lan5:t'
        list ports 'lan6:t'
        list ports 'lan7:t'
        list ports 'lan8:t'

config interface 'xan'
        option proto 'static'
        option device 'switch.22'
        option ipaddr '10.0.0.1'
        option netmask '255.255.255.0'
```

According to your description you would not be able to mange the switch through lets say lan1?

AreYouLoco, 2025-08-05:
Correct. I have tagged only 22/23/24 ports and switch seems isolated from the point of view of devices connected to tagged ports. All the rest ports are untagged and when I set static IP (because DHCP is not passed from tagged port) the I am able to reach that switch using any of untagged ports.

plappermaul, 2025-08-05:
Maybe stupid question. Do you have the VLAN local flag enabled in WebUI? Or looking at /etc/config/network. Have deleted the local=0 setting?

```
config bridge-vlan
        option device 'switch'
        option vlan '23'
        option local '0' <- Only switch between real ports.

```

plappermaul, 2025-08-05:
Just collecting some wild ideas. Can you give me the following memory data from the running NON FUNCTIONAL system.

```
root@OpenWrt:~# devmem 0x1b003a7c
0x55555555
root@OpenWrt:~# devmem 0x1b003a80
0x00555555
root@OpenWrt:~# devmem 0x1b003a84
0x1FFFFFFF
root@OpenWrt:~# devmem 0x1b00a530
0x00000001
root@OpenWrt:~# devmem 0x1b00a534
0x00000001
root@OpenWrt:~# devmem 0x1b00a5a0
0x00000001
```

plappermaul, 2025-08-05:
/dev/mem support in kernel needed for devmem.

SimonFischer04, 2025-08-05:
> /dev/mem support in kernel needed for devmem.

Tried to buld an image by enabling:
Global build settings - Kernel build options - /dev/mem virtual device support

then (after some other issues) I'm stuck with an to be image:
```
/openwrt/build_dir/target-mips_24kc_musl/linux-realtek_rtl838x/tmp/openwrt-realtek-rtl838x-zyxel_gs1900-24e-initramfs-kernel.bin is too big: 7629678 > 7143424
```

How to make it smaller.? already tried to remove luci

SimonFischer04, 2025-08-07:
> Just \"stock\" image with standard packages to test out.

Same here. But apparently with the debug packages its already to much.

>  If you manage faster to check /dev/mem values asked then it would be nice.

Probably won't hurt if we check both, just to be sure. But will build pr soon.

SimonFischer04, 2025-08-07:
> Just collecting some wild ideas. Can you give me the following memory data from the running NON FUNCTIONAL system.
>
> ```
> root@OpenWrt:~# devmem 0x1b003a7c
> 0x55555555
> root@OpenWrt:~# devmem 0x1b003a80 [...]

Did build an image now with the pr:
```
root@OpenWrt:~# devmem 0x1b003a7c
0x55555555
root@OpenWrt:~# devmem 0x1b003a80
0x00005555
root@OpenWrt:~# devmem 0x1b003a84
0x1FFFFFFF
root@OpenWrt:~# devmem 0x1b00a530
0x00000001
root@OpenWrt:~# devmem 0x1b00a534
0x00000001
root@OpenWrt:~# devmem 0x1b00a5a0
0x00000001
```

The first two commands responded with 0x00000000
but after some time got the values above.

Full serial log attached

[lognew.txt](https://github.com/user-attachments/files/21672348/lognew.txt)

plappermaul, 2025-08-07:
Ok. so no problem here. The next thing we should test is, if we have issues with incoming or outgoing packets. So first check if the CPU port receives a packet from the outside world. Please add this to the rtl83xx ethernet driver:

```
if (dsa) {
	priv->r->decode_tag(h, &tag);
	skb->data[len - 4] = 0x80;
	skb->data[len - 3] = tag.port;
	pr_info("%s received packet from port %d\n", __func__, tag.port); <---- ADD
	skb->data[len - 2] = 0x10;
	skb->data[len - 1] = 0x00;
	if (tag.l2_offloaded)
		skb->data[len - 3] |= 0x40;
}
```
Before pinging/accessing the switch wait for the the DSA ports to come up. If we see messages here coming from the non functioning VLAN port we can assume, that the inbound VLAN config works.

SimonFischer04, 2025-08-08:
> Made a little test config to verify this VLAN management.
>
> ```
> config bridge-vlan 'lan_vlan'
>         option device 'switch'
>         option vlan '1'
>         list ports 'lan9'
>         list ports [...]

Did a test with the current openwrt main and this config. Got it working once (able to access the switch via - directly attached' but tagged vlan22 -)

Could the rt-loader have fixed something here? Or were there other related prs merged in the last few days?

After testing a bit more and trying to apply me usual config the lan1 interface now does not want to come up again after network reload.

Just to be sure also rebootet the switch a few times. ftpboot the same kernel image. initially lan1 switch light lights up. But after i enter this exact same config into /etc/config/network
And then reload with
/etc/init.d/network restart

dmesg just logs:
```
[  141.032173] in rtl838x_eth_stop
[  141.036858] rtl838x-eth 1b00a300.ethernet eth0: Link is Down
[  141.546769] rtl83xx_fib_event: FIB_RULE ADD/DEL for IPv6 not supported
[  141.554331] rtl83xx_fib_event: FIB_RULE ADD/DEL for IPv6 not supported
[  141.561740] rtl83xx_fib_event: FIB_RULE ADD/DEL for IPv6 not supported
[  141.569257] rtl83xx_fib_event: FIB_RULE ADD/DEL for IPv6 not supported
[  143.076694] rtl83xx_fib4_del: no such gateway: 0.0.0.0
[  143.082536] rtl83xx_fib4_del: no such gateway: 0.0.0.0
[  143.091636] rtl83xx_fib_event: FIB_RULE ADD/DEL for IPv6 not supported
[  143.144117] rtl83xx_fib4_del: no such gateway: 0.0.0.0
```

but then does not come up anymore after some time - like it would on initial boot.
(nothing after 10 minutes)

@AreYouLoco please also test once more with new main and see what results you get.
I'm kinda confused now.
Will need to do more testing myself also.

plappermaul, 2025-08-09:
Network reconfiguration at least from command line is tricky sometimes. Always stop network before you modify the configuration.

AreYouLoco, 2025-08-09:
@SimonFischer04 please do a reboot to confirm. Because for me first time I discovered the problem also it worked randomly after upgrade to 24.10.1+ but then I got power loss and after switch rebooted that same symptoms.

SimonFischer04, 2025-08-09:
@AreYouLoco did not reboot because was always testing in tftpboot initramfs.

Did now also flash the sysupgrade to be able to make persistent changes.

Tried for hours but can't get the switch to establish a link anymore using the test config.
link led off and ip link also shows down.

Reboot, re-plugging power, ... already tried dozen of times.
I think I somehow bricked the switch.

SimonFischer04, 2025-08-09:
> Try to revert to 24.10.0 this is the last working release

Already tried reverting to a snapshot that got at least that far.
But will also try revert back to that release.

plappermaul, 2025-08-09:
The RTL838x is the best explored target. Looking back this thread we still have no clear commit that is the issue. I do not even know what image is currently runnning and why we suddenly have no network at all. Slowly we are heading towards everywhere but not in the right direction. If it is not possible to bisect this then poking around 8000 registers and writing arbitrary debug messages is useless.

SimonFischer04, 2025-08-09:
Sorry, I should have explained in more clearly:
To add your debug messages I used yesterdays latest main (3aee42001f493952b72a2ef3cd458bf9417c8940).
And added the pr_info line.

After network not working at all anymore I then tried switching back to just before the rt-loader (ebee9462273d3d30dbd29434e4f715ce39f80efc).
I know it should change anything at runtime but it was working before there.
(with check-size removed so I can leave devmem in there)

But even does not work now. I don't know how, but that why I think I might have hard bricked something.

SimonFischer04, 2025-08-09:
> > But even [ebee946](https://github.com/openwrt/openwrt/commit/ebee9462273d3d30dbd29434e4f715ce39f80efc) does not work now. I don't know how, but that why I think I might have hard bricked something.
> 
> This was commited yesterday. Try 24.10.0 tag

Yes, I know. But it worked before!
I'm now trying the 24.10.0 tag.

Please try to add the pr_info to your build to help out plap while I see whats wrong with my switch.

AreYouLoco, 2025-08-10:
Will see because today I spent time upgrading my build server from Debian 12 to 13.

And in general openwrt build system builds toolchain for you that is compatible. Scroll up you will se my example work flow when building.

Sorry I could not build images duting this weekend as life happened.

p7996619, 2025-08-14:
Ran into the same issue today on a GS1900-24E when trying to upgrade to 24.10.2. Had to restore from TFTP.
Only version 24.10.**0** and current SNAPSHOT works. I tried 24.10.**1**, 24.10.**2** and 24.10-SNAPSHOT.
I have the older hardware version, same board revision as OP.

Upgrade itself worked and bootlog, config etc. all normal and as expected, but no RX packets on any interface.
Can't confirm that it's only with VLANs. I could only get a connection in U-Boot. I tried with failsafe, default initramfs image, tagged and untagged ports. Also not a single packet passed through from other devices.

Please let me know if I can help, but I'm also somewhat limited with testing as I still need the device for connectivity.

SimonFischer04, 2025-08-15:
> > Ok. so no problem here. The next thing we should test is, if we have issues with incoming or outgoing packets. So first check if the CPU port receives a packet from the outside world. Please add this to the rtl83xx ethernet driver:
> > ```
> > if (dsa) {
> > 	priv->r->decode_tag(h, &tag);
> > 	skb->data[len - 4] = 0x80;
> > 	skb->data[len - 3] = tag.port;
> > 	pr_info("%s received packet from port %d\n", __func__, tag.port); <---- ADD
> > 	skb->data[len - 2] = 0x10;
> > 	skb->data[len - 1] = 0x00;
> > 	if (tag.l2_offloaded)
> > 		skb->data[len - 3] |= 0x40;
> > }
> > ```
> > 
> > 
> >     
> >       
> >     
> > 
> >       
> >     
> > 
> >     
> >   
> > Before pinging/accessing the switch wait for the the DSA ports to come up. If we see messages here coming from the non functioning VLAN port we can assume, that the inbound VLAN config works.
> 
> [@plappermaul](https://github.com/plappermaul) in which file it goes? Tomorrow I will have time to build with that and test out

Just search for the line before:
skb->data[len - 3] = tag.port;

you will find it very easily:
target/linux/realtek/files-6.6/drivers/net/ethernet/rtl838x_eth.c

SimonFischer04, 2025-08-15:
> > But even [ebee946](https://github.com/openwrt/openwrt/commit/ebee9462273d3d30dbd29434e4f715ce39f80efc) does not work now. I don't know how, but that why I think I might have hard bricked something.
> 
> This was commited yesterday. Try 24.10.0 tag instead https://downloads.openwrt.org/releases/24.10.0/targets/realtek/rtl838x/openwrt-24.10.0-realtek-rtl838x-zyxel_gs1900-24e-squashfs-sysupgrade.bin

Messing around with my build way too long again. Even after adding gnu17, ... I just got more and more errors ...
After hours I went and just downloaded the 24.10.0 release, just to test if i could get networking there.

Even with 24.10. the network only worked with openwrt's initial default config with port1 192.168.1.1 UNTAGGED.
After applying plappermaul's test config above with vlans, I was back at no link at all.

Kind of randomly I tried reverting back to stock, testing there and then switching back to openwrt.
And woala, tagged access now works again with 24.10.

@AreYouLoco Not sure, maybe this is also related to the gpio reset issues you mentioned above. Maybe you managed yourself into a state that did not get reset while bisecting this.
Like something was really broken here with my switch, that stock firmware fixed by re-initialising everything properly.
Again, not sure. Just some guess I wanted to share. Maybe it could help figure out the regression.

Will probably need to setup a debian 12 buildserver myself, as building should work there according to your setup? 
Then I can help with bisecting, ...

plappermaul, 2025-08-16:
I don't know what to say to such feedback. But everybody brings themselves along ... To look back and be convinced that I have done my best to assist you, I built several extra images for GS1900-24E in between 24.10.0 and 24.10.1 for you. You can download them at https://github.com/plappermaul/gs1900-24e

They contain devmem but no other features. The numbers indicate the first digits of the last commit they are based on. E.g. xxx-e5ac0c is upto and including https://git.openwrt.org/?p=openwrt/openwrt.git;a=commit;h=e5ac0c

Do me a favour. Organize yourselves and find a common way to get reproducible test results. For now my only hope that is left, is that current main snapshots magically work again. Afterwards feel free to come back. 

Good luck.

plappermaul, 2025-08-16:
@p7996619 thanks for testing. Nevertheless I must assume that this is not the right way to test. The result sums up as:

- It works until half way between 24.10.0 and 24.10.1
- It breaks for some commits in between
- Afterwards it works again somewhere before 24.10.1

That does not fit to all previous reportings. As you did not post if you powered off the switch or just rebootet and obviously did no installation this must be repeated.

I guess @AreYouLoco can best describe the compete end-to-end test sequence.

SimonFischer04, 2025-10-02:
So coming back to this, 
after I got some time to dig deeper and do a (hopefully) proper test:

After 3 days of testing, I'm now confident about https://github.com/openwrt/openwrt/commit/b2d17dbb68c232393739e6fb48245f1f4bebb698 being the culprit.
Previous tests were close, but had a few fatal flaws. Here's why:

---

(1. I started by setting up myself a debian 11 environment to have old enough packages, to be able to build older versions and properly bisect this on main.)

So I did a:
```
git bisect start 263721dd0ae14587703f328500c39b887774e6b3 0ae0766f29e537484748ce6681d78d33ecea8ec6
```

with the following config:
```
config bridge-vlan 'lan_vlan'
        option device 'switch'
        option vlan '1'
        list ports 'lan10'
        list ports 'lan11'
        list ports 'lan12'
        list ports 'lan13'
        list ports 'lan14'
        list ports 'lan15'
        list ports 'lan16'
        list ports 'lan17'
        list ports 'lan18'
        list ports 'lan19'
        list ports 'lan20'
        list ports 'lan21'
        list ports 'lan22'
        list ports 'lan23'
        list ports 'lan24'
        list ports 'lan25'
        list ports 'lan26'
        list ports 'lan27'
        list ports 'lan28'

config device
        option name 'switch.1'
        option macaddr '00:ad:24:3c:92:a0'

config interface 'lan'
        option device 'switch.1'
        option proto 'static'
        list ipaddr '192.168.2.71/24'
        option ip6assign '60'

config bridge-vlan
        option device 'switch'
        option vlan '22'
        list ports 'eth0:t'
        list ports 'lan1:t'
        list ports 'lan2:t'
        list ports 'lan3:t'
        list ports 'lan4:t'
        list ports 'lan5:t'
        list ports 'lan6:t'
        list ports 'lan7:t'
        list ports 'lan8:t'
        list ports 'lan9:t'

config interface 'xan'
        option proto 'static'
        option device 'switch.22'
        option ipaddr '10.0.0.1'
        option netmask '255.255.255.0'
```
[... 103 lines cut ...]

---

To make sure we found the preproduceable cause, please @AreYouLoco @p7996619 also test the following:

- e9fad02c101d7d7f45fd6e5eef8a80b848718a44 should work
- b2d17dbb68c232393739e6fb48245f1f4bebb698 is broken.

connect to lan1!
Then just tftpboot these two images and try to ping the switch from your device should be enough (ping 192.168.1.1)

Then build a current image (i used same base as my bisect - 263721dd0ae14587703f328500c39b887774e6b3) with the following lines commented out:

https://github.com/openwrt/openwrt/blob/263721dd0ae14587703f328500c39b887774e6b3/target/linux/realtek/dts/rtl8380_zyxel_gs1900_gpio.dtsi#L8-L10

this should also works. 
(At least for me ping 192.168.1.1. and after config 10.0.0.1 works)

@AreYouLoco please also test again on your newer 24e. So we are sure the regression is only for the old revision and not actually on both, but testing was flawed.

---

Summary, again. Because all of these I'm confident that the gpio reset is at fault.

If there is any other details / testing I can help with, please ask, I'm glad to provide.

plappermaul, 2025-10-03:
Thanks for taking the time to nail it down. 

You own the devices. So you are the ones who decide what is best for them. If the activation of the GPIO pin is really the issue, then send a PR to deactivate that.

SimonFischer04, 2025-10-03:
That's only half of the questiion ...

The reset is already in the common rtl8380_zyxel_gs1900_gpio.dtsi used by more than 1900-24e.

SimonFischer04, 2025-10-03:
> With freedom I mean that you can create files/dependencies as you like. For this would be
> 
> * take everything from the include besides the reset and copy it into ...gpio_noreset.dtsi
> * remove everthing from inlude exept reset and include ...gpio_noreset.dtsi

👍, thanks for the info. But lets wait first until @AreYouLoco can confirm, so we are sure.

SimonFischer04, 2025-10-03:
@AreYouLoco 
I don't know how good you are hardware wise, but as you have both switches, it would be easier to compare.
To confirm if the GPIO pin may have changed, it would be nice if you could take both apart and check (with multimeter) if the GPIO pin actually connects to where it's supposed to be (RESET of rtl8231) / another GPIO connects there on the older revision.

Also, please check (in stock) firmware if newer ones actually identify as B1.
After again reading your initial post (both being labelled as V1.2 and only minor D0 vs V3 difference) I unfortunately believe both are shown as A1.
If so, I'm not sure how to best label them, not to cause confusion for the users.

SimonFischer04, 2025-10-03:
> Lemmie check continuity with multimeter tomorrow. Will grab datasheet real quick which pin number it is. I will just check the broken one so older revision. As there is no problem on newer. So I guess its right
> 
> Edit: rtl8382M pin 114 is #RESET 
> Exact model taken from Svan's wiki https://svanheule.net/switches/models
> 
> Correct labelling later on. Depending on if the fix is different GPIO for each models or some label like (older/no reset pin/physical power switch).

114 is rest of the rtl8382M.

What we need is gpio1 of the rtl8382 (so pin 112) which connects to Reset OF the rtl8231. (GPIO extender).
Is what i think should be what we need, after reading the commit that originally introduced the rest / svanheule wiki.

AreYouLoco, 2025-10-04:
I am sorry peeps but today I cannot find time to take it apart and check. But now that its almost confirmed I will try to prioritize it in the upcomming days.

[5 more replies]

closed 2026-02-25 by svanheule