openwrt/openwrt #18627: TP-Link Archer MR200 v6 not working access point on wifi 5 GHz

toranon47, 2025-04-28:
### Describe the bug

On TP-Link Archer MR200 v6 i flashed firmware from v5 - https://downloads.openwrt.org/releases/24.10.0/targets/ramips/mt76x8/openwrt-24.10.0-ramips-mt76x8-tplink_archer-mr200-v5-squashfs-tftp-recovery.bin - no radio wifi 5 GHz

On v6 wifi 5 GHz on MT7613BEN chip - i tried wifi driver kmod-mt7615e - MediaTek MT7613BE 802.11ac/n radio1 appeared, but in the disabled state regardless of the settings

### OpenWrt version

r28427-6df0e3d02a

### OpenWrt release

24.10.0

### OpenWrt target/subtarget

ramips/mt76x8

### Device

TP-Link Archer MR200 v6

### Image kind

Official downloaded image

### Additional info

[image]

### Diffconfig

```text

```

### Terms

- [x] I am reporting an issue for OpenWrt, not an unsupported fork.

brada4, 2025-04-28:
Please ask in forum.

Openwrtfunboy, 2025-04-28:
24 is broken. Use 23.

DragonBluep, 2025-04-29:
You need to install `kmod-mt7615e` and `kmod-mt7663-firmware-ap`.

brada4, 2025-04-29:
Try to compile your findings adding V6 profile next to existing V5 ;-)

brada4, 2025-11-15:
You mean diff€rent manufacture week? Any problems open a new issue.