openwrt/openwrt #18651: unable to create issue

AnoBene, 2025-04-30:
unable to create issue
i want to submit bug report but i get message "unable to create issue"

AnoBene, 2025-05-01:
[image]
Nope still cant send the bug report i wrote up.. its waiting on another tab still..

stklcode, 2025-05-03:
According to your screenshot, you did not accept the terms, i.e. the mandatory checkbox "I am reporting an issue for OpenWrt, not an unsupported fork."

AnoBene, 2025-05-03:
I did, the unchecked box is "Create more"
If anything is unchecked the error is according to waht is unchecked.. "unable to create issue" error is a fkn mystery tho.
Im using "Dark Reader" to give all sites Dark Mode.. i guess thats why it is "white" as checked.