openwrt/openwrt #18662: 24.10/24.10.1 LuCI: scan and select process removes radio0 and its network

leanne63, 2025-05-01:
### Describe the bug

Using LuCI, in both 24.10 and 24.10.1, when I scan and select a network, the radio0 (connecting to the scanned network) disappears. After Save & Apply, the radio0 is still gone.

This occurs on two different travel routers: a TP-Link WR902AC(US) V3 and a GL.iNet MT3000 (Beryl AX).

### OpenWrt version

r28597-0425664679

### OpenWrt release

24.10.1

### OpenWrt target/subtarget

TP-Link WR902AC: ramips/mt76x8 -- GL.iNet GL-MT3000: mediatek/filogic

### Device

TP-Link TL-WR902AC V3 -- GL.iNet GL-MT3000 (Beryl AX)

### Image kind

Official downloaded image

### Steps to reproduce

* Go to Network -> Wireless
* (radio0 and radio1 should both be active/enabled)

[image]

* On radio0, click Scan
* Select a network and click Join Network

[image]

* Either check 'Replace wireless configuration' or don't check it; doesn't matter
* Enter the network's passphrase, if it has one

[image]

Note: I repeated this with a fresh install, so the firewall zone 'wan' was default, without 'wireguard0', as shown in the following from the GL.iNet MT3000:

[image]

* Click Submit
* On next page, under Device Configuration, at Network, click dropdown arrow, and check to add 'lan'

[image]

* Click Save
* Upon returning to the radio page: radio0 and the connection beneath it are both gone; radio 1 and its connection are still present. Clicking Save & Apply does not restore the two items.

WR902AC:
[image]

GL.iNet:
[image]

### Actual behaviour

In LuCI, at Networking -> Wireless, after selecting and configuring a network to join to radio0, saving causes the radio0 and its newly associated network to disappear.

### Expected behaviour

At Networking -> Wireless, with both radios (radio0 and radio1) enabled. Radio1 is actively providing a 5 GHz connection to clients (in Master mode). Radio0 is enabled, and may or may not be actively connected to a WISP hotspot. I 'Scan' for an active network, then select to join a valid network appearing in that scan. Upon saving and configuring the network, then returning to the Networking - > Wireless page, both radios are present with their respective networks underneath them. I then can Save & Apply, and the router will connect to the WISP I selected and configured. Both radios and their networks will show their associated activity.

### Additional info

The list of changes that would be submitted upon clicking Save & Apply appear to be deleting the radio and its associated network.

[image]

### Diffconfig

```text

```

### Terms

- [x] I am reporting an issue for OpenWrt, not an unsupported fork.

leanne63, 2025-05-01:
Yes, this is from two supported devices. Wasn't sure how to note that. 

The target for the TP-Link WR902AC is 'ramips/mt76x8'

The target for the GL.iNet GL-MT3000 (Beeyl AX) is 'mediatek/filogic'

brada4, 2025-05-01:
Start with setting wifi country code

brada4, 2025-05-01:
Stop with pictures, thry are hard to zoom on mobile. No country code in your pictures

Run `iw list` and rxtract "supported intetface combinations" section and the channel maps )list with frequencirs)

leanne63, 2025-05-01:
`iw list` "supported interface combinations": I only see "supported interface modes" - is that what you want?
	Supported interface modes; same for both Wiphy phy1 and Wiphy phy0:
		 * IBSS
		 * managed
		 * AP
		 * AP/VLAN
		 * monitor
		 * mesh point
		 * P2P-client
		 * P2P-GO

What is the "channel maps list"? I have this under "valid interface combinations":
		   total <= 19, #channels <= 1, STA/AP BI must match, radar detect widths: { 20 MHz (no HT), 20 MHz, 40 MHz, 80 MHz, 160 MHz }

Again, that's the same for both radios.

Also, since you can't see the pictures, and since you suggested updating the country code, which I can't see having any relation whatsoever to what's happening (and I just know enough to be dangerous, probably, so I'm just seeking understanding here) - let me reiterate in text: **when I click Save after scanning/configuring a network for radio0, the resulting change list for Save & Apply shows that radio0 will be deleted (and it is; it's gone from all interfaces/devices after clicking Save & Apply)**.

brada4, 2025-05-01:
> which I can't see

It is YOUR legal obligation.

leanne63, 2025-05-01:
> It is YOUR legal obligation.

Umm... "... which I can't see _having any relation whatsoever to what's happening_...". 

After a few re-installs of OpenWRT to see which version(s) were affected and whether the issue was consistent, I got tired of repeatedly entering the country code. After a few more, I went ahead and entered it again - because I understand "It is [MY] legal obligation." if I'm going to be using the router. As a result, I now know, from a troubleshooting perspective, that the country code doesn't affect whether the radio and its associated network disappears/is deleted or not.

Now, can you provide any tips as to how to fix this "bug"? Or, maybe you can test to see if you're having the same problem with your own router(s) running 24.10 or 24.10.1? Or anything else that might be relevant to solving the problem/determining whether it's a valid bug in those two OpenWRT versions? It's happening on two different routers of mine, so I'd appreciate input on whether it's happening on other routers as well.

brada4, 2025-05-02:
You should know for certain that you are using passivr scan due to no-ir restriction in the world domain. You can not outsmart it, no access point outside 80mhz ch36 without country code.

leanne63, 2025-05-13:
Just installed the latest Firmware Selector firmware versions for each of my devices. The problem seems to no longer be occurring on that firmware.

Both the GL.iNet and the TP-Link now show:
> Powered by LuCI (HEAD detached at 2ac26e56) branch (25.103.51521~2ac26e5) / OpenWrt 24.10.1 (r28597-0425664679)

Sorry, I didn't get that information from my prior firmwares. They were both 24.10.1, but I got them from the TOH for each of the devices. (The Firmware Selector was having issues when I downloaded the earlier versions). Then, I read that sometimes the firmware is further updated over a short time, and the Firmware Selector is now working, so I tried the download from there.

Thanks much for your consideration of this issue!

closed 2025-05-13 by leanne63