openwrt/openwrt #18673: Low network speed without off-floating

WeSTManCoder, 2025-05-01:
### Describe the bug

Without off-floating, the network speed is a maximum of 100 Mbps. But on the Archer C6U and Beeline Smartbox Giga devices, the speed reaches up to 1 Gbps without off-floating.

### OpenWrt version

r28427-6df0e3d02a

### OpenWrt release

24.10.0

### OpenWrt target/subtarget

ramips/mt7621

### Device

netis N6

### Image kind

Official downloaded image

### Steps to reproduce

install openwrt 24.10
Test network speed (speedtest example)

### Actual behaviour

The load on one core is 100%. With a load on all cores - a maximum of 300 Mbps

### Expected behaviour

Speeds around or above 1 Gbps without hardware or software flow offloading

### Diffconfig

```text

```

### Terms

- [x] I am reporting an issue for OpenWrt, not an unsupported fork.

brada4, 2025-05-01:
Show links o 9 graphs using  24.10.1 and
https://www.waveform.com/tools/bufferbloat

Specs are nearly identical, should be +-10% apart max.

russkinz, 2025-05-02:
Interestingly I am seeing this on x86-64 as well, iperf3 shows full speed to the router, and the router to the internet is fine, but all traffic to the internet from any devices in my network seems to hit a hard limit of 100mbit.

WeSTManCoder, 2025-05-02:
> Interestingly I am seeing this on x86-64 as well, iperf3 shows full speed to the router, and the router to the internet is fine, but all traffic to the internet from any devices in my network seems to hit a hard limit of 100mbit.

maybe enable packet steering all cores? 
network -> interfaces -> global network options -> packet steering enable on all cores
or enable hardware offloating
network -> firewall -> Flow offloading type hardware

WeSTManCoder, 2025-05-02:
> Show links to 9 graphs using 24.10.1 and https://www.waveform.com/tools/bufferbloat
> 
> Specs are nearly identical, should be +-10% apart max.

NETIS N6 (24.10.0 r28427-6df0e3d02a):

without offloating
https://www.waveform.com/tools/bufferbloat?test-id=4c0a3207-1c18-4cd0-817c-610692a33c04
https://www.waveform.com/tools/bufferbloat?test-id=67df4cbf-f0ee-413d-856d-28e40ed9da1a
https://www.waveform.com/tools/bufferbloat?test-id=96127aec-8a02-4214-ba8e-b8fd5fc87132
https://www.waveform.com/tools/bufferbloat?test-id=37271c1e-a7a9-4334-ad17-bc48ed31886a
https://www.waveform.com/tools/bufferbloat?test-id=15a4cad5-3d31-400e-b09a-caf8f25ba6f3
https://www.waveform.com/tools/bufferbloat?test-id=e5f1fb79-630d-4244-bba3-802fd84a5558
https://www.waveform.com/tools/bufferbloat?test-id=f724756c-afea-4af2-a5dc-85c57dd25a32
https://www.waveform.com/tools/bufferbloat?test-id=0f138f08-1c93-45ee-b541-9a33afbadbd2
https://www.waveform.com/tools/bufferbloat?test-id=06de9981-f423-4c34-9b59-106ba2cc7078

using hardware offloating:
https://www.waveform.com/tools/bufferbloat?test-id=68f5b050-a052-4dde-93b6-9ecf5505fcaa
https://www.waveform.com/tools/bufferbloat?test-id=2f841395-4fe0-44d1-a469-9e7b8d12fcd7
https://www.waveform.com/tools/bufferbloat?test-id=b88375d6-d599-4825-8c38-301034a0d697

using software offloating
https://www.waveform.com/tools/bufferbloat?test-id=8e18951b-3dd5-4440-81f6-8a938662c53c
https://www.waveform.com/tools/bufferbloat?test-id=723947ae-ee03-4857-b31d-4e4deaa8e406
https://www.waveform.com/tools/bufferbloat?test-id=1e865cf9-7746-4444-bb6a-25deaaa75c75

Archer C6U v1 (23.05.0 r23497-6637af95aa)

without offloating
https://www.waveform.com/tools/bufferbloat?test-id=2baa883d-c740-4a2e-8a34-bcbc00a3d443
https://www.waveform.com/tools/bufferbloat?test-id=f629ee63-7ed9-4959-8935-e7e9102e3670
https://www.waveform.com/tools/bufferbloat?test-id=2d4ec571-8752-4deb-833a-db5a404a8261

Archer C6U v1 (24.10.1 r28597-0425664679)

without offloating
https://www.waveform.com/tools/bufferbloat?test-id=fe0dccff-ea8b-4d1d-8860-e5edf57a1570
https://www.waveform.com/tools/bufferbloat?test-id=01058639-14ba-400b-b02c-48afaab1b82b
https://www.waveform.com/tools/bufferbloat?test-id=da08830f-f555-4f50-87a1-f7c027676729

DIFFERENT by default:
	24.10.0 packet steering using one core
	24.10.1 packet steering using one core
	23.05.0 packet steering using all cores

It looks like the situation is identical +/-

brada4, 2025-05-02:
Is there some flow control on wan and lanX port? (ethtool -a to check -A to turn all 3 off -r to renegotiate)?
Speeds differ 3x which is major regression.
 You can steer to all cores in 24.10 to get back old behaviour, new default steers to one hyperthread of the each of 2 cores

WeSTManCoder, 2025-05-02:
> Is there some flow control on wan and lanX port? (ethtool -a to check -A to turn all 3 off -r to renegotiate)? Speeds differ 3x which is major regression. You can steer to all cores in 24.10 to get back old behaviour, new default steers to one hyperthread of the each of 2 cores

ethtool is not installed, I have already enabled packet steering on all cores and the speed has increased, but if I use it at maximum, then my connection quality deteriorates very much, even to the point of freezing, for example, my voice in discord

brada4, 2025-05-03:
zapret software does ruleset.uc patching to support offloads by itself, depending on detected/chosen settings, say its detect script to enable offload, dont use luci and you are good. Should not affect bulk up/downloads after initial packet manipulations. Or in other words all devices suffer equally.

You can keep one of 2 good devices aside, leave netis and one you find easier to reset.

Try `iperf3 -Z --bidir <server> -p 5201` against server: 
https://iperf.fr/iperf-servers.php

While doing that run htop (just like ethtool - install it) and then in F2 - Setup enable "Detailed CPU Time (System/IO-Wait/...)" and disable "Hide Kenrel threads" and press Ctrl-S to freeze terminal mid-test and make screenshot , then Ctrl-Q to unfreeze.

Share just one from netis one from good device (no idea yet whats messing 2 nearly identical devices down)

WeSTManCoder, 2025-05-03:
> Share just one from netis one from good device (no idea yet whats messing 2 nearly identical devices down)

Thank you very much, I will do the tests and come back with the results.