openwrt/openwrt #18678: Compilation error

grosjo, 2025-05-02:
### Describe the bug

Error got:

```
In file included from mips-opc.c:29:
mips-opc.c: In function 'decode_mips_operand':
mips-formats.h:86:7: error: expected identifier or '(' before 'static_assert'
   86 |       static_assert[(1 << (SIZE)) == ARRAY_SIZE (MAP)]; \
      |       ^~~~~~~~~~~~~
mips-opc.c:214:15: note: in expansion of macro 'MAPPED_REG'
  214 |     case 'z': MAPPED_REG (0, 0, GP, reg_0_map);
      |               ^~~~~~~~~~
make[8]: *** [Makefile:1077: mips-opc.lo] Error 1
make[8]: Leaving directory '/files/src/openwrt/kuwfi830/build_dir/toolchain-mips_24kc_gcc-14.2.0_musl/binutils-2.42/opcodes'
```

### OpenWrt version

Not relevant

### OpenWrt release

Not relevant

### OpenWrt target/subtarget

Not relevant

### Device

Not relevant

### Image kind

Official downloaded image

### Steps to reproduce

Build with GCC 15.1

### Actual behaviour

Not compiling

### Expected behaviour

Compiling

### Diffconfig

```text

```

### Terms

- [x] I am reporting an issue for OpenWrt, not an unsupported fork.

grosjo, 2025-05-02:
Porbably upgrade bintuiles to 2.44 (latest) ?

closed 2025-05-02 by openwrt-bot

grosjo, 2025-05-04:
@robimarko still not enough

Error:
```
In file included from mips-opc.c:29:
mips-opc.c: In function 'decode_mips_operand':
mips-formats.h:86:7: error: expected identifier or '(' before 'static_assert'
   86 |       static_assert((1 << (SIZE)) == ARRAY_SIZE (MAP)); \
      |       ^~~~~~~~~~~~~
mips-opc.c:214:15: note: in expansion of macro 'MAPPED_REG'
  214 |     case 'z': MAPPED_REG (0, 0, GP, reg_0_map);
      |               ^~~~~~~~~~

```

robimarko, 2025-05-04:
It has to be enough, I had the exact same issue and backport fixed it.

You gotta do make toolchain/binutils/clean do though first

commit d3216173 (2025-05-02): toolchain: binutils: fix compilation with GCC15

GCC15 has switched the C language default from GNU17 to GNU23[1] and this
causes builds to fail with:
In file included from mips-opc.c:29:
mips-opc.c: In function 'decode_mips_operand':
mips-formats.h:86:7: error: expected identifier or '(' before 'static_assert'
   86 |       static_assert[(1 << (SIZE)) == ARRAY_SIZE (MAP)]; \
      |       ^~~~~~~~~~~~~
mips-opc.c:214:15: note: in expansion of macro 'MAPPED_REG'
  214 |     case 'z': MAPPED_REG (0, 0, GP, reg_0_map);
      |               ^~~~~~~~~~

So, backport upstream fix for this[2] to fi [... cut]
commit 069f988d (2025-05-03): toolchain: binutils: fix compilation with GCC15

GCC15 has switched the C language default from GNU17 to GNU23[1] and this
causes builds to fail with:
In file included from mips-opc.c:29:
mips-opc.c: In function 'decode_mips_operand':
mips-formats.h:86:7: error: expected identifier or '(' before 'static_assert'
   86 |       static_assert[(1 << (SIZE)) == ARRAY_SIZE (MAP)]; \
      |       ^~~~~~~~~~~~~
mips-opc.c:214:15: note: in expansion of macro 'MAPPED_REG'
  214 |     case 'z': MAPPED_REG (0, 0, GP, reg_0_map);
      |               ^~~~~~~~~~

So, backport upstream fix for this[2] to fi [... cut]
commit bd9b964b (2025-06-08): toolchain: binutils: fix compilation with GCC15

GCC15 has switched the C language default from GNU17 to GNU23[1] and this
causes builds to fail with:
In file included from mips-opc.c:29:
mips-opc.c: In function 'decode_mips_operand':
mips-formats.h:86:7: error: expected identifier or '(' before 'static_assert'
   86 |       static_assert[(1 << (SIZE)) == ARRAY_SIZE (MAP)]; \
      |       ^~~~~~~~~~~~~
mips-opc.c:214:15: note: in expansion of macro 'MAPPED_REG'
  214 |     case 'z': MAPPED_REG (0, 0, GP, reg_0_map);
      |               ^~~~~~~~~~

So, backport upstream fix for this[2] to fi [... cut]