openwrt/openwrt #18680: Xiaomi Router AX3000T random crash (mtk_soc_eth 15100000.ethernet eth0: NETDEV WATCHDOG: CPU: 1: transmit queue 3 timed out)

SnIPeRSnIPeR, 2025-05-02:
### Describe the bug

Getting the following crash last night.

Wi-Fi connection continue to function without internet connection. Works fine after reboot.

`Fri May  2 12:46:10 2025 kern.crit kernel: [252307.692383] mtk_soc_eth 15100000.ethernet eth0: NETDEV WATCHDOG: CPU: 1: transmit queue 3 timed out 19125000 ms
Fri May  2 12:46:15 2025 kern.crit kernel: [252312.651161] mtk_soc_eth 15100000.ethernet eth0: NETDEV WATCHDOG: CPU: 1: transmit queue 3 timed out 19129960 ms
Fri May  2 12:46:20 2025 kern.crit kernel: [252317.689922] mtk_soc_eth 15100000.ethernet eth0: NETDEV WATCHDOG: CPU: 1: transmit queue 3 timed out 19135000 ms
Fri May  2 12:46:25 2025 kern.crit kernel: [252322.648700] mtk_soc_eth 15100000.ethernet eth0: NETDEV WATCHDOG: CPU: 1: transmit queue 3 timed out 19139960 ms
Fri May  2 12:46:30 2025 kern.crit kernel: [252327.687461] mtk_soc_eth 15100000.ethernet eth0: NETDEV WATCHDOG: CPU: 1: transmit queue 3 timed out 19145000 ms
Fri May  2 12:46:35 2025 kern.crit kernel: [252332.646237] mtk_soc_eth 15100000.ethernet eth0: NETDEV WATCHDOG: CPU: 1: transmit queue 3 timed out 19149960 ms
Fri May  2 12:46:40 2025 kern.crit kernel: [252337.684995] mtk_soc_eth 15100000.ethernet eth0: NETDEV WATCHDOG: CPU: 1: transmit queue 3 timed out 19155000 ms
Fri May  2 12:46:45 2025 kern.crit kernel: [252342.643774] mtk_soc_eth 15100000.ethernet eth0: NETDEV WATCHDOG: CPU: 1: transmit queue 3 timed out 19159960 ms
Fri May  2 12:46:50 2025 kern.crit kernel: [252347.682531] mtk_soc_eth 15100000.ethernet eth0: NETDEV WATCHDOG: CPU: 1: transmit queue 3 timed out 19165000 ms
Fri May  2 12:46:55 2025 kern.crit kernel: [252352.641317] mtk_soc_eth 15100000.ethernet eth0: NETDEV WATCHDOG: CPU: 1: transmit queue 3 timed out 19169960 ms
Fri May  2 12:47:00 2025 kern.crit kernel: [252357.690065] mtk_soc_eth 15100000.ethernet eth0: NETDEV WATCHDOG: CPU: 1: transmit queue 3 timed out 19175010 ms
Fri May  2 12:47:05 2025 kern.crit kernel: [252362.638851] mtk_soc_eth 15100000.ethernet eth0: NETDEV WATCHDOG: CPU: 1: transmit queue 3 timed out 19179960 ms
Fri May  2 12:47:10 2025 kern.crit kernel: [252367.677604] mtk_soc_eth 15100000.ethernet eth0: NETDEV WATCHDOG: CPU: 1: transmit queue 3 timed out 19185000 ms
Fri May  2 12:47:15 2025 kern.crit kernel: [252372.636388] mtk_soc_eth 15100000.ethernet eth0: NETDEV WATCHDOG: CPU: 1: transmit queue 3 timed out 19189960 ms
Fri May  2 12:47:20 2025 kern.crit kernel: [252377.675142] mtk_soc_eth 15100000.ethernet eth0: NETDEV WATCHDOG: CPU: 1: transmit queue 3 timed out 19195000 ms
Fri May  2 12:47:25 2025 kern.crit kernel: [252382.633930] mtk_soc_eth 15100000.ethernet eth0: NETDEV WATCHDOG: CPU: 1: transmit queue 3 timed out 19199960 ms
Fri May  2 12:47:30 2025 kern.crit kernel: [252387.682675] mtk_soc_eth 15100000.ethernet eth0: NETDEV WATCHDOG: CPU: 1: transmit queue 3 timed out 19205010 ms
Fri May  2 12:47:35 2025 kern.crit kernel: [252392.631488] mtk_soc_eth 15100000.ethernet eth0: NETDEV WATCHDOG: CPU: 1: transmit queue 3 timed out 19209960 ms
Fri May  2 12:47:40 2025 kern.crit kernel: [252397.670220] mtk_soc_eth 15100000.ethernet eth0: NETDEV WATCHDOG: CPU: 1: transmit queue 3 timed out 19215000 ms
Fri May  2 12:47:45 2025 kern.crit kernel: [252402.628998] mtk_soc_eth 15100000.ethernet eth0: NETDEV WATCHDOG: CPU: 1: transmit queue 3 timed out 19219960 ms
Fri May  2 12:47:50 2025 kern.crit kernel: [252407.677753] mtk_soc_eth 15100000.ethernet eth0: NETDEV WATCHDOG: CPU: 1: transmit queue 3 timed out 19225010 ms
Fri May  2 12:47:55 2025 kern.crit kernel: [252412.626538] mtk_soc_eth 15100000.ethernet eth0: NETDEV WATCHDOG: CPU: 1: transmit queue 3 timed out 19229960 ms
Fri May  2 12:48:00 2025 kern.crit kernel: [252417.675285] mtk_soc_eth 15100000.ethernet eth0: NETDEV WATCHDOG: CPU: 1: transmit queue 3 timed out 19235010 ms
Fri May  2 12:48:05 2025 kern.crit kernel: [252422.624071] mtk_soc_eth 15100000.ethernet eth0: NETDEV WATCHDOG: CPU: 1: transmit queue 3 timed out 19239960 ms
Fri May  2 12:48:10 2025 kern.crit kernel: [252427.662829] mtk_soc_eth 15100000.ethernet eth0: NETDEV WATCHDOG: CPU: 1: transmit queue 3 timed out 19245000 ms
Fri May  2 12:48:15 2025 kern.crit kernel: [252432.621617] mtk_soc_eth 15100000.ethernet eth0: NETDEV WATCHDOG: CPU: 1: transmit queue 3 timed out 19249960 ms
Fri May  2 12:48:20 2025 kern.crit kernel: [252437.660363] mtk_soc_eth 15100000.ethernet eth0: NETDEV WATCHDOG: CPU: 1: transmit queue 3 timed out 19255000 ms
Fri May  2 12:48:25 2025 kern.crit kernel: [252442.619144] mtk_soc_eth 15100000.ethernet eth0: NETDEV WATCHDOG: CPU: 1: transmit queue 3 timed out 19259960 ms
Fri May  2 12:48:30 2025 kern.crit kernel: [252447.657899] mtk_soc_eth 15100000.ethernet eth0: NETDEV WATCHDOG: CPU: 1: transmit queue 3 timed out 19265000 ms
Fri May  2 12:48:35 2025 kern.crit kernel: [252452.616683] mtk_soc_eth 15100000.ethernet eth0: NETDEV WATCHDOG: CPU: 1: transmit queue 3 timed out 19269960 ms
Fri May  2 12:48:40 2025 kern.crit kernel: [252457.655436] mtk_soc_eth 15100000.ethernet eth0: NETDEV WATCHDOG: CPU: 1: transmit queue 3 timed out 19275000 ms
Fri May  2 12:48:45 2025 kern.crit kernel: [252462.614310] mtk_soc_eth 15100000.ethernet eth0: NETDEV WATCHDOG: CPU: 1: transmit queue 3 timed out 19279960 ms
Fri May  2 12:48:50 2025 kern.crit kernel: [252467.662970] mtk_soc_eth 15100000.ethernet eth0: NETDEV WATCHDOG: CPU: 1: transmit queue 3 timed out 19285010 ms
Fri May  2 12:48:55 2025 kern.crit kernel: [252472.611760] mtk_soc_eth 15100000.ethernet eth0: NETDEV WATCHDOG: CPU: 1: transmit queue 3 timed out 19289960 ms
Fri May  2 12:49:00 2025 kern.crit kernel: [252477.650519] mtk_soc_eth 15100000.ethernet eth0: NETDEV WATCHDOG: CPU: 1: transmit queue 3 timed out 19295000 ms
Fri May  2 12:49:05 2025 kern.crit kernel: [252482.609405] mtk_soc_eth 15100000.ethernet eth0: NETDEV WATCHDOG: CPU: 1: transmit queue 3 timed out 19299960 ms
Fri May  2 12:49:10 2025 kern.crit kernel: [252487.648048] mtk_soc_eth 15100000.ethernet eth0: NETDEV WATCHDOG: CPU: 1: transmit queue 3 timed out 19305000 ms
Fri May  2 12:49:15 2025 kern.crit kernel: [252492.616827] mtk_soc_eth 15100000.ethernet eth0: NETDEV WATCHDOG: CPU: 1: transmit queue 3 timed out 19309970 ms
Fri May  2 12:49:20 2025 kern.crit kernel: [252497.645585] mtk_soc_eth 15100000.ethernet eth0: NETDEV WATCHDOG: CPU: 1: transmit queue 3 timed out 19315000 ms
Fri May  2 12:49:25 2025 kern.crit kernel: [252502.604369] mtk_soc_eth 15100000.ethernet eth0: NETDEV WATCHDOG: CPU: 1: transmit queue 3 timed out 19319960 ms
Fri May  2 12:49:30 2025 kern.crit kernel: [252507.653121] mtk_soc_eth 15100000.ethernet eth0: NETDEV WATCHDOG: CPU: 1: transmit queue 3 timed out 19325010 ms
Fri May  2 12:49:35 2025 kern.crit kernel: [252512.611895] mtk_soc_eth 15100000.ethernet eth0: NETDEV WATCHDOG: CPU: 1: transmit queue 3 timed out 19329970 ms
Fri May  2 12:49:40 2025 kern.crit kernel: [252517.640657] mtk_soc_eth 15100000.ethernet eth0: NETDEV WATCHDOG: CPU: 1: transmit queue 3 timed out 19335000 ms
Fri May  2 12:49:45 2025 kern.crit kernel: [252522.599438] mtk_soc_eth 15100000.ethernet eth0: NETDEV WATCHDOG: CPU: 1: transmit queue 3 timed out 19339960 ms
Fri May  2 12:49:50 2025 kern.crit kernel: [252527.638197] mtk_soc_eth 15100000.ethernet eth0: NETDEV WATCHDOG: CPU: 1: transmit queue 3 timed out 19345000 ms
Fri May  2 12:49:55 2025 kern.crit kernel: [252532.596974] mtk_soc_eth 15100000.ethernet eth0: NETDEV WATCHDOG: CPU: 1: transmit queue 3 timed out 19349960 ms
Fri May  2 12:50:00 2025 kern.crit kernel: [252537.635734] mtk_soc_eth 15100000.ethernet eth0: NETDEV WATCHDOG: CPU: 1: transmit queue 3 timed out 19355000 ms
Fri May  2 12:50:05 2025 kern.crit kernel: [252542.594512] mtk_soc_eth 15100000.ethernet eth0: NETDEV WATCHDOG: CPU: 1: transmit queue 3 timed out 19359960 ms
Fri May  2 12:50:10 2025 kern.crit kernel: [252547.633270] mtk_soc_eth 15100000.ethernet eth0: NETDEV WATCHDOG: CPU: 1: transmit queue 3 timed out 19365000 ms
Fri May  2 12:50:15 2025 kern.crit kernel: [252552.592049] mtk_soc_eth 15100000.ethernet eth0: NETDEV WATCHDOG: CPU: 1: transmit queue 3 timed out 19369960 ms
Fri May  2 12:50:20 2025 kern.crit kernel: [252557.640804] mtk_soc_eth 15100000.ethernet eth0: NETDEV WATCHDOG: CPU: 1: transmit queue 3 timed out 19375010 ms
Fri May  2 12:50:25 2025 kern.crit kernel: [252562.589593] mtk_soc_eth 15100000.ethernet eth0: NETDEV WATCHDOG: CPU: 1: transmit queue 3 timed out 19379960 ms
[... cut ...]
Fri May  2 12:51:20 2025 kern.crit kernel: [252617.616027] mtk_soc_eth 15100000.ethernet eth0: NETDEV WATCHDOG: CPU: 1: transmit queue 3 timed out 19435000 ms
Fri May  2 12:51:25 2025 kern.crit kernel: [252622.584801] mtk_soc_eth 15100000.ethernet eth0: NETDEV WATCHDOG: CPU: 1: transmit queue 3 timed out 19439970 ms
[... 16 lines cut ...]
### OpenWrt release

24.10.1

### OpenWrt target/subtarget

mediatek/filogicv

### Device

Xiaomi Mi Router AX3000T

### Image kind

Official downloaded image

### Diffconfig

```text

```

### Terms

- [x] I am reporting an issue for OpenWrt, not an unsupported fork.

SnIPeRSnIPeR, 2025-05-02:
> mediatek/filogicv

mediatek/filogic
Xiaomi AX3000T: https://openwrt.org/inbox/toh/xiaomi/ax3000t

brada4, 2025-05-03:
If you search "NETDEV WATCHDOG" in this repo youll find dozen sibling issues. First - ensure cooling vents are not covered. Then read other threads.

lessload, 2025-05-06:
Weird wan port on it , why it use wan\lan2\lan3\lan4 , port order should be lan1\lan2\lan3\wan , someone should fix it

NitroOxid, 2025-05-06:
Same problem on Routerich AX3000 V1, using openwrt 24.10.1
With the same hardware (MediaTek MT7981BA)

brada4, 2025-05-06:
It is long running issue, cause not yet found.

https://github.com/search?q=repo%3Aopenwrt%2Fopenwrt+NETDEV+WATCHDOG+mtk_soc_eth&type=issues