openwrt/openwrt #18687: 24.10.1: Asus RT-AX89X incorrect wi-fi modules binding

RiddlerX2, 2025-05-02:
### Describe the bug

While configuring in LuCI I have two radio modules which is detect its own frequency props. But on setup completion only one network (2.4GHz) appears on other devices, which is surely support 5GHz bandwidth (they works on original firmware). Disabling 2.4GHz module does not help 5GHz module not available for discovery on other devices (SSID not hidden for sure). Some times on device restart 5GHz network available for short time but disconnected quickly after 5-30 minutes. Also I have found that 2 LED on device works incorrectly, if I disable 5GHz module 2.4GHz LED turned off and same with 2.4GHz module it turns off 5GHz LED.

### OpenWrt version

r28597-0425664679

### OpenWrt release

24.10.1

### OpenWrt target/subtarget

qualcommax/ipq807x

### Device

Asus RT-AX89X rev. B2

### Image kind

Official downloaded image

### Actual behaviour

5GHz bandwidth not available most of time.
LEDs for WI-FI are mixed up and has wrong bandwidth assignment.

### Expected behaviour

5GHz bandwidth should be available for discovery.
LEDs for WI-FI have correct corresponding to their bandwidth.

### Diffconfig

```text

```

### Terms

- [x] I am reporting an issue for OpenWrt, not an unsupported fork.

Djfe, 2025-05-17:
Was this issue apparent right from when you installed OpenWrt or has it worked before?
Please post the output of:
`iw phy phy0 info`
`iw phy phy1 info`

RiddlerX2, 2025-05-17:
Yes, problem appears right after OpenWRT installation.
```root@wrt:~# iw phy phy0 info
Wiphy phy0
        wiphy index: 0
        max # scan SSIDs: 16
        max scan IEs length: 398 bytes
        max # sched scan SSIDs: 0
        max # match sets: 0
        Retry short limit: 7
        Retry long limit: 4
        Coverage class: 0 (up to 0m)
        Device supports AP-side u-APSD.
        Available Antennas: TX 0xff RX 0xff
        Configured Antennas: TX 0xff RX 0xff
        Supported interface modes:
                 * managed
                 * AP
                 * monitor
                 * mesh point
        Band 2:
                Capabilities: 0x19e7
                        RX LDPC
                        HT20/HT40
                        Dynamic SM Power Save
                        RX HT20 SGI
                        RX HT40 SGI
                        TX STBC
                        RX STBC 1-stream
                        Max AMSDU length: 7935 bytes
                        DSSS/CCK HT40
                Maximum RX AMPDU length 65535 bytes (exponent: 0x003)
                Minimum RX AMPDU time spacing: No restriction (0x00)
                HT TX/RX MCS rate indexes supported: 0-63
                VHT Capabilities (0x339ff9f6):
                        Max MPDU length: 11454
                        Supported Channel Width: 160 MHz
                        RX LDPC
                        short GI (80 MHz)
                        short GI (160/80+80 MHz)
                        TX STBC
                        SU Beamformer
                        SU Beamformee
                        MU Beamformer
                        MU Beamformee
                        RX antenna pattern consistency
                        TX antenna pattern consistency
                VHT RX MCS set:
                        1 streams: MCS 0-9
                        2 streams: MCS 0-9
                        3 streams: MCS 0-9
                        4 streams: MCS 0-9
                        5 streams: MCS 0-9
                        6 streams: MCS 0-9
                        7 streams: MCS 0-9
                        8 streams: MCS 0-9
                VHT RX highest supported: 0 Mbps
                VHT TX MCS set:
                        1 streams: MCS 0-9
                        2 streams: MCS 0-9
                        3 streams: MCS 0-9
                        4 streams: MCS 0-9
                        5 streams: MCS 0-9
                        6 streams: MCS 0-9
                        7 streams: MCS 0-9
                        8 streams: MCS 0-9
                VHT TX highest supported: 0 Mbps
                VHT extended NSS: supported
                HE Iftypes: managed
                        HE MAC Capabilities (0x000b9a181040):
                                +HTC HE Supported
[... 487 lines cut ...]
                 * maximum A-MSDU length
                 * supported channel width
                 * short GI for 40 MHz
                 * max A-MPDU length exponent
                 * min MPDU start spacing
        max # scan plans: 1
        max scan plan interval: -1
        max scan plan iterations: 0
        Maximum associated stations in AP mode: 512
        Supported extended features:
                * [ RRM ]: RRM
                * [ SET_SCAN_DWELL ]: scan dwell setting
                * [ FILS_STA ]: STA FILS (Fast Initial Link Setup)
                * [ CQM_RSSI_LIST ]: multiple CQM_RSSI_THOLD records
                * [ CONTROL_PORT_OVER_NL80211 ]: control port over nl80211
                * [ ACK_SIGNAL_SUPPORT ]: ack signal level support
                * [ TXQS ]: FQ-CoDel-enabled intermediate TXQs
                * [ ENABLE_FTM_RESPONDER ]: enable FTM (Fine Time Measurement) responder
                * [ STA_TX_PWR ]: TX power control per station
                * [ CONTROL_PORT_NO_PREAUTH ]: disable pre-auth over nl80211 control port support
                * [ SCAN_FREQ_KHZ ]: scan on kHz frequency support
                * [ CONTROL_PORT_OVER_NL80211_TX_STATUS ]: tx status for nl80211 control port support
                * [ BSS_COLOR ]: BSS coloring support
                * [ POWERED_ADDR_CHANGE ]: can change MAC address while up
```

Djfe, 2025-05-17:
ok that's helpful
For the full picture: what is the output of these commands?
`uci show wireless`
`iwinfo`

RiddlerX2, 2025-05-17:
Sure
```root@wrt:~# uci show wireless
wireless.radio0=wifi-device
wireless.radio0.type='mac80211'
wireless.radio0.path='platform/soc@0/c000000.wifi'
wireless.radio0.band='5g'
wireless.radio0.channel='auto'
wireless.radio0.htmode='HE160'
wireless.radio0.cell_density='0'
wireless.default_radio0=wifi-iface
wireless.default_radio0.device='radio0'
wireless.default_radio0.network='lan'
wireless.default_radio0.mode='ap'
wireless.default_radio0.ssid='RiddleWrt'
wireless.default_radio0.encryption='sae-mixed'
wireless.default_radio0.key='***'
wireless.default_radio0.ocv='0'
wireless.default_radio0.ifname='wls6'
wireless.default_radio0.wmm='0'
wireless.radio1=wifi-device
wireless.radio1.type='mac80211'
wireless.radio1.path='platform/soc@0/c000000.wifi+1'
wireless.radio1.band='2g'
wireless.radio1.channel='auto'
wireless.radio1.htmode='HE40'
wireless.radio1.legacy_rates='1'
wireless.radio1.cell_density='0'
wireless.default_radio1=wifi-iface
wireless.default_radio1.device='radio1'
wireless.default_radio1.network='lan'
wireless.default_radio1.mode='ap'
wireless.default_radio1.ssid='RiddleWrt5'
wireless.default_radio1.encryption='psk-mixed'
wireless.default_radio1.ifname='wls5'
wireless.default_radio1.key='***'
```

```root@wrt:~# iwinfo
wls5      ESSID: "RiddleWrt5"
          Access Point: 58:11:22:34:4A:78
          Mode: Master  Channel: 1 (2.412 GHz)  HT Mode: HE20
          Center Channel 1: 1 2: unknown
          Tx-Power: 20 dBm  Link Quality: 68/70
          Signal: -42 dBm  Noise: -101 dBm
          Bit Rate: 46.0 MBit/s
          Encryption: mixed WPA/WPA2 PSK (CCMP)
          Type: nl80211  HW Mode(s): 802.11ax/b/g/n
          Hardware: embedded [Qualcomm Atheros IPQ8074]
          TX power offset: none
          Frequency offset: none
          Supports VAPs: yes  PHY name: phy1

wls6      ESSID: "RiddleWrt"
          Access Point: 58:11:22:34:4A:7C
          Mode: Master  Channel: 100 (5.500 GHz)  HT Mode: HE160
          Center Channel 1: 114 2: unknown
          Tx-Power: 30 dBm  Link Quality: unknown/70
          Signal: unknown  Noise: -107 dBm
          Bit Rate: unknown
          Encryption: mixed WPA2/WPA3 PSK/SAE (CCMP)
          Type: nl80211  HW Mode(s): 802.11ac/ax/n
          Hardware: embedded [Qualcomm Atheros IPQ8074]
          TX power offset: none
          Frequency offset: none
          Supports VAPs: yes  PHY name: phy0
```

Djfe, 2025-05-17:
Please set `wireless.radio0.country` and `wireless.radio1.country`
Either via uci or LuCI

this is set `wireless.radio0.channel='auto'` but not `wireless.radio0.channels='36 100'`
I'm not sure whether "channels" is required, but maybe `logread` helps us out.
maybe hostapd error somewhere in the log (around the section about ACS)

if you want them to switch names, you only need to change SSIDs

phy0/radio0 is 5GHz and runs with config `wireless.default_radio0.ssid='RiddleWrt'`
phy1/radio1 is 2.4GHz and runs with config `wireless.default_radio1.ssid='RiddleWrt5'``

RiddlerX2, 2025-05-17:
I'll try next options:
Set `wireless.radio0.country='US'` and `wireless.radio1.country='US'` this will not set `wireless.radio[0-1].channels` automatically.
Restart and still 5GHz network not available.
Have nothing suspicious in the log:
```
Sat May 17 23:22:03 2025 daemon.notice hostapd: wls6: interface state UNINITIALIZED->COUNTRY_UPDATE
Sat May 17 23:22:03 2025 daemon.notice hostapd: ACS: Automatic channel selection started, this may take a bit
Sat May 17 23:22:03 2025 daemon.notice hostapd: wls6: interface state COUNTRY_UPDATE->ACS
Sat May 17 23:22:03 2025 daemon.notice hostapd: wls6: ACS-STARTED
Sat May 17 23:22:03 2025 daemon.notice netifd: Wireless device 'radio1' is now up
Sat May 17 23:22:03 2025 daemon.notice netifd: Wireless device 'radio0' is now up
Sat May 17 23:22:04 2025 kern.info kernel: [817148.062407] br-lan: port 11(wls5) entered disabled state
Sat May 17 23:22:04 2025 kern.info kernel: [817148.062702] br-lan: port 12(wls6) entered disabled state
Sat May 17 23:22:04 2025 daemon.notice hostapd: wls6: DFS-PRE-CAC-EXPIRED freq=5500 ht_enabled=0 chan_offset=0 chan_width=0 cf1=5500 cf2=0
Sat May 17 23:22:04 2025 daemon.notice hostapd: wls6: DFS-PRE-CAC-EXPIRED freq=5520 ht_enabled=0 chan_offset=0 chan_width=0 cf1=5520 cf2=0
Sat May 17 23:22:04 2025 daemon.notice hostapd: wls6: DFS-PRE-CAC-EXPIRED freq=5540 ht_enabled=0 chan_offset=0 chan_width=0 cf1=5540 cf2=0
Sat May 17 23:22:04 2025 daemon.notice hostapd: wls6: DFS-PRE-CAC-EXPIRED freq=5560 ht_enabled=0 chan_offset=0 chan_width=0 cf1=5560 cf2=0
Sat May 17 23:22:04 2025 daemon.notice hostapd: wls6: DFS-PRE-CAC-EXPIRED freq=5580 ht_enabled=0 chan_offset=0 chan_width=0 cf1=5580 cf2=0
Sat May 17 23:22:04 2025 daemon.notice hostapd: wls6: DFS-PRE-CAC-EXPIRED freq=5600 ht_enabled=0 chan_offset=0 chan_width=0 cf1=5600 cf2=0
Sat May 17 23:22:04 2025 daemon.notice hostapd: wls6: DFS-PRE-CAC-EXPIRED freq=5620 ht_enabled=0 chan_offset=0 chan_width=0 cf1=5620 cf2=0
Sat May 17 23:22:04 2025 daemon.notice hostapd: wls6: DFS-PRE-CAC-EXPIRED freq=5640 ht_enabled=0 chan_offset=0 chan_width=0 cf1=5640 cf2=0
Sat May 17 23:22:12 2025 daemon.notice hostapd: wls5: ACS-COMPLETED freq=2437 channel=6
Sat May 17 23:22:12 2025 daemon.notice hostapd: wls5: interface state ACS->HT_SCAN
Sat May 17 23:22:14 2025 daemon.notice hostapd: 20/40 MHz operation not permitted on channel pri=6 sec=10 based on overlapping BSSes
Sat May 17 23:22:14 2025 daemon.notice hostapd: Fallback to 20 MHz
Sat May 17 23:22:14 2025 kern.info kernel: [817158.270543] br-lan: port 11(wls5) entered blocking state
Sat May 17 23:22:14 2025 kern.info kernel: [817158.270595] br-lan: port 11(wls5) entered forwarding state
Sat May 17 23:22:14 2025 daemon.notice netifd: Network device 'wls5' link is up
Sat May 17 23:22:14 2025 daemon.notice hostapd: wls5: interface state HT_SCAN->ENABLED
Sat May 17 23:22:14 2025 daemon.notice hostapd: wls5: AP-ENABLED
Sat May 17 23:22:18 2025 daemon.info hostapd: wls5: STA b8:4d:43:9a:68:2f IEEE 802.11: authenticated
Sat May 17 23:22:18 2025 daemon.info hostapd: wls5: STA b8:4d:43:9a:68:2f IEEE 802.11: associated (aid 1)
Sat May 17 23:22:18 2025 daemon.notice hostapd: wls5: AP-STA-CONNECTED b8:4d:43:9a:68:2f auth_alg=open
Sat May 17 23:22:18 2025 daemon.info hostapd: wls5: STA b8:4d:43:9a:68:2f WPA: pairwise key handshake completed (RSN)
Sat May 17 23:22:18 2025 daemon.notice hostapd: wls5: EAPOL-4WAY-HS-COMPLETED b8:4d:43:9a:68:2f
Sat May 17 23:22:18 2025 daemon.info dnsmasq-dhcp[1]: DHCPDISCOVER(br-lan) b8:4d:43:9a:68:2f
Sat May 17 23:22:18 2025 daemon.info dnsmasq-dhcp[1]: DHCPOFFER(br-lan) 192.168.3.195 b8:4d:43:9a:68:2f
Sat May 17 23:22:18 2025 daemon.info dnsmasq-dhcp[1]: DHCPREQUEST(br-lan) 192.168.3.195 b8:4d:43:9a:68:2f
Sat May 17 23:22:18 2025 daemon.info dnsmasq-dhcp[1]: DHCPACK(br-lan) 192.168.3.195 b8:4d:43:9a:68:2f dreame_vacuum_p2150a
Sat May 17 23:22:22 2025 daemon.info hostapd: wls5: STA 68:76:27:34:7b:e5 IEEE 802.11: authenticated
Sat May 17 23:22:22 2025 daemon.info hostapd: wls5: STA 68:76:27:34:82:5e IEEE 802.11: authenticated
Sat May 17 23:22:22 2025 daemon.info hostapd: wls5: STA 68:76:27:34:7b:e5 IEEE 802.11: associated (aid 2)
Sat May 17 23:22:22 2025 daemon.info hostapd: wls5: STA 68:76:27:34:82:5e IEEE 802.11: associated (aid 3)
Sat May 17 23:22:22 2025 daemon.notice hostapd: wls5: AP-STA-CONNECTED 68:76:27:34:7b:e5 auth_alg=open
Sat May 17 23:22:22 2025 daemon.info hostapd: wls5: STA 68:76:27:34:7b:e5 WPA: pairwise key handshake completed (RSN)
Sat May 17 23:22:22 2025 daemon.notice hostapd: wls5: EAPOL-4WAY-HS-COMPLETED 68:76:27:34:7b:e5
Sat May 17 23:22:22 2025 daemon.notice hostapd: wls5: AP-STA-CONNECTED 68:76:27:34:82:5e auth_alg=open
Sat May 17 23:22:22 2025 daemon.info hostapd: wls5: STA 68:76:27:34:82:5e WPA: pairwise key handshake completed (RSN)
Sat May 17 23:22:22 2025 daemon.notice hostapd: wls5: EAPOL-4WAY-HS-COMPLETED 68:76:27:34:82:5e
Sat May 17 23:22:23 2025 daemon.info hostapd: wls5: STA 26:37:ca:32:10:7f IEEE 802.11: authenticated
Sat May 17 23:22:23 2025 daemon.info hostapd: wls5: STA 26:37:ca:32:10:7f IEEE 802.11: associated (aid 4)
Sat May 17 23:22:23 2025 daemon.notice hostapd: wls5: AP-STA-CONNECTED 26:37:ca:32:10:7f auth_alg=open
Sat May 17 23:22:23 2025 daemon.info hostapd: wls5: STA 26:37:ca:32:10:7f WPA: pairwise key handshake completed (RSN)
Sat May 17 23:22:23 2025 daemon.notice hostapd: wls5: EAPOL-4WAY-HS-COMPLETED 26:37:ca:32:10:7f
[... cut ...]
wireless.radio1.path='platform/soc@0/c000000.wifi+1'
wireless.radio1.band='2g'
wireless.radio1.channel='auto'
wireless.radio1.htmode='HE40'
wireless.radio1.legacy_rates='1'
wireless.radio1.cell_density='0'
wireless.radio1.country='US'
wireless.default_radio1=wifi-iface
wireless.default_radio1.device='radio1'
wireless.default_radio1.network='lan'
wireless.default_radio1.mode='ap'
wireless.default_radio1.ssid='RiddleWrt5'
wireless.default_radio1.encryption='psk-mixed'
wireless.default_radio1.ifname='wls5'
wireless.default_radio1.key='***'
```

Djfe, 2025-05-17:
does 5GHz work with wireless.radio0.htmode='HE20' and wireless.radio0.channel='36' ?
Does it stay relatively fast then? If it does we can check further configs. Channel 36 with 20, 40 or 80MHz doesn't require DFS. This is what I'm testing for.

irrelevant to your issue, but I suggest setting wireless.radio1.legacy_rates='0' (1 would allow legacy 802.11b on 2.4GHz which does not make sense anymore nowadays)

why do you name your 2.4GHz network "RiddleWrt5" like it's the 5GHz one? ^^

RiddlerX2, 2025-05-18:
Yes 5GHz works at 36 channel and HE20/40/80 mode, but still on downgraded speed (about 15-20 Mbit at any mode). On HE160 5GHz disappears.
And I have confused what I see, looks like device (mobile phone) can't negotiate speed up and always stay on minimum.

[image]

About legacy_rates, looks like it was enabled by accident. Disable there is nothing changed in environment.
About naming I'm using Wi-Fi versions (5 = old) as an anchor, steel have devices which able to use 802.11n/ac only, but may be this is another mistake about Wi-Fi understanding.

While experimenting found if disable wmm `wireless.default_radio[0-1].wmm='0'`, speed always reduced. But if enable then speed become normal even on 5GHz up to 80MHz bandwidth.

So about speed limit it was misconfiguration. Steel have problem with HE160 mode only.

Djfe, 2025-05-25:
Are you certain your wifi client not be cause of the issue?
Maybe your client doesn't support 160MHz well 🤔
Just a last idea I have.