openwrt/openwrt #18692: realtek: XikeStor SKS8300-8X port definition

plappermaul, 2025-05-03:
I'm planning to enhance the realtek MDIO codig. Looking at the [XikeStor SKS8300-8X DTS](https://git.openwrt.org/?p=openwrt/openwrt.git;a=blob;f=target/linux/realtek/dts/rtl9303_xikestor_sks8300-8x.dts;h=e62636776b77973b7804240486d8427990e891df;hb=HEAD). I'm wondering why the mdio ports have so strange register numbers? From my understanding the rule is:

- Look into SerDes registers of <sds> if phy-is-integrated
- Otherwise look into the MDIO bus

@musashino205 Why did you decide on this reg = 0, 8, 16 on the mdio bus definition?

@janh How does the MIB reading (e.g. your last patches) find the right internal port numbers for an interface? From the mdio-bus numbering or from the switch/ports numbering?

Why am I asking? The idea to split the mdio bus register readings into two sections:

- 0x00-0x3f read from bus
- 0x40-... read from sds <nr-64>

for that model it would boild down to:

```
&ethernet0 {
  mdio: mdio-bus {
  compatible = "realtek,rtl838x-mdio";
  regmap = <&ethernet0>;
  #address-cells = <1>;
  #size-cells = <0>;
  
  phy0: ethernet-phy@0 {
    compatible = "ethernet-phy-ieee802.3-c22";
    reg = <0x42>;
  };
```

musashino205, 2025-05-03:
> @musashino205 Why did you decide on this reg = 0, 8, 16 on the mdio bus definition?

They are the base ports of XSGMII on RTL9303.

https://svanheule.net/switches/rtl93xx#section9303

plappermaul, 2025-05-03:
Understood. So you simply chose the "would-be-external-phy-address". Can you do me a favour and check if changing the first three bus ports to & registers to 1, 7 and 18 (0x12) will work too?

janh, 2025-05-03:
> [@janh](https://github.com/janh) How does the MIB reading (e.g. your last patches) find the right internal port numbers for an interface? From the mdio-bus numbering or from the switch/ports numbering?

It uses the port index from the DSA callback functions directly.

musashino205, 2025-05-05:
> Can you do me a favour and check if changing the first three bus ports to & registers to 1, 7 and 18 (0x12) will work too?

Tested and it works.

musashino205, 2025-05-05:
> The idea to split the mdio bus register readings into two sections:
>
> 0x00-0x3f read from bus
> 0x40-... read from sds

IMO, I don't like using values ​​that are not related to the base port or the index of SerDes and are not significant.

---
BTW, this is not a bug, please discuss at the Forum.

plappermaul, 2025-05-05:
Thanks for your help.

closed 2025-05-05 by plappermaul