openwrt/openwrt #18693: BPI-R4, wifi got worse with the latest snapshot. [r29440-bf91381451]

relink2013, 2025-05-03:
I just tried the latest snapshot build of OpenWRT [27dd2cb3114ba03f219c8d80ca58454caee8b7af72bc9a01c3205fc3834db023] and somehow the WiFi has actually gotten worse. 

In the previous snapshots, and in 24.10.1 (r28597-0425664679), I was able to select upto 7 dBm transmit power, now im limited to 6 dBm, either way it should be 22-23 dBm. Speeds seem stable, but only if I am a few ft away, 15-20ft away the network disappears entirely (Tested with iPhone 15 Pro and iPhone 16). 6GHz is completely disabled, and is not even an option on any of the radios. 

Everything appears to work fine on MTK builds so it's not a hardware issue. 

Did a reboot, immediately disabled all radios, and reenabled all radios. [Logs are here](https://pastebin.com/AcsBdD29)

kirdesde, 2025-05-03:
This is not a bug in OpenWrt nor the mt76 driver.

There are faulty BE14 boards those missing the proper eeprom values, that's why (MTK builds are using a default eeprom file as a fallback)

My BE14 has the right eeprom values and is working fine with OpenWrt.

Issue needs to be discussed with Sinvoip as manufacturer.

damnms, 2025-05-03:
afaik this should be fixed in master, see https://github.com/openwrt/openwrt/issues/17489

kirdesde, 2025-05-03:
the actual PR is still open:

https://github.com/openwrt/mt76/pull/954

damnms, 2025-05-03:
you are right, thought thats already in master.
i remove the patch and see if it still works (on my machine only, not sure if i was affected at all) with the latest firmware bin

for reference: https://forum.banana-pi.org/t/be1400-terrible-range-slow-speeds-and-one-of-the-radios-is-disabled-this-is-with-only-a-single-device-connected/23153/22

relink2013, 2025-05-03:
> This is not a bug in OpenWrt nor the mt76 driver.
> There are faulty BE14 boards those missing the proper eeprom values

Ooof, Ok. That would explain why I keep seeing such wildly different performance claims.

> the actual PR is still open:

I did a quick read through of the issue regarding the tx_power check. Even in that single issue there are two different patches mentioned with differing results. 

Regardless, this 100% sounds like the issue I'm having, if the PR is still open then that means it's not in the snapshot builds yet right?

If anyone has compiled an image with the patches I would be thrilled to test it out and report back. If not, then what would I need to do in order to compile it myself with the patches applied? Doing that would be completely new territory for me, and I wouldn't even begin to know where to start.

I apologize if there are any typos, I've been up for 24hr straight lol.

damnms, 2025-05-03:
@relink2013 as frank-w mentioned, simplest solution is probably to just update the firmware file.

download https://drive.google.com/file/d/1qneVF1a6ZGMDoLLPpQw2LPYZsRKXto1T/view?usp=sharing and save it to /lib/firmware/mediatek/mt7996/mt7996_eeprom_233_2i5i6i.bin

PentiumB, 2025-05-03:
@kirdesde 

> There are faulty BE14 boards those missing the proper eeprom values, that's why (MTK builds are using a default eeprom file as a fallback)
> 
> My BE14 has the right eeprom values and is working fine with OpenWrt.
> 
> Issue needs to be discussed with Sinvoip as manufacturer.

This is fixed by the openwrt patch. And when you try to tell the seller that this is an arbitrary defect, they simply send a [link with information about this patch](https://docs.banana-pi.org/en/BPI-R4/GettingStarted_BPI-R4#_improve_the_performance_of_radio_frequency_under_openwrt24_10_firmware).

relink2013, 2025-05-03:
> [@relink2013](https://github.com/relink2013) as frank-w mentioned, simplest solution is probably to just update the firmware file.
> 
> download https://drive.google.com/file/d/1qneVF1a6ZGMDoLLPpQw2LPYZsRKXto1T/view?usp=sharing and save it to /lib/firmware/mediatek/mt7996/mt7996_eeprom_233_2i5i6i.bin, then reboot

I just saw his comment on the bpi thread. I applied the patch and rebooted but im still stuck at 6 dBm for some reason. 

```
root@OpenWrt:~# ls -lh /lib/firmware/mediatek/mt7996/
-rw-r--r--    1 root     root      187.2K May  2 17:30 mt7996_dsp.bin
-rw-r--r--    1 root     root        7.5K May  2 17:30 mt7996_eeprom.bin
-rw-r--r--    1 root     root        7.5K May  2 17:30 mt7996_eeprom_233.bin
-rw-r--r--    1 root     root        7.5K May  3 20:16 mt7996_eeprom_233_2i5i6i.bin
-rw-r--r--    1 root     root        7.5K May  2 17:30 mt7996_eeprom_2i5i6i.bin
-rw-r--r--    1 root     root       23.4K May  2 17:30 mt7996_rom_patch.bin
-rw-r--r--    1 root     root       23.4K May  2 17:30 mt7996_rom_patch_233.bin
-rw-r--r--    1 root     root      504.2K May  2 17:30 mt7996_wa.bin
-rw-r--r--    1 root     root      504.2K May  2 17:30 mt7996_wa_233.bin
-rw-r--r--    1 root     root        2.5M May  2 17:30 mt7996_wm.bin
-rw-r--r--    1 root     root        2.5M May  2 17:30 mt7996_wm_233.bin
``` 
I've tried restarting the radios, disabling and then re-enabling them, tried changing the setting and changing it back. Nothing so far has made the additional power levels appear as options. 

Im going to also read more of the new comments on the bpi thread.

ptpt52, 2025-05-04:
ping @nbd168

relink2013, 2025-05-05:
> ping [@nbd168](https://github.com/nbd168)

Sorry, im not sure what you mean?

relink2013, 2025-05-10:
Is anyone aware of any progress for the wifi issues? I haven't found much on the bpi forums or anything else so far.

danpawlik, 2025-05-11:
Is it not duplication of https://github.com/openwrt/openwrt/issues/17489 ?

CAMOBAP, 2025-11-09:
Is this issue still there?