openwrt/openwrt #18703: [24.10] mt7981, mt7986: slow TX with batman-adv wireless mesh

csharper2005, 2025-05-04:
### Describe the bug

Slow TX speed (2-3 Mbps) on mt7981 or mt7986 batman-adv wireless mesh nodes. Good speed on ath79 or mt7621 mesh nodes. 

5 GHz backhaul, 80 MHz BW, OpenWrt 24.10.1.

An illustration:
[image]

### OpenWrt version

r28597-0425664679

### OpenWrt release

24.10.1

### OpenWrt target/subtarget

mediatek/filogic

### Device

MERCUSYS MR90X v1 (UBI)

### Image kind

Official downloaded image

### Steps to reproduce

1. Set up batman-adv 5 Ghz wireless mesh. At least one of the wireless nodes should be with mt7981 or mt7986 wireless.
2. Run `iperf3 -s` on one of the mesh nodes.
3. Run iperf3 on the different node to test the speed in both directions:
`iperf3 -c <first_node_ip> -P 10`
`iperf3 -c <first_node_ip> -P 10 -R`
4. Make sure that mt7981 or mt7986 has slow TX speed (2-3 Mbps).

### Actual behaviour

1. Slow TX speed (2-3 Mbps) on mt7981 or mt7986 batman-adv wireless mesh nodes.
2. Good TX speed on ath79 or mt7621 mesh nodes.

### Expected behaviour

At least 50-60 Mbps TX speed on mt7981 or mt7986 batman-adv wireless mesh nodes.

### Additional info

batman config:
```
config interface 'bat0'
	option proto 'batadv'
	option routing_algo 'BATMAN_IV'
	option bridge_loop_avoidance '1'
	option gw_mode 'client'
	option hop_penalty '30'
	option defaultroute '0'

config interface 'bat0_5g'
	option proto 'batadv_hardif'
	option master 'bat0'
	option mtu '1536'
	option defaultroute '0'
```
mesh config:
```
config wifi-device 'radio1'
	option type 'mac80211'
	option path 'platform/soc/18000000.wifi+1'
	option band '5g'
	option htmode 'HE80'
	option cell_density '0'
	option country 'DE'
	option channel '36'
	option channels '36 40 44 48'
	option txpower '23'

config wifi-iface 'mesh1'
	option device 'radio1'
	option mode 'mesh'
	option encryption 'sae'
	option mesh_id '5000'
	option mesh_fwding '0'
	option mesh_rssi_threshold '0'
	option mesh_ttl '1'
	option key 'xxxxxxxxxxxx'
	option mcast_rate '24000'
	option network 'bat0_5g'
	option skip_inactivity_poll '1'
	option disassoc_low_ack '0'
```

### Diffconfig

```text

```

### Terms

- [x] I am reporting an issue for OpenWrt, not an unsupported fork.

csharper2005, 2025-05-04:
Station dump from mt7986 node:
```
iw phy1-mesh0 station dump
Station xx:xx:xx:xx:xx:xx (on phy1-mesh0)
        inactive time:  100 ms
        rx bytes:       675871
        rx packets:     3121
        tx bytes:       250163854
        tx packets:     346829
        tx retries:     37832
        tx failed:      37822
        rx drop misc:   177
        signal:         -52 [-54, -56, -64, -63] dBm
        signal avg:     -52 [-54, -55, -64, -63] dBm
        Toffset:        15177895017 us
        tx bitrate:     1170.0 MBit/s VHT-MCS 7 80MHz VHT-NSS 4
        tx duration:    124077529 us
        rx bitrate:     1560.0 MBit/s VHT-MCS 8 80MHz short GI VHT-NSS 4
        rx duration:    41711675 us
        last ack signal:-51 dBm
        avg ack signal: -51 dBm
        airtime weight: 256
        mesh llid:      0
        mesh plid:      0
        mesh plink:     ESTAB
        mesh airtime link metric: 16
        mesh connected to gate: no
        mesh connected to auth server:  no
        mesh local PS mode:     ACTIVE
        mesh peer PS mode:      ACTIVE
        mesh non-peer PS mode:  ACTIVE
        authorized:     yes
        authenticated:  yes
        associated:     yes
        preamble:       long
        WMM/WME:        yes
        MFP:            yes
        TDLS peer:      no
        DTIM period:    2
        beacon interval:100
        connected time: 19205 seconds
        associated at [boottime]:       79374.649s
        associated at:  1746355544086 ms
        current time:   1746374748465 ms
Station xx:xx:xx:xx:xx:xx (on phy1-mesh0)
        inactive time:  10 ms
        rx bytes:       70795
        rx packets:     516
        tx bytes:       5785824
        tx packets:     10732
        tx retries:     1698
        tx failed:      1698
        rx drop misc:   12
        signal:         -45 [-60, -46, -62, -49] dBm
        signal avg:     -44 [-59, -46, -61, -48] dBm
        Toffset:        7145755021 us
        tx bitrate:     1200.9 MBit/s 80MHz HE-MCS 11 HE-NSS 2 HE-GI 0 HE-DCM 0
        tx duration:    11237417 us
        rx bitrate:     6.0 MBit/s 80MHz
        rx duration:    18403360 us
        last ack signal:-46 dBm
        avg ack signal: -46 dBm
        airtime weight: 256
        mesh llid:      0
        mesh plid:      0
        mesh plink:     ESTAB
        mesh airtime link metric: 12
        mesh connected to gate: no
        mesh connected to auth server:  no
        mesh local PS mode:     ACTIVE
        mesh peer PS mode:      ACTIVE
        mesh non-peer PS mode:  ACTIVE
        authorized:     yes
        authenticated:  yes
        associated:     yes
        preamble:       long
        WMM/WME:        yes
        MFP:            yes
        TDLS peer:      no
        DTIM period:    2
        beacon interval:100
        connected time: 11211 seconds
        associated at [boottime]:       87368.846s
        associated at:  1746363538283 ms
        current time:   1746374748466 ms
Station xx:xx:xx:xx:xx:xx (on phy1-mesh0)
        inactive time:  60 ms
        rx bytes:       7462
        rx packets:     57
        tx bytes:       29752
        tx packets:     491
        tx retries:     1
        tx failed:      1
        rx drop misc:   18
        signal:         -47 [-48, -51, -67, -68] dBm
        signal avg:     -47 [-48, -51, -67, -67] dBm
        Toffset:        18446744072652809768 us
        tx bitrate:     270.0 MBit/s MCS 15 40MHz
        tx duration:    112712 us
        rx bitrate:     300.0 MBit/s MCS 15 40MHz short GI
        rx duration:    36737 us
        last ack signal:-46 dBm
        avg ack signal: -45 dBm
        airtime weight: 256
        mesh llid:      0
        mesh plid:      0
        mesh plink:     ESTAB
        mesh airtime link metric: 77
        mesh connected to gate: no
        mesh connected to auth server:  no
        mesh local PS mode:     ACTIVE
        mesh peer PS mode:      ACTIVE
        mesh non-peer PS mode:  ACTIVE
        authorized:     yes
        authenticated:  yes
        associated:     yes
        preamble:       long
        WMM/WME:        yes
        MFP:            yes
        TDLS peer:      no
        DTIM period:    2
        beacon interval:100
        connected time: 3003 seconds
        associated at [boottime]:       95592.757s
        associated at:  1746371762195 ms
        current time:   1746374748466 ms
```

csharper2005, 2025-05-04:
Low TX on mt7986 node (iperf3 is sending but TX rate is 6 Mbit/s only):

[image]

csharper2005, 2025-05-05:
> what about use a fakemesh (`AP(wds)` + `STA(wds)`)? Are there performance issues as well?

No any issues with `wds` or pure `802.11s` setup (speed is hundreds of Mbit/sec).

ptpt52, 2025-05-06:
ping @nbd168

lorand-horvath, 2025-05-06:
@csharper2005 Have you tried one of the latest snapshots instead of 24.10.1?

csharper2005, 2025-05-06:
> [@csharper2005](https://github.com/csharper2005) Have you tried one of the latest snapshots instead of 24.10.1?

The issue exists in SNAPSHOT. I also tried the newest MediaTek firmware dated 2024.12. The same result.

Just a guess - with new firmware `batman-adv` frames (`ethertype Unknown (0x4305)`) are unable to wake up the radio from power saving mode.

```
iperf3 -c 192.168.204.131 -P 5
Connecting to host 192.168.204.131, port 5201
[  5] local 192.168.204.132 port 43086 connected to 192.168.204.131 port 5201
[  7] local 192.168.204.132 port 43096 connected to 192.168.204.131 port 5201
[  9] local 192.168.204.132 port 43112 connected to 192.168.204.131 port 5201
[ 11] local 192.168.204.132 port 43116 connected to 192.168.204.131 port 5201
[ 13] local 192.168.204.132 port 43120 connected to 192.168.204.131 port 5201
[ ID] Interval           Transfer     Bitrate         Retr  Cwnd
[  5]   0.00-1.00   sec   128 KBytes  1.05 Mbits/sec    2   18.4 KBytes
[  7]   0.00-1.00   sec  0.00 Bytes  0.00 bits/sec    6   11.3 KBytes
[  9]   0.00-1.00   sec  0.00 Bytes  0.00 bits/sec    5   11.3 KBytes
[ 11]   0.00-1.00   sec  0.00 Bytes  0.00 bits/sec    6   11.3 KBytes
[ 13]   0.00-1.00   sec  0.00 Bytes  0.00 bits/sec    7   2.83 KBytes
[SUM]   0.00-1.00   sec   128 KBytes  1.05 Mbits/sec   26
- - - - - - - - - - - - - - - - - - - - - - - - -
[  5]   1.00-2.00   sec  0.00 Bytes  0.00 bits/sec    9   12.7 KBytes
[  7]   1.00-2.00   sec  0.00 Bytes  0.00 bits/sec    3   5.66 KBytes
[  9]   1.00-2.00   sec  0.00 Bytes  0.00 bits/sec    3   4.24 KBytes
[ 11]   1.00-2.00   sec  0.00 Bytes  0.00 bits/sec    4   4.24 KBytes
[ 13]   1.00-2.00   sec  0.00 Bytes  0.00 bits/sec    0   5.66 KBytes
[SUM]   1.00-2.00   sec  0.00 Bytes  0.00 bits/sec   19
- - - - - - - - - - - - - - - - - - - - - - - - -
[  5]   2.00-3.00   sec  0.00 Bytes  0.00 bits/sec    3   8.48 KBytes
[  7]   2.00-3.00   sec   128 KBytes  1.05 Mbits/sec    2   5.66 KBytes
[  9]   2.00-3.00   sec   128 KBytes  1.05 Mbits/sec    2   4.24 KBytes
[ 11]   2.00-3.00   sec   128 KBytes  1.05 Mbits/sec    1   2.83 KBytes
[ 13]   2.00-3.00   sec  0.00 Bytes  0.00 bits/sec    2   4.24 KBytes
[SUM]   2.00-3.00   sec   384 KBytes  3.15 Mbits/sec   10
- - - - - - - - - - - - - - - - - - - - - - - - -
[  5]   3.00-4.00   sec  0.00 Bytes  0.00 bits/sec    3   5.66 KBytes
[  7]   3.00-4.00   sec  0.00 Bytes  0.00 bits/sec    2   2.83 KBytes
[  9]   3.00-4.00   sec  0.00 Bytes  0.00 bits/sec    1   4.24 KBytes
[ 11]   3.00-4.00   sec  0.00 Bytes  0.00 bits/sec    1   2.83 KBytes
[ 13]   3.00-4.00   sec   128 KBytes  1.05 Mbits/sec    1   4.24 KBytes
[SUM]   3.00-4.00   sec   128 KBytes  1.05 Mbits/sec    8
- - - - - - - - - - - - - - - - - - - - - - - - -
[  5]   4.00-5.00   sec   128 KBytes  1.05 Mbits/sec    2   4.24 KBytes
[  7]   4.00-5.00   sec  0.00 Bytes  0.00 bits/sec    1   4.24 KBytes
[  9]   4.00-5.00   sec  0.00 Bytes  0.00 bits/sec    1   4.24 KBytes
[ 11]   4.00-5.00   sec  0.00 Bytes  0.00 bits/sec    0   4.24 KBytes
[ 13]   4.00-5.00   sec  0.00 Bytes  0.00 bits/sec    2   2.83 KBytes
[SUM]   4.00-5.00   sec   128 KBytes  1.05 Mbits/sec    6
- - - - - - - - - - - - - - - - - - - - - - - - -
[  5]   5.00-6.00   sec  0.00 Bytes  0.00 bits/sec    1   4.24 KBytes
[  7]   5.00-6.00   sec  0.00 Bytes  0.00 bits/sec    1   4.24 KBytes
[  9]   5.00-6.00   sec  0.00 Bytes  0.00 bits/sec    1   4.24 KBytes
[ 11]   5.00-6.00   sec  0.00 Bytes  0.00 bits/sec    1   2.83 KBytes
[ 13]   5.00-6.00   sec  0.00 Bytes  0.00 bits/sec    2   2.83 KBytes
[SUM]   5.00-6.00   sec  0.00 Bytes  0.00 bits/sec    6
- - - - - - - - - - - - - - - - - - - - - - - - -
[  5]   6.00-7.00   sec  0.00 Bytes  0.00 bits/sec    2   2.83 KBytes
[  7]   6.00-7.00   sec  0.00 Bytes  0.00 bits/sec    1   4.24 KBytes
[  9]   6.00-7.00   sec  0.00 Bytes  0.00 bits/sec    1   2.83 KBytes
[ 11]   6.00-7.00   sec  0.00 Bytes  0.00 bits/sec    1   5.66 KBytes
[ 13]   6.00-7.00   sec  0.00 Bytes  0.00 bits/sec    0   5.66 KBytes
[SUM]   6.00-7.00   sec  0.00 Bytes  0.00 bits/sec    5
- - - - - - - - - - - - - - - - - - - - - - - - -
[  5]   7.00-8.00   sec  0.00 Bytes  0.00 bits/sec    2   2.83 KBytes
[  7]   7.00-8.00   sec  0.00 Bytes  0.00 bits/sec    1   2.83 KBytes
[  9]   7.00-8.00   sec  0.00 Bytes  0.00 bits/sec    1   2.83 KBytes
[ 11]   7.00-8.00   sec  0.00 Bytes  0.00 bits/sec    2   4.24 KBytes
[ 13]   7.00-8.00   sec  0.00 Bytes  0.00 bits/sec    2   2.83 KBytes
[SUM]   7.00-8.00   sec  0.00 Bytes  0.00 bits/sec    8
[... 6 lines cut ...]
[SUM]   8.00-9.00   sec  0.00 Bytes  0.00 bits/sec    4
- - - - - - - - - - - - - - - - - - - - - - - - -
[  5]   9.00-10.00  sec  0.00 Bytes  0.00 bits/sec    1   4.24 KBytes
[  7]   9.00-10.00  sec   128 KBytes  1.05 Mbits/sec    0   4.24 KBytes
[  9]   9.00-10.00  sec  0.00 Bytes  0.00 bits/sec    1   2.83 KBytes
[ 11]   9.00-10.00  sec   128 KBytes  1.05 Mbits/sec    2   4.24 KBytes
[ 13]   9.00-10.00  sec   128 KBytes  1.05 Mbits/sec    1   5.66 KBytes
[SUM]   9.00-10.00  sec   384 KBytes  3.14 Mbits/sec    5
- - - - - - - - - - - - - - - - - - - - - - - - -
[ ID] Interval           Transfer     Bitrate         Retr
[  5]   0.00-10.00  sec   256 KBytes   210 Kbits/sec   26            sender
[  5]   0.00-10.09  sec   128 KBytes   104 Kbits/sec                  receiver
[  7]   0.00-10.00  sec   256 KBytes   210 Kbits/sec   18            sender
[  7]   0.00-10.09  sec   128 KBytes   104 Kbits/sec                  receiver
[... cut ...]
[  9]   0.00-10.09  sec   128 KBytes   104 Kbits/sec                  receiver
[ 11]   0.00-10.00  sec   256 KBytes   210 Kbits/sec   19            sender
[ 11]   0.00-10.09  sec   128 KBytes   104 Kbits/sec                  receiver
[ 13]   0.00-10.00  sec   256 KBytes   210 Kbits/sec   17            sender
[ 13]   0.00-10.09  sec   128 KBytes   104 Kbits/sec                  receiver
[SUM]   0.00-10.00  sec  1.12 MBytes   944 Kbits/sec   97             sender
[SUM]   0.00-10.09  sec   640 KBytes   520 Kbits/sec                  receiver

iperf Done.
```

andrewjlamarche, 2025-05-08:
FWIW, I have a B.A.T.M.A.N mesh across 2 MT7986 with poor signal on 24.10.1. Seeing >200mbps still.

lorand-horvath, 2025-05-08:
Not sure if it would help, but I've just noticed there's an update for the `mt7986-wo-firmware` package in the latest snapshot:

[image]

csharper2005, 2025-05-08:
@lorand-horvath it looks like the fw version is the same:
`DEV_00000020221012175032` - snapshot 
`DEV_00000020221012175032` - 24.10.1

@andrewjlamarche can you share your wireless mesh config and batman interfaces network config, please?

lorand-horvath, 2025-05-08:
@csharper2005 How can I get the fw version as you did?

csharper2005, 2025-05-08:
> [@csharper2005](https://github.com/csharper2005) How can I get the fw version as you did?

You can open the file in any hex or even text editor. Fw version is at the end of the file.

[image]

lorand-horvath, 2025-05-08:
It looks like I have the same firmware version as well (DEV_00000020221012175032) in the last 2 snapshots 29467 and 29487. Nothing has changed, the files inside `/lib/firmware/mediatek` are exactly the same - I checked with file comparison.

csharper2005, 2025-05-09:
And even with workaround applied the speed between wireless nodes doesn't exceed 50 Mbits/sec. Server - mt7986, client - mt7981. And it doesn't depend on the number of iperf3 threads.
```
root@ax3000:~# iperf3 -c 192.168.204.131 -P 10
Connecting to host 192.168.204.131, port 5201
[  5] local 192.168.204.132 port 58914 connected to 192.168.204.131 port 5201
[  7] local 192.168.204.132 port 58916 connected to 192.168.204.131 port 5201
[  9] local 192.168.204.132 port 58928 connected to 192.168.204.131 port 5201
[ 11] local 192.168.204.132 port 58932 connected to 192.168.204.131 port 5201
[ 13] local 192.168.204.132 port 58946 connected to 192.168.204.131 port 5201
[ 15] local 192.168.204.132 port 58956 connected to 192.168.204.131 port 5201
[ 17] local 192.168.204.132 port 58968 connected to 192.168.204.131 port 5201
[ 19] local 192.168.204.132 port 58982 connected to 192.168.204.131 port 5201
[ 21] local 192.168.204.132 port 58988 connected to 192.168.204.131 port 5201
[ 23] local 192.168.204.132 port 58994 connected to 192.168.204.131 port 5201
[ ID] Interval           Transfer     Bitrate         Retr  Cwnd
[  5]   0.00-1.00   sec   768 KBytes  6.28 Mbits/sec    0   59.4 KBytes
[  7]   0.00-1.00   sec   768 KBytes  6.28 Mbits/sec    0   59.4 KBytes
[  9]   0.00-1.00   sec   640 KBytes  5.24 Mbits/sec    0   56.6 KBytes
[ 11]   0.00-1.00   sec   640 KBytes  5.24 Mbits/sec    0   58.0 KBytes
[ 13]   0.00-1.00   sec   640 KBytes  5.24 Mbits/sec    0   58.0 KBytes
[ 15]   0.00-1.00   sec   640 KBytes  5.24 Mbits/sec    0   60.8 KBytes
[ 17]   0.00-1.00   sec   640 KBytes  5.24 Mbits/sec    0   56.6 KBytes
[ 19]   0.00-1.00   sec   640 KBytes  5.24 Mbits/sec    0   52.3 KBytes
[ 21]   0.00-1.00   sec   640 KBytes  5.24 Mbits/sec    0   58.0 KBytes
[ 23]   0.00-1.00   sec   640 KBytes  5.24 Mbits/sec    0   58.0 KBytes
[SUM]   0.00-1.00   sec  6.50 MBytes  54.5 Mbits/sec    0
- - - - - - - - - - - - - - - - - - - - - - - - -
[  5]   1.00-2.00   sec   256 KBytes  2.10 Mbits/sec    1   67.9 KBytes
[  7]   1.00-2.00   sec   384 KBytes  3.15 Mbits/sec    0   72.1 KBytes
[  9]   1.00-2.00   sec   384 KBytes  3.15 Mbits/sec    1   69.3 KBytes
[ 11]   1.00-2.00   sec   384 KBytes  3.15 Mbits/sec    1   73.5 KBytes
[ 13]   1.00-2.00   sec   384 KBytes  3.15 Mbits/sec    0   63.6 KBytes
[ 15]   1.00-2.00   sec   640 KBytes  5.24 Mbits/sec    0   74.9 KBytes
[ 17]   1.00-2.00   sec   256 KBytes  2.10 Mbits/sec    1   66.5 KBytes
[ 19]   1.00-2.00   sec   256 KBytes  2.10 Mbits/sec    0   60.8 KBytes
[ 21]   1.00-2.00   sec   384 KBytes  3.15 Mbits/sec    1   67.9 KBytes
[ 23]   1.00-2.00   sec   384 KBytes  3.15 Mbits/sec    0   70.7 KBytes
[SUM]   1.00-2.00   sec  3.62 MBytes  30.4 Mbits/sec    5
- - - - - - - - - - - - - - - - - - - - - - - - -
[  5]   2.00-3.00   sec   640 KBytes  5.24 Mbits/sec    0   77.8 KBytes
[  7]   2.00-3.00   sec   512 KBytes  4.19 Mbits/sec    0   77.8 KBytes
[  9]   2.00-3.00   sec   640 KBytes  5.24 Mbits/sec    0   82.0 KBytes
[ 11]   2.00-3.00   sec   768 KBytes  6.29 Mbits/sec    0   83.4 KBytes
[ 13]   2.00-3.00   sec   512 KBytes  4.19 Mbits/sec    0   63.6 KBytes
[ 15]   2.00-3.00   sec   512 KBytes  4.19 Mbits/sec    0   80.6 KBytes
[ 17]   2.00-3.00   sec   640 KBytes  5.24 Mbits/sec    0   72.1 KBytes
[ 19]   2.00-3.00   sec   640 KBytes  5.24 Mbits/sec    0   76.4 KBytes
[ 21]   2.00-3.00   sec   640 KBytes  5.24 Mbits/sec    0   83.4 KBytes
[ 23]   2.00-3.00   sec   768 KBytes  6.29 Mbits/sec    0   83.4 KBytes
[SUM]   2.00-3.00   sec  6.12 MBytes  51.4 Mbits/sec    0
- - - - - - - - - - - - - - - - - - - - - - - - -
[  5]   3.00-4.00   sec   384 KBytes  3.15 Mbits/sec    0   77.8 KBytes
[  7]   3.00-4.00   sec   384 KBytes  3.15 Mbits/sec    0   77.8 KBytes
[  9]   3.00-4.00   sec   768 KBytes  6.29 Mbits/sec    0   82.0 KBytes
[ 11]   3.00-4.00   sec   768 KBytes  6.29 Mbits/sec    0   97.6 KBytes
[ 13]   3.00-4.00   sec   512 KBytes  4.19 Mbits/sec    0   63.6 KBytes
[ 15]   3.00-4.00   sec   512 KBytes  4.19 Mbits/sec    0   80.6 KBytes
[ 17]   3.00-4.00   sec   384 KBytes  3.15 Mbits/sec    0   72.1 KBytes
[ 19]   3.00-4.00   sec   384 KBytes  3.15 Mbits/sec    0   79.2 KBytes
[ 21]   3.00-4.00   sec   896 KBytes  7.34 Mbits/sec    0   97.6 KBytes
[ 23]   3.00-4.00   sec   640 KBytes  5.24 Mbits/sec    0   89.1 KBytes
[SUM]   3.00-4.00   sec  5.50 MBytes  46.1 Mbits/sec    0
- - - - - - - - - - - - - - - - - - - - - - - - -
[  5]   4.00-5.00   sec   512 KBytes  4.19 Mbits/sec    0   80.6 KBytes
[  7]   4.00-5.00   sec   512 KBytes  4.19 Mbits/sec    0   93.3 KBytes
[  9]   4.00-5.00   sec   512 KBytes  4.19 Mbits/sec    0   82.0 KBytes
[ 11]   4.00-5.00   sec   512 KBytes  4.19 Mbits/sec    0   97.6 KBytes
[ 13]   4.00-5.00   sec   384 KBytes  3.15 Mbits/sec    0   63.6 KBytes
[ 15]   4.00-5.00   sec   768 KBytes  6.29 Mbits/sec    0   80.6 KBytes
[ 17]   4.00-5.00   sec   512 KBytes  4.19 Mbits/sec    0   74.9 KBytes
[... 267 lines cut ...]

iperf Done.
root@ax3000:~# iperf3 -c 192.168.204.131 -P 1 -R
Connecting to host 192.168.204.131, port 5201
Reverse mode, remote host 192.168.204.131 is sending
[  5] local 192.168.204.132 port 34100 connected to 192.168.204.131 port 5201
[ ID] Interval           Transfer     Bitrate
[  5]   0.00-1.00   sec  5.75 MBytes  48.2 Mbits/sec
[  5]   1.00-2.00   sec  6.12 MBytes  51.4 Mbits/sec
[  5]   2.00-3.00   sec  5.25 MBytes  44.0 Mbits/sec
[  5]   3.00-4.00   sec  5.62 MBytes  47.2 Mbits/sec
[... cut ...]
[  5]   5.00-6.00   sec  5.88 MBytes  49.3 Mbits/sec
[  5]   6.00-7.00   sec  6.00 MBytes  50.3 Mbits/sec
[  5]   7.00-8.00   sec  5.88 MBytes  49.3 Mbits/sec
[  5]   8.00-9.00   sec  3.75 MBytes  31.5 Mbits/sec
[  5]   9.00-10.00  sec  5.88 MBytes  49.3 Mbits/sec
- - - - - - - - - - - - - - - - - - - - - - - - -
[ ID] Interval           Transfer     Bitrate         Retr
[  5]   0.00-10.01  sec  58.0 MBytes  48.6 Mbits/sec    1             sender
[  5]   0.00-10.00  sec  56.0 MBytes  47.0 Mbits/sec                  receiver

iperf Done.

```

csharper2005, 2025-05-09:
mt798x is slow even with workaround. TX on ipq8065 is much faster.
[image]

Djfe, 2025-05-17:
maybe you want to remove [24.10] from the issue title since it also affects master

csharper2005, 2025-05-18:
> maybe you want to remove [24.10] from the issue title since it also affects master

Ok. Done.

csharper2005, 2025-05-22:
@andrewjlamarche thank you. Do you use vlans in mesh?

andrewjlamarche, 2025-05-22:
@csharper2005 yes, I have probably 8 or 10.

Djfe, 2025-05-24:
and if you give ath79 or ramips those octets they still mesh just fine? sounds really interesting

csharper2005, 2025-05-24:
Ramips (mt7621 + mt7615n) feels nice too with a new IP.

[image]