openwrt/openwrt #18706: Fritz!Box 7360 v2 version > 22.03.7 resulting in no DSL

Amenia1999, 2025-05-04:
### Describe the bug

I wanted to install the current release, than I figured out the the DSL Firmware will not work until I have a < OpenWRT 22.03.7 release.

### OpenWrt version

22.03.7

### OpenWrt release

22.03.7 r20341-591b7e93d3

### OpenWrt target/subtarget

lantiq/xrx200

### Device

Fritz!Box 7360 v2

### Image kind

Official downloaded image

### Steps to reproduce

Upgrade or clean install from any version higher then 22.03.7 resulting in no DSL activity.

### Actual behaviour

The System looks like there is no firmware for the DSL.

### Expected behaviour

Working DSL and sync

### Additional info

Testet on two identical devices.

### Diffconfig

```text

```

### Terms

- [x] I am reporting an issue for OpenWrt, not an unsupported fork.

aimacintyre, 2025-05-06:
24.10.x and 23.05.x include exactly the same firmware as included in 22.03.7, and these devices have mature support.  Have you attempted to raise this in the OpenWrt forums where you're more likely to get help?

At a minimum you will need to add to this issue a copy of the system log for an affected version showing the startup of the device and any attempts for the xDSL function to connect to your ISP.  NB: obscure any passwords or WAN IP addresses in the log before uploading!

Amenia1999, 2025-05-06:
Ohh, noo, noo. I dont mean the included Firmware, I mean the orginal VDS Firmware what is not included and you need to extract it from the original Fritz!Box OS.

aimacintyre, 2025-05-07:
> I mean the orginal VDSL Firmware what is not included and you need to extract it from the original Fritz!Box OS.

Then you also need to show where and/or how you have installed this external firmware on the device.  Attaching a copy of /etc/config/network would also be helpful (again, please obscure any passwords or WAN IP addresses before attaching).

You might also find some useful info in [this forum thread](https://forum.openwrt.org/t/need-help-for-fritzbox-7362-sl-dsl-modem-driver-vr9-dsl-bin/182293), even though it is for a different AVM device (but the same xDSL modem chipset).

Amenia1999, 2025-05-15:
I took the Firmwarefile from a working Freetz Version of the same Fritz Box, transferd it via SSH.

This is the log from the working Version OpenWrt 22.03.7 r20341-591b7e93d3:

Kernel: 
```
[    0.000000] Linux version 5.10.221 (builder@buildhost) (mips-openwrt-linux-musl-gcc (OpenWrt GCC 11.2.0 r20341-591b7e93d3) 11.2.0, GNU ld (GNU Binutils) 2.37) #0 SMP Mon Jul 15 22:25:54 2024
[    0.000000] SoC: xRX200 rev 1.2
[    0.000000] printk: bootconsole [early0] enabled
[    0.000000] CPU0 revision is: 00019556 (MIPS 34Kc)
[    0.000000] MIPS: machine is AVM FRITZ!Box 7360 V2
[    0.000000] Initrd not found or empty - disabling initrd
[    0.000000] Detected 1 available secondary CPU(s)
[    0.000000] Primary instruction cache 32kB, VIPT, 4-way, linesize 32 bytes.
[    0.000000] Primary data cache 32kB, 4-way, VIPT, cache aliases, linesize 32 bytes
[    0.000000] Zone ranges:
[    0.000000]   Normal   [mem 0x0000000000000000-0x0000000007ffffff]
[    0.000000] Movable zone start for each node
[    0.000000] Early memory node ranges
[    0.000000]   node   0: [mem 0x0000000000000000-0x0000000007ffffff]
[    0.000000] Initmem setup node 0 [mem 0x0000000000000000-0x0000000007ffffff]
[    0.000000] On node 0 totalpages: 32768
[    0.000000]   Normal zone: 288 pages used for memmap
[    0.000000]   Normal zone: 0 pages reserved
[    0.000000]   Normal zone: 32768 pages, LIFO batch:7
[    0.000000] percpu: Embedded 14 pages/cpu s26800 r8192 d22352 u57344
[    0.000000] pcpu-alloc: s26800 r8192 d22352 u57344 alloc=14*4096
[    0.000000] pcpu-alloc: [0] 0 [0] 1 
[    0.000000] Built 1 zonelists, mobility grouping on.  Total pages: 32480
[    0.000000] Kernel command line: console=ttyLTQ0,115200
[    0.000000] Dentry cache hash table entries: 16384 (order: 4, 65536 bytes, linear)
[    0.000000] Inode-cache hash table entries: 8192 (order: 3, 32768 bytes, linear)
[    0.000000] Writing ErrCtl register=0002808a
[    0.000000] Readback ErrCtl register=0002808a
[    0.000000] mem auto-init: stack:off, heap alloc:off, heap free:off
[    0.000000] Memory: 119448K/131072K available (6982K kernel code, 628K rwdata, 1112K rodata, 1244K init, 213K bss, 11624K reserved, 0K cma-reserved)
[    0.000000] SLUB: HWalign=32, Order=0-3, MinObjects=0, CPUs=2, Nodes=1
[    0.000000] rcu: Hierarchical RCU implementation.
[    0.000000] 	Tracing variant of Tasks RCU enabled.
[    0.000000] rcu: RCU calculated value of scheduler-enlistment delay is 10 jiffies.
[    0.000000] NR_IRQS: 256
[    0.000000] CPU Clock: 500MHz
[    0.000000] clocksource: MIPS: mask: 0xffffffff max_cycles: 0xffffffff, max_idle_ns: 7645041786 ns
[    0.000012] sched_clock: 32 bits at 250MHz, resolution 4ns, wraps every 8589934590ns
[    0.008006] Calibrating delay loop... 331.77 BogoMIPS (lpj=1658880)
[    0.074082] pid_max: default: 32768 minimum: 301
[    0.079005] Mount-cache hash table entries: 1024 (order: 0, 4096 bytes, linear)
[    0.086200] Mountpoint-cache hash table entries: 1024 (order: 0, 4096 bytes, linear)
[    0.098478] rcu: Hierarchical SRCU implementation.
[    0.103734] dyndbg: Ignore empty _ddebug table in a CONFIG_DYNAMIC_DEBUG_CORE build
[    0.111835] smp: Bringing up secondary CPUs ...
[    0.117808] Primary instruction cache 32kB, VIPT, 4-way, linesize 32 bytes.
[    0.117825] Primary data cache 32kB, 4-way, VIPT, cache aliases, linesize 32 bytes
[    0.117922] CPU1 revision is: 00019556 (MIPS 34Kc)
[    0.171558] Synchronize counters for CPU 1: done.
[    0.195870] smp: Brought up 1 node, 2 CPUs
[    0.208293] clocksource: jiffies: mask: 0xffffffff max_cycles: 0xffffffff, max_idle_ns: 19112604462750000 ns
[    0.218066] futex hash table entries: 512 (order: 2, 16384 bytes, linear)
[    0.225007] pinctrl core: initialized pinctrl subsystem
[    0.233010] NET: Registered protocol family 16
[    0.238373] thermal_sys: Registered thermal governor 'step_wise'
[    0.276184] dcdc-xrx200 1f106a00.dcdc: Core Voltage : 1016 mV
[    0.289957] pinctrl-xway 1e100b10.pinmux: Init done
[    0.304885] dma-xway 1e104100.dma: Init done - hw rev: 7, ports: 7, channels: 28
[    0.312701] pcie-xrx200 1d900000.pcie: failed to get the PCIe PHY
[    0.393085] usbcore: registered new interface driver usbfs
[    0.398696] usbcore: registered new interface driver hub
[    0.404190] usbcore: registered new device driver usb
[    0.418978] clocksource: Switched to clocksource MIPS
[    0.426781] NET: Registered protocol family 2
[... 192 lines cut ...]
[   41.881509] device lan2 entered promiscuous mode
[   41.925587] gswip 1e108000.switch lan3: configuring for phy/rmii link mode
[   41.931677] 8021q: adding VLAN 0 to HW filter on device lan3
[   41.941160] br-lan: port 3(lan3) entered blocking state
[   41.944973] br-lan: port 3(lan3) entered disabled state
[   41.954376] device lan3 entered promiscuous mode
[   42.810946] device br-lan entered promiscuous mode
[   44.830860] gswip 1e108000.switch lan1: Link is Up - 1Gbps/Full - flow control rx/tx
[   44.837309] br-lan: port 1(lan1) entered blocking state
[   44.842774] br-lan: port 1(lan1) entered forwarding state
[   44.869164] IPv6: ADDRCONF(NETDEV_CHANGE): br-lan: link becomes ready
[  107.890497] Lantiq PTM driver, version 1.0.27, (c) 2001-2013 Lantiq Deutschland GmbH
[  107.890497]     PTM (E1) firmware version 2.16.0
[  107.901554] ifxmips_ptm: PTM init succeed
[  107.949795] br-lan: port 4(dsl0.7) entered blocking state
[  107.953783] br-lan: port 4(dsl0.7) entered disabled state
[  107.960244] device dsl0.7 entered promiscuous mode
[  107.963975] device dsl0 entered promiscuous mode
[  393.792405] enter showtime
[  393.793874] IPv6: ADDRCONF(NETDEV_CHANGE): dsl0: link becomes ready
[  393.805088] br-lan: port 4(dsl0.7) entered blocking state
[  393.809308] br-lan: port 4(dsl0.7) entered forwarding state
[  393.822538] enter showtime

```

Rondom, 2025-05-15:
It is expected that the firmware that you manually copied to the device will disappear on upgrade.

You could theoretically put the path in /etc/sysupgrade.conf, but it has the disadvantage that it would also be included in config-backup and restore which would be unexpected.

I am not familiar with the Lantiq firmware situation, but isn't there an OpenWrt package, these days? Or does it not apply to this device?

Rondom, 2025-05-15:
Quoting the [TOH page](https://openwrt.org/toh/avm/avm_fritz_box_7360_v2)

> After OpenWrt is flashed on the device there is an additional (xdsl vectoring enabled) firmware required in order to support vectoring (otherwise if the network supports VDSL vectoring, but the VDSL modem does not support it, the device will be put into a fall back mode using only the lower 2.2 MHz of the band, this results in reduced rates like 13 MBit/s down and 1.4 MBit/s up instead of 50 MBit/s).

I had hoped we had now permission to ship the firmware.

Amenia1999, 2025-05-15:
I will post the log from the not working Version if I am able to shutown the Fritz!Box.

I know how to import the Firmware, the Firmware just wont work on a new version.