openwrt/openwrt #18742: Unneeded spectre mitigation for Cortex-A53 (Mediatek Filogic 820, 830)

Wizard160888, 2025-05-07:
### Describe the bug

Commits:
[mediatek: Refresh kernel configuration](https://git.openwrt.org/?p=openwrt/openwrt.git;a=commit;h=c70d0a630404b4d243f21a4dfe17c039dd667a0b)
[mediatek: filogic: enable CONFIG_MITIGATE_SPECTRE_BRANCH_HISTORY](https://git.openwrt.org/?p=openwrt/openwrt.git;a=commit;h=d00ed19ed4cbeb446741ef977b63d7c96a045323)
add spectre mitigation for filogic subtarget, but The Cortex-A53 in MediaTek Filogic 820 (MT7981BA) and 830 (mt7986) not vulnerable.

### OpenWrt version

24.10.1

### OpenWrt release

d00ed19ed4cbeb446741ef977b63d7c96a045323

### OpenWrt target/subtarget

mediatek/filogic

### Device

Routerich AX3000

### Image kind

Official downloaded image

### Diffconfig

```text

```

### Terms

- [x] I am reporting an issue for OpenWrt, not an unsupported fork.

DragonBluep, 2025-05-08:
This fix is for MT7988 (Cortex-A73). Have you suffered significant performance losses?

LGA1150, 2025-05-08:
The mitigation will detect CPU ID at runtime and will only affect vulnerable CPUs.

Djfe, 2025-05-17:
Feels to me like you can close this now @Wizard160888 
Thanks for reporting but unless you report any issues everything likely works as expected :)

closed 2025-05-19 by Wizard160888