openwrt/openwrt #18760: realtek: rtl839x: no traffic on SFP ports for HP 1920-48G(-poe)

jwh, 2025-05-09:
Following on from https://github.com/openwrt/openwrt/pull/18737#issuecomment-2859458169

Previous to that the ports were working, just to be sure it wasn't my tree (even though its clean), I've flashed a snapshot with fresh config and still the same, is there a snapshot archive anywhere?  Annoyingly I can't seem to build much older commits as I get spurious build errors even with a fresh build_dir/staging_dir etc...

plappermaul, 2025-05-09:
Thanks for opening this. I'm working on different topics right now. So I need help to switch the context. Can you explain the regression you noticed? And if you are unsure at least check if it works on 24.10.0 or 24.10.1 default images

jwh, 2025-05-09:
On release I don't get link at all, and no EEPROM either of course - testing on non-poe 48G currently as that is what I was testing at the time, but it no longer works on poe either - but the problem seems to be there is absolutely no rx/tx, the remote port receives no frames and the local port doesn't receive any either, otherwise everything looks fine (has link, etc)

plappermaul, 2025-05-09:
Do you at least get Link Up on current main?

jwh, 2025-05-09:
Yeah, link comes up fine, just no traffic passed... it's really odd - I don't see anything that would have broken it and it obviously was working absolutely fine before

jwh, 2025-05-09:
on 24.10.1 I don't get link even if I switch mode, I guess I could go into tftp menu pre boot to get the same effect as `rtk network on` - on main with the already merged stuff I get link but no traffic, the thing is this was working fine on poe (all 4 ports), now ports don't work on either, so I'm a little confused as to why

plappermaul, 2025-05-09:
There is so much that needs to be done to get the link working. We wont solve this easily but we can check at least some basic bits.

Is the SerDes for the RTL8214FC active and running in QSGMII mode?

```
devmem 0xbb000008
devmem 0xbb00000c
```

https://svanheule.net/realtek/cypress/register/mac_serdes_if_ctrl

jwh, 2025-05-09:
I thought I had a winning combination:

main + 18402 + revert 1b7fd8464c2b170d0862e0c19fff11fd8dc50611

It wouldn't work with either just the PR or the revert, but now it works after sysupgrade without the revert, so I'm a little confused

jwh, 2025-05-09:
Yeah I'm out of idea, now I booted with clean main and its fine, nothing has physically changed, only reboots for sysupgrade and a cold boot in between just in case it was some weird state... I'll reboot/cold boot it a few more times and see if it stops working again

jwh, 2025-05-09:
Ok so, warm reboot and its not passing traffic again... can I give you console access to this if you have time? I have no idea what I'm looking for really

jwh, 2025-05-09:
Done (it's currently working again):

```
root@OpenWrt:~# devmem 0x1b000008
0x66666666
root@OpenWrt:~# devmem 0x1b00000c
0x00066666
root@OpenWrt:~#
```

T-X, 2025-05-10:
> I thought I had a winning combination:
> 
> main + 18402 + revert [1b7fd84](https://github.com/openwrt/openwrt/commit/1b7fd8464c2b170d0862e0c19fff11fd8dc50611)
> 
> It wouldn't work with either just the PR or the revert, but now it works after sysupgrade without the revert, so I'm a little confused

I'm a bit confused about that revert. Any idea why that change would touch rtl839x at all?

jwh, 2025-05-10:
Yeah I didn't think so I was just reverting possibly related stuff and it happened to work on that reboot, completely unrelated it seems, sorry!

plappermaul, 2025-05-10:
Regarding those two devmem above. That means SerDes 0-12 are running with QSGMII mode. This looks sane because

- each RTL8218x is connected via 2 QSGMII SerDes -> Mostly SerDes 0-11
- A RTL8214FC is connected via 1 QSGMII SerDes -> Mostly SerDes 12

Now you can try deactivationg and reactivating the SerDes. Just write to the above registers and replace a 6 by a 0 to power off a SerDes. The connected ports will no longer pass traffic Afterwards power it on again. And see what happens. E.g.

- devmem 0x1b000008 0x66666066 -> power off SerDes 2
- Ports 8-11 should no langer pass traffic
- devmem 0x1b000008 0x66666666 -> power on SerDes 2
- Ports 8-11 should work normal again

In your no connection case just power off+on all SerDes with both registers and see what happens.

If we had already had https://patchwork.kernel.org/project/linux-phy/patch/20250417124813.2406007-1-markus.stockhausen@gmx.de/ this would be easier.

plappermaul, 2025-05-10:
Set both registers to 0 and back to the old value. This power cycles all SerDes.

jwh, 2025-05-10:
Annoyingly I can't actually break it now, it's consistently working - I'm not sure what specifically prevents them working occasionally but it's hard to reproduce on demand it would appear

jwh, 2025-05-10:
If nobody else can reproduce this by next week I'll close it, it's a minor annoyance really and is "fixed" by just checking for a dhcp lease at boot after a small delay, if it's failed then reboot - eventually it'll work (although apparently now it just works all the time, sigh)

plappermaul, 2025-05-17:
One reason may be https://github.com/openwrt/openwrt/pull/18824

jwh, 2025-05-17:
aha, yes that would make sense, unfortunately it's extremely hard to reproduce, I'm guessing it depends on config (firstboot/initramfs is more likely to trigger it) and probe order - it's actually been consistently working now since last week with multiple reboots for PRs etc, so even if that isn't the cause it's not a huge issue anymore

jwh, 2025-05-19:
@plappermaul on a related note, if there is high load (like a loop), the switch will stop forwarding all traffic until reboot - would it be possible to detect this and reset the switch on it's own?

plappermaul, 2025-05-19:
Do not know what you mean. The SOC consists of

- a switching core -> Responsible for packet processing/forwarding
- a MIPS CPU -> configuring the switch core through registers

IF MIPS CPU loops the switch should still work. No idea how to detect if the switchcore hangs. But in this case we might have configured something wrong.

jwh, 2025-05-19:
Yeah, it *might* be the CPU side thats broken, but it definitely still sends TX (ie; for dhcp) it just never hears the response - I mean a loop like two ports plugged into each switch so its an endless cycle of broadcast spam, sometimes they recover sometimes not

plappermaul, 2025-05-19:
Switching topics in this issue is not helpful. Asking generic questions even less. I'm sorry but cannot follow here. If you suspect an error at some special code location (e.g. port->CPU receive) add deubg messages and report back.

jwh, 2025-05-21:
I think this issue has completely gone away now with the various fixes, so I'll close it, thanks!

closed 2025-05-21 by jwh