openwrt/openwrt #18777: realtek: HP 1920-48G-POE - /etc/config/poe not being generated/updated correctly

jwh, 2025-05-10:
### Describe the bug

Although 02_network has the correct definition:

```
hpe,1920-48g-poe)
        ucidef_set_poe 370 "lan8 lan7 lan6 lan5 lan4 lan3 lan2 lan1 lan16 lan15 lan14 lan13 lan12 lan11 lan10 lan9 lan24 lan23
                        lan22 lan21 lan20 lan19 lan18 lan17 lan32 lan31 lan30 lan29 lan28 lan27 lan26 lan25 lan40 lan39 lan38 lan37
                        lan36 lan35 lan34 lan33 lan48 lan47 lan46 lan45 lan44 lan43 lan42 lan41"
        ;;
```

It seems that on first boot a generic default is created that has 170w allowance and only 'lan1' as a port entry, I don't see where it even gets called, but perhaps someone else knows?

### OpenWrt version

r622-703e7d2d5b

### OpenWrt release

24.10

### OpenWrt target/subtarget

realtek/rtl839x

### Device

HPE 1920-48G-PoE (JG928A)

### Image kind

Self-built image

### Diffconfig

```text

```

### Terms

- [x] I am reporting an issue for OpenWrt, not an unsupported fork.

Djfe, 2025-05-20:
I don't see it either but how does the poe entry in your uci look like?

jwh, 2025-05-21:
I don't have the default now, but it looks the same on other switches, allowance of 77 and lan1 configured which I think is just the default baked in config anyway - it is not regenerated at all if missing

jwh, 2025-05-21:
Working poe config for the 1920-48G for others who want to use it:

```
config global
        option budget   '370'

config port
        option enable   '1'
        option id       '1'
        option name     'lan08'
        option poe_plus '1'
        option priority '2'

config port
        option enable   '1'
        option id       '2'
        option name     'lan07'
        option poe_plus '1'
        option priority '2'

config port
        option enable   '1'
        option id       '3'
        option name     'lan06'
        option poe_plus '1'
        option priority '2'

config port
        option enable   '1'
        option id       '4'
        option name     'lan05'
        option poe_plus '1'
        option priority '2'

config port
        option enable   '1'
        option id       '5'
        option name     'lan04'
        option poe_plus '1'
        option priority '2'

config port
        option enable   '1'
        option id       '6'
        option name     'lan03'
        option poe_plus '1'
        option priority '2'

config port
        option enable   '1'
        option id       '7'
        option name     'lan02'
        option poe_plus '1'
        option priority '2'

config port
        option enable   '1'
        option id       '8'
        option name     'lan01'
        option poe_plus '1'
        option priority '2'

config port
        option enable   '1'
        option id       '9'
        option name     'lan16'
        option poe_plus '1'
        option priority '2'

config port
[... 247 lines cut ...]
        option name     'lan44'
        option poe_plus '1'
        option priority '2'

config port
        option enable   '1'
        option id       '46'
        option name     'lan43'
        option poe_plus '1'
        option priority '2'

config port
        option enable   '1'
        option id       '47'
        option name     'lan42'
        option poe_plus '1'
        option priority '2'

config port
        option enable   '1'
        option id       '48'
        option name     'lan41'
        option poe_plus '1'
        option priority '2'
```

Djfe, 2025-05-21:
does the script maybe run too early before all ethernet ports are available or something?

jwh, 2025-05-21:
No, they're always available - it isn't run at all, even after nuking config and calling config_generate - there is no place where it is actually called.

Djfe, 2025-05-21:
what happens if you run /etc/board.d/02_network manually?

grobian, 2025-06-08:
perhaps this is because the case is not matching?  On my 1920-24G-PoE 370W I see this: ```
# cat /tmp/sysinfo/board_name 
hpe,1920-24g-poe-370w
```
does that return `hpe,1920-48g-poe` or `hpe,1920-48g-poe-370w`?

tim-seoss, 2025-11-06:
@Neustradamus Hi, please don't comment just to follow an issue - use the `Notifications` -> `Subscribe` button instead. Thanks.