openwrt/openwrt #18807: Race condition in netifd WAN teardown causes spurious ubus notify_proto errors

aon33, 2025-05-15:
### Describe the bug

During 'ifdown wan', netifd logs 'Command failed: ubus call network.interface notify_proto (Permission denied)' despite:

-     Manual ubus calls succeeding
-     ACL permissions being correct
-     The error being harmless (interface still tears down properly)

Findings:

        The error occurs when netifd attempts to notify network.interface via ubus after DHCP teardown.

        Timing Issue?: Adding a 3-second delay in proto_notify() for WAN eliminates the error (see below).

Reproduction:

    ifdown wan
    # Logs show error, but manual call works (after 3 seconds for my env):
    ubus call network.interface notify_proto '{"action":0,"link-up":false,"keep":false,"interface":"wan"}'

My simple workaround fix:

    Patch /lib/netifd/netifd-proto.sh to handle WAN teardown race condition:
diff:

```
proto_notify() {
   [ "$INTERFACE" = "wan" ] && sleep 3  # Workaround
.....
}
```

I think the 3s delay suggests network.interface takes 2-3s to re-register with ubus during teardown.  Hopefully this can help devs find a more elegant solution and others bothered by it.

### OpenWrt version

24.10.1

### OpenWrt release

24.10.1

### OpenWrt target/subtarget

ramips/mt7621

### Device

Linksys 7500v2

### Image kind

Official downloaded image

### Diffconfig

```text

```

### Terms

- [x] I am reporting an issue for OpenWrt, not an unsupported fork.

aon33, 2025-05-15:
Yes downloaded stable 24.10.1

aon33, 2025-05-19:
I not familiar with this, how to get this bot to fix this, remove the 'invalid' after I entered the correct version number?  Anyone?

isaacquirke-boop, 2026-02-17:
Thank You!  This fix solved a problem where my Nighthawk R7800 running 24.10.5 was reverting the WAN interface from 1000 to 100 mbps speed after a soft reboot.  I found the same permission denied error which led me here.  Before applying this fix, my only solution was to physically unplug and plug back in the WAN cable or Power cable.

proto_notify() {
   [ "$INTERFACE" = "wan" ] && sleep 3  # Workaround
.....
}