openwrt/openwrt #18810: dnsmasq / network stack not properly initialized at boot without manual restarts (Raspberry Pi 4)

briskshop, 2025-05-15:
### Describe the bug

On OpenWRT running on a Raspberry Pi 4 used as a wireless access point, dnsmasq does not assign IPs after boot. Clients are unable to obtain IP addresses until I manually run:

/etc/init.d/network restart
/etc/init.d/firewall restart
/etc/init.d/dnsmasq restart

Putting those restarts into /etc/rc.local with a delay resolves the issue, which suggests a startup race condition or dependency problem.

Hardware:
Raspberry Pi 4
Ethernet connected to main router (DHCP)
WiFi configured as access point (hostapd) on separate subnet (e.g. 192.168.100.0/24)

LOG:

cat  /tmp/bootlog.txt
Thu May 15 20:10:32 2025 daemon.info procd: - init complete -
Thu May 15 20:10:33 2025 daemon.notice hostapd: Set new config for phy phy0:
Thu May 15 20:10:33 2025 daemon.notice wpa_supplicant[615]: Set new config for phy phy0
Thu May 15 20:10:33 2025 daemon.notice wpa_supplicant[615]: Set new config for phy phy0
Thu May 15 20:10:33 2025 daemon.notice hostapd: Set new config for phy phy0: /var/run/hostapd-phy0.conf
Thu May 15 20:10:33 2025 daemon.notice hostapd: Restart interface for phy phy0
Thu May 15 20:10:33 2025 kern.info kernel: [   12.234489] brcmfmac mmc1:0001:1 phy0-ap0: renamed from wlan0
Thu May 15 20:10:33 2025 daemon.notice netifd: Interface 'lan' is enabled
Thu May 15 20:10:33 2025 daemon.notice netifd: Interface 'lan' is setting up now
Thu May 15 20:10:33 2025 daemon.notice hostapd: Configuration file: data: driver=nl80211 logger_syslog=127 logger_syslog_level=2 logger_stdout=127 logger_stdout_level=2 country_code=DE ieee80211d=1 ieee80211h=1 hw_mode=a beacon_int=100 stationary_ap=1 chanlist=36 tx_queue_data2_burst=2.0 #num_global_macaddr=1 #macaddr_base= ieee80211n=1 ht_coex=0 ht_capab=[HT40+][SHORT-GI-20][SHORT-GI-40][DSSS_CCK-40] ieee80211ac=1 vht_oper_chwidth=1 vht_oper_centr_freq_seg0_idx=42 vht_capab=[SHORT-GI-80][SU-BEAMFORMEE][MAX-A-MPDU-LEN-EXP0] channel=36  interface=phy0-ap0 bssid=fr:a6:32:f2:rr:ce ctrl_interface=/var/run/hostapd bss_load_update_period=60 chan_util_avg_period=600 disassoc_low_ack=1 skip_inactivity_poll=0 preamble=1 wmm_enabled=1 ignore_broadcast_ssid=0 uapsd_advertisement_enabled=1 utf8_ssid=1 multi_ap=0 wpa_passphrase=********* wpa_psk_file=/var/run/hostapd-phy0-ap0.psk auth_algs=1 wpa=2 wpa_pairwise=CCMP ssid=OpenWrt wpa_disable_eapol_key_retries=0 wpa_key_mgmt=WPA-PSK okc=0 disable_pmksa_caching=1 dynamic_vlan=0 vlan_naming=1
Thu May 15 20:10:33 2025 daemon.notice netifd: Interface 'lan' is now up
Thu May 15 20:10:33 2025 daemon.notice netifd: Interface 'lan' is now down
Thu May 15 20:10:33 2025 daemon.notice netifd: Interface 'lan' is disabled
Thu May 15 20:10:33 2025 daemon.notice netifd: Interface 'lan' is enabled
Thu May 15 20:10:33 2025 daemon.notice netifd: Interface 'lan' is setting up now
Thu May 15 20:10:33 2025 daemon.notice hostapd: phy0-ap0: interface state UNINITIALIZED->COUNTRY_UPDATE
Thu May 15 20:10:33 2025 daemon.notice netifd: Interface 'lan' is now up
Thu May 15 20:10:33 2025 daemon.notice hostapd: phy0-ap0: interface state COUNTRY_UPDATE->HT_SCAN
Thu May 15 20:10:33 2025 daemon.notice netifd: Wireless device 'radio0' is now up
Thu May 15 20:10:33 2025 daemon.notice netifd: Network device 'phy0-ap0' link is up
Thu May 15 20:10:33 2025 daemon.notice netifd: Interface 'lan' has link connectivity
Thu May 15 20:10:33 2025 daemon.notice hostapd: phy0-ap0: interface state HT_SCAN->ENABLED
Thu May 15 20:10:33 2025 daemon.notice hostapd: phy0-ap0: AP-ENABLED
Thu May 15 20:10:36 2025 daemon.notice netifd: Network device 'eth0' link is up
Thu May 15 20:10:36 2025 daemon.notice netifd: Interface 'wan' has link connectivity
Thu May 15 20:10:36 2025 daemon.notice netifd: Interface 'wan' is setting up now
Thu May 15 20:10:36 2025 kern.info kernel: [   14.965325] bcmgenet fd580000.ethernet eth0: Link is Up - 1Gbps/Full - flow control off
Thu May 15 20:10:36 2025 daemon.notice netifd: wan (1524): udhcpc: started, v1.36.1
Thu May 15 20:10:36 2025 daemon.notice netifd: wan (1524): udhcpc: broadcasting discover
Thu May 15 20:10:36 2025 daemon.notice netifd: wan (1524): udhcpc: broadcasting select for 192.168.5.76, server 192.168.5.1
Thu May 15 20:10:36 2025 daemon.notice netifd: wan (1524): udhcpc: lease of 192.168.5.76 obtained from 192.168.5.1, lease time 864000
Thu May 15 20:10:36 2025 daemon.notice netifd: Interface 'wan' is now up
Thu May 15 20:15:19 2025 authpriv.info dropbear[1682]: Child connection from 192.168.5.20:64225
Thu May 15 20:15:19 2025 authpriv.notice dropbear[1682]: Auth succeeded with blank password for 'root' from 192.168.5.20:64225

Steps to Reproduce:
Install OpenWRT on Raspberry Pi 4
Configure as an AP with static LAN interface and DHCP server (dnsmasq)
Reboot — wireless clients cannot obtain an IP
Manually restart network, firewall, and dnsmasq — everything works

### OpenWrt version

v24

### OpenWrt release

v24.10.1

### OpenWrt target/subtarget

bcm27xx/bcm2711

### Device

Raspberry Pi 4 Model B Rev 1.2

### Image kind

Official downloaded image

### Steps to reproduce

Install OpenWRT on Raspberry Pi 4
Configure as an AP with static LAN interface and DHCP server (dnsmasq)
Reboot — wireless clients cannot obtain an IP
Manually restart network, firewall, and dnsmasq — everything works

### Diffconfig

```text

```

### Terms

- [x] I am reporting an issue for OpenWrt, not an unsupported fork.

briskmbr, 2025-05-15:
I got it from here:

https://firmware-selector.openwrt.org/?version=24.10.1&target=bcm27xx%2Fbcm2711&id=rpi-4

brada4, 2025-05-16:
What is "static lan interface"? How is it different from default br-lan?

brada4, 2025-05-16:
Remove wifi device, check "empty bridge" checkbox and choose lan as network in wifi, complain to luci that it permits you to create broken config.

briskmbr, 2025-05-16:
Thanks for the suggestion!

 I see what you're pointing out that LuCI allows creating a bridge without any actual interface attached, which leads to a broken config.

In my case, I'm trying to run a Raspberry Pi as an access point with its own subnet (192.168.100.x), providing DHCP and NAT through Wi-Fi only no bridging with eth0.

If I follow your suggestion literally (removing the Wi-Fi device, checking "empty bridge", assigning to lan), it results in clients getting no IP address because the bridge (br-lan) has no interfaces. DHCP can't bind properly and clients have no routing.

But just to clarify: for a routed AP like mine (not bridged to the main LAN), the clean solution seems to be assigning the wireless directly to a non-bridged lan interface with a static IP, running DHCP, and configuring proper firewall/NAT rules to forward to wan.

brada4, 2025-05-16:
Nope, it allows to attach wifi interface as a member when correct way is that hostap joins the bridge

briskmbr, 2025-05-17:
ok configured as suggested, but I still have to Manually restart network, firewall, and dnsmasq after reboot.

[image]

[image]

[image]

brada4, 2025-05-17:
Did it work as out of the box?