openwrt/openwrt #18817: TP-Link Archer AX23 v1.20 - Unable to boot sometimes

xMikooo, 2025-05-16:
### Describe the bug

Router TP-Link Archer ax23 v.120 is unable to boot properly since release 23.05.5, with older 23.05.4 works flawlesly. This issue still occurs in the newest snapshot. The boot process will happen as usual, then hang at power LED and will stay like that forever. I have connected serial connection to one router running snapshot firmware and factory reset. I thought the router might kernel panic or hang at uboot somewhere, but no. It boots normally, you can access the console, the interfaces supposedly have IP addresses. However no ping and any kind of access is possible. Below is the bootlog, I'm not competent enough to troubleshoot further. This issue makes it not practical to use such router in a setup without backup power. In my case i have to reboot the router around 5 times to make it boot properly.
```
U-Boot 1.1.3 (Feb 16 2022 - 16:26:34)

Board: Ralink APSoC DRAM:  128 MB
relocate_code Pointer at: 87f58000

Config XHCI 40M PLL
******************************
Software System Reset Occurred
******************************
flash manufacture id: 20, device id 40 18
Warning: un-recognized chip ID, please update bootloader!
*** Warning - bad CRC, using default environment

============================================
Ralink UBoot Version: 5.0.0.0
--------------------------------------------
ASIC MT7621A DualCore (MAC to MT7530 Mode)
DRAM_CONF_FROM: Auto-Detection
DRAM_TYPE: DDR3
DRAM bus: 16 bit
Xtal Mode=3 OCP Ratio=1/3
Flash component: SPI Flash
Date:Feb 16 2022  Time:16:26:34
============================================
THIS IS uboot
icache: sets:256, ways:4, linesz:32 ,total:32768
dcache: sets:256, ways:4, linesz:32 ,total:32768

 ##### The CPU freq = 880 MHZ ####
 estimate memory size =128 Mbytes

Press '4' or 't' to break the booting process

Press 'x' to enter recovery web server                                                                                                                                                     0
nm_init:791
nm_initFwupPtnStruct:276
nm_lib_readPtnTable:738
[NM_Debug](nm_lib_readPtnTable) 00743: NM_PTN_TABLE_BASE = 0xfe0000
[NM_Debug](nm_lib_readPtnFromNvram) 00569: partition_used_len = 1054, requried len = 8192
[NM_Debug](nm_lib_readPtnTable) 00751: Reading Partition Table from NVRAM ... OK

[NM_Debug](nm_lib_readPtnTable) 00759: Parsing Partition Table ... OK

[NM_Debug](nm_lib_readPtnFromNvram) 00569: partition_used_len = 2, requried len = 3
factory boot check integer ok.

3: System Boot system code via Flash.
## Booting image at bc040000 ...
   Image Name:   MIPS OpenWrt Linux-6.6.89
   Image Type:   MIPS Linux Kernel Image (lzma compressed)
   Data Size:    3403996 Bytes =  3.2 MB
   Load Address: 80001000
   Entry Point:  80001000
   Verifying Checksum ... OK
   Uncompressing Kernel Image ... OK
No initrd
[NM_Debug](nm_lib_readPtnFromNvram) 00569: partition_used_len = 2, requried len = 3
[NM_Error](nm_api_checkTestMode) 00417: factory boot check testmode flag is not 1.

### test_mode=disable,boot_args:console=ttyS1,115200 root=/dev/mtdblock2 rootfstype=squashfs init=/etc/preinit mtdparts=raspi:256k(uboot),4096k(uImage),11968k@0x440000(rootfs),64k@0xff0000(ART) mem=128M test_mode=disable
## Transferring control to Linux (at address 80001000) ...
## Giving linux memsize in MB, 128

Starting kernel ...

[    0.000000] Linux version 6.6.89 (builder@buildhost) (mipsel-openwrt-linux-musl-gcc (OpenWrt GCC 14.2.0 r29528-ffde9a9fe9) 14.2.0, GNU ld (GNU Binutils) 2.42) #0 SMP Fri May  9 14:38:19 2025
[... 353 lines cut ...]
### Image kind

Official downloaded image

### Steps to reproduce

Install any version newer than 23.05.4 and try to boot the router. 

### Actual behaviour

When booting the router hangs at power LED, no ping, ssh etc. 80% of the time

### Expected behaviour

Router boots normally

### Diffconfig

```text

```

### Terms

- [x] I am reporting an issue for OpenWrt, not an unsupported fork.

xMikooo, 2025-05-17:
> Your log is normal boot, could you produce one relevant to the issue you describe and edit away irrelevant excessive log?

Well that is that failed boot, as I said it’s weird that openwrt is accessible on serial console, but no LEDs other than power and no response through Ethernet

brada4, 2025-05-17:
Experiment 1:
OK, it says no  cable connected anywhere. while with serial run `logread -f` and plug cable end in all ports (wan and one lan) to see if any is working.
Experiment 2:
Add wifi - > can you get lan like luci web from there if restoof net fails at boot?

xMikooo, 2025-05-17:
Ok I’ll try experiment 1 in my free time, as for experiment 2 WiFi also doesn’t get brought up, now I’m starting to think that the only led being the power one doesn’t mean it doesn’t hang just that it doesn’t bring any interface wired or wireless one so no led lights up

brada4, 2025-05-17:
Add 2.4ghz ap, it comes up immediately ( normally)

xMikooo, 2025-05-20:
> Add 2.4ghz ap, it comes up immediately ( normally)

Unfortunately WiFi also doesn’t come on, no led and no network found on my phone

brada4, 2025-05-20:
Only visible thing is ports dont detect cable. Jost dont reboot in the meantime and let sysupgrade and power utility to do that for you ;-)

pestasoft, 2025-06-18:
I also have an AX23 v1.20 router and also after reboot the same symptoms as xMikooo, is there any progress in solving the bug?

msebald, 2025-06-23:
I'm having the same problem with four AX23 devices, also see here: https://forum.openwrt.org/t/tp-link-archer-ax23-does-not-reboot-every-time-with-23-05-5/212435

xMikooo, 2025-06-23:
Unfortunately the device I tested it on is running by in production and  I don’t have the time to troubleshoot this further as i would have to take down that AP. I downgraded that device to 23.05.04 and it’s running flawlessly as a dumb AP which is what I need it to do. I also have other AX23 running 24.10.1 but it’s on a battery backup so not that big of of a problem. Once the device successfully boots it runs flawlessly, that one has over 70 days of uptime. I’m happy to see this issue pick up, since others might be able to troubleshoot and fix this issue in later releases. In quite puzzled that the device seemingly boots on the serial console even when the boot failed. Seems to me like some bug in the network stack, as there is no communication on Ethernet and the WiFi isn’t brought up. For now I recommend to stick to 23.05.4 as in most use cases there is no benefit in using 24.10.x and the device is stable and no performance hit in older release. To my knowledge the issue still persists in the latest snapshots

mraksoll4, 2026-01-14:
need check what i say, but i cannot do it (if i brick it i possible not restore it ) , so i now use TPlink default what was get recently update. And also wait for fix. 

Idea is - it simple skip up network interfaces due boot and we get shutdown all netowrk.. Default TPlink firmware is very slow boot , alsmot 1minute or more , possible wait network up...
So firmware need wait before they up/ expand timeouts etc..

sparky8512, 2026-01-17:
I think I found the root cause of the boot hang most people are hitting on these MT7621 devices. I posted details on issue [#18472](https://github.com/openwrt/openwrt/issues/18472#issuecomment-3764349185), including potential work arounds.

I was going to post that on this issue as there is a bit more info here, but it turns out the original report on this issue must have been something other than the kernel/driver deadlock I found, and I didn't want to confuse things further. In the original post, OP was able to run `ip addr` command on a serial terminal, which would not be possible once that deadlock triggered. The behavior noted in that [forum post](https://forum.openwrt.org/t/tp-link-archer-ax23-does-not-reboot-every-time-with-23-05-5/212435), however, would be consistent with it.

DragonBluep, 2026-03-29:
This issue can be closed, completed via https://github.com/openwrt/openwrt/pull/22136. The patch was backported to the stable kernel and will be included in the next release.

Ping @robimarko