openwrt/openwrt #18826: Iw dev phy1-sta0 set bitrates legacy-2.4 1 ht-mcs-2.4 not set TX speed to 1Mbit

AlexeiGHub, 2025-05-17:
### Describe the bug

Maybe someone can do a similar experiment and confirm the behavior.
In version 23.05.5 everything worked as I expected.

Maybe someone can tell me the commit that led to the behavior change?

### OpenWrt version

r28597-0425664679

### OpenWrt release

24.10.1

### OpenWrt target/subtarget

qualcommax/ipq807x

### Device

Redmi AX6

### Image kind

Official downloaded image

### Steps to reproduce

radio1 (2.4) act as Wifi Client (STA)  

execute "**iw dev phy1-sta0 set bitrates legacy-2.4 1 ht-mcs-2.4**" 
Can anyone tell me if this is a new feature of iw, IPQ driver or am I being stupid? Everything worked before.

### Actual behaviour

not set TX speed to 1Mbit
I try 1 and 2 Mbit, all less then 5.5 set 5.5(6Mbit)

### Expected behaviour

I expect STAinterface set 1Mbit Tx rate.

### Additional info

Can anyone tell me if this is a new feature of iw or IPQ driver? Everything worked before.

### Diffconfig

```text

```

### Terms

- [x] I am reporting an issue for OpenWrt, not an unsupported fork.

brada4, 2025-05-17:
Since phy "acting" involves hostapd you must &se hostapd_cli or ubus or (l)uci to influence its workings. Take a note that setting sub-par parameters on one AP degrades other interfaces on same radio.
If you need help with tools ask in forum. There is no bug exposed, just the way things work for ages.

AlexeiGHub, 2025-05-18:
> Since phy "acting" involves hostapd you must &se hostapd_cli or ubus or (l)uci to influence its workings. 

I think wpa-supplicant is used for STA mode
And parameters from iw are passed through API mac80211
I don't really understand how it works.

> Take a note that setting sub-par parameters on one AP degrades other interfaces on same radio.

On the interface (radio) I activate only STA mode

> If you need help with tools ask in forum. There is no bug exposed, just the way things work for ages.

I tried asking on the forum and the expert [lleachii](https://forum.openwrt.org/u/lleachii), and as I understood him, in the OWRT philosophy my actions are a hack and should not be guaranteed to work.

I hope maybe [robimarco](https://github.com/openwrt/openwrt/commits?author=robimarko) will have his say?

Intuitively, it seems to me that between 23.05.5 and 24.10.1 the driver version changed and this is the reason.
It feels like 802.11b codes (DSSS modulation) are prohibited.

brada4, 2025-05-18:
What do you have in radio config section?

AlexeiGHub, 2025-05-18:
Specially sterilized

config wifi-device 'radio1'
	option type 'mac80211'
	option path 'platform/soc@0/c000000.wifi+1'
	option band '2g'
	option channel 'auto'
	option legacy_rates '1'
	option cell_density '0'

current config

AlexeiGHub, 2025-05-18:
after execute **iw dev  phy1-sta0 set bitrates legacy-2.4 1  ht-mcs-2.4**
i have **iwinfo phy1-sta0 info**

phy1-sta0 ESSID: "********"
          Access Point: yy:yy:yy:yy:yy:yy
          Mode: Client  Channel: 11 (2.462 GHz)  HT Mode: HT40
          Center Channel 1: 9 2: unknown
          Tx-Power: 27 dBm  Link Quality: 25/70
          Signal: -** dBm  Noise: -107 dBm
          **Bit Rate: 6.0 MBit/s**
          Encryption: WPA2 PSK (CCMP)
          Type: nl80211  HW Mode(s): 802.11ax/b/g/n
          Hardware: embedded [Qualcomm Atheros IPQ8074]
          TX power offset: none
          Frequency offset: none
          Supports VAPs: yes  PHY name: phy1

brada4, 2025-05-18:
You need to disable ht in configuration.
 https://w1.fi/cgit/hostap/tree/wpa_supplicant/wpa_supplicant.conf#n1721

AlexeiGHub, 2025-05-18:
OK, I try/

AlexeiGHub, 2025-05-18:
But again, I'm in STA mode.

This parameter, as I understand, is for AP mode?!

Can I specify it via UCI or directly in /etc/config/wireless?

AlexeiGHub, 2025-05-18:
[image]

brada4, 2025-05-18:
wpa_supplicant is the STA mode
checked scripts
https://openwrt.org/docs/guide-user/network/wifi/basic
`option htmode NOHT`
into 2.4ghz radio section.

AlexeiGHub, 2025-05-18:
OK

/etc/config/wireles

**config wifi-device 'radio1'
	option type 'mac80211'
	option path 'platform/soc@0/c000000.wifi+1'
	option band '2g'
	option channel '1'
	option legacy_rates '1'
	option cell_density '0'
	option htmode 'NOHT'**

1) **Only STA mode**  

iwinfo phy1-sta0 info > "HT Mode: HT40' previous behavior

2) **Only AP mode**  

iwinfo phy1-ap0  info > "HT Mode: NOHT'.  
speeds 802.11g   
**iw dev  phy1-ap0 set bitrates legacy-2.4 1**  has an effect with/without **ht-mcs-2.4** , the speed is blocked. But the same only not lower than 6 Mbit 

3) **AP+STA mode**

iwinfo phy1-sta0 info > "HT Mode: HT40' previous behavior
iwinfo phy1-sta0  info > "HT Mode: NOHT'.  has an effect, the speed is blocked. But the same only not lower than 6 Mbit 

!!!**IIMPORTANT**!!!! in AP+STA  tx speed and HTxx mode  for sta0 and ap0 **sets independed** :)

AlexeiGHub, 2025-05-18:
> wpa_supplicant is the STA mode
> checked scripts

Please describe this moment in more detail.

AlexeiGHub, 2025-05-18:
/var/run/wpa_supplicant-phy1-sta0.conf

**network={
	scan_ssid=1
	ssid="XXXXXXXX"
	key_mgmt=WPA-PSK
	psk="********"
	proto=RSN
	beacon_int=100
}**

this is all :(

AlexeiGHub, 2025-05-18:
> https://w1.fi/cgit/hostap/tree/wpa_supplicant/wpa_supplicant.conf#n1721

I haven't figured out how to implement this yet :(

AlexeiGHub, 2025-05-18:
> !!!**IIMPORTANT**!!!! in AP+STA tx speed and HTxx mode for sta0 and ap0 **sets independed** via **iw** command :)

This contradicts the concept of disabling 'HT' for the entire radio1

brada4, 2025-05-18:
Watch the tickmarks and check PHY not STA conf in same dir

AlexeiGHub, 2025-05-18:
> Watch the tickmarks and check PHY not STA conf in same dir

I didn't understand what directory and what to do?

brada4, 2025-05-18:
/etc/config/wireless
You noticed it I think ;-)

Now check iwinfo.... should be NOHT all around.

AlexeiGHub, 2025-05-19:
OK.

**config wifi-device 'radio0'
	option type 'mac80211'
	option path 'platform/soc@0/c000000.wifi'
	option band '5g'
	option cell_density '0'
	option htmode 'NOHT'
	option channel '36'**

**config wifi-device 'radio1'
	option type 'mac80211'
	option path 'platform/soc@0/c000000.wifi+1'
	option band '2g'
	option channel '1'
	option legacy_rates '1'
	option cell_density '0'
	option htmode 'NOHT'**

**iwinfo phy0-ap0 info**
Mode: Master  Channel: 36 (5.180 GHz)  HT Mode: NOHT
          Center Channel 1: 36 2: unknown
          Tx-Power: 10 dBm  Link Quality: 53/70
          Signal: -57 dBm  Noise: -101 dBm
          Bit Rate: 42.0 MBit/s
          Encryption: mixed WPA2/WPA3 PSK/SAE (CCMP)
          Type: nl80211  HW Mode(s): 802.11ac/ax/n
          Hardware: embedded [Qualcomm Atheros IPQ8074]
          TX power offset: none
          Frequency offset: none
          Supports VAPs: yes  PHY name: phy0

**iwinfo phy1-ap0 info**
Mode: Master  Channel: 1 (2.412 GHz)  HT Mode: NOHT
          Center Channel 1: 1 2: unknown
          Tx-Power: 27 dBm  Link Quality: 67/70
          Signal: -43 dBm  Noise: -106 dBm
          Bit Rate: 48.0 MBit/s
          Encryption: WPA2 PSK (CCMP)
          Type: nl80211  HW Mode(s): 802.11ax/b/g/n
          Hardware: embedded [Qualcomm Atheros IPQ8074]
          TX power offset: none
          Frequency offset: none
          Supports VAPs: yes  PHY name: phy1

**iwinfo phy1-sta0 info**
 Mode: Client  Channel: 1 (2.412 GHz)  HT Mode: HT40
          Center Channel 1: 3 2: unknown
          Tx-Power: 27 dBm  Link Quality: 59/70
          Signal: -51 dBm  Noise: -106 dBm
          Bit Rate: 243.0 MBit/s
          Encryption: WPA2 PSK (CCMP)
          Type: nl80211  HW Mode(s): 802.11ax/b/g/n
          Hardware: embedded [Qualcomm Atheros IPQ8074]
          TX power offset: none
          Frequency offset: none
          Supports VAPs: yes  PHY name: phy1

!!!!**All at one time**!!!

Everything is as I expected.
What did you expect?

AlexeiGHub, 2025-05-19:
I think similar results can be obtained on AX3600 (the same chip)

brada4, 2025-05-19:
"supplicant script overrides forced wifi standard downgrade on phy"
If you have time please check snapshot, that has new generation of script in ucode. Now the ideal illustration of encountered problem.
Can you add code box `< >` as "actual behavior" in original report with NOHT in conf and iwinfo showing setting ignored?

AlexeiGHub, 2025-05-19:
> Can you add code box `< >` as "actual behavior" in original report with NOHT in conf and iwinfo showing setting ignored?

Did I understand you correctly and make changes to the first post?

> If you have time please check snapshot, that has new generation of script in ucode. Now the ideal illustration of encountered problem

If the nature of the changes allows, can I make changes to the script in the current instance using the overlay file system without reflashing the router from the stable firmware to the snapshot?

brada4, 2025-05-19:
Yes, some superfluous user may read just that.
Can you check that mode HT20 is also exceeded towards HT40?

You can make config backup, upgrade to snapshot using owut, then downgrade resetting conf and restoring backed up one. 

Nope, 24.10 will not run ucode  wifi scripts, but a nice try.

AlexeiGHub, 2025-05-19:
> Can you check that mode HT20 is also exceeded towards HT40?

Enabling HT20 mode will block transmission at 40 MHz (HT40). Did I understand correctly?
This is also a long-standing problem.

Or will it also work on recieve at 20 MHz (HT20)?

brada4, 2025-05-19:
HT40 is nearly impossible if you agree with standards. Since you are happy with 1Mbps I think bw is not of concern.

AlexeiGHub, 2025-05-20:
I did a similar experiment in VMWARE with a USB stick.
Everything works as I expect. Speeds are set to 1Mbit, 2Mbit. Checked in 24.10.1 and in snapshot.

[image]

I **haven't** checked for IPQ807x yet. ASU and OWUT are unavailable for a long time.

Please send a link to the commit with fixes that you think can change the situation.