openwrt/openwrt #18828: apk: strongswan cortex-a53 (no such package) on SNAPSHOT

nikolainv225, 2025-05-17:
### Describe the bug

Environment: mediatek/filogic (aarch64_cortex-a53)  SNAPSHOT r29487-044c7d55b3

Description: I can't install strongswan-related packages from the repository. [snapshots/faillogs](https://downloads.openwrt.org/snapshots/faillogs/aarch64_cortex-a53/packages/strongswan/) have errors when building the package

### OpenWrt version

r29487-044c7d55b3

### OpenWrt release

SNAPSHOT

### OpenWrt target/subtarget

mediatek/filogic

### Device

Cudy WR3000H v1

### Image kind

Official downloaded image

### Steps to reproduce

apk add --simulate strongswan

### Diffconfig

```text

```

### Terms

- [x] I am reporting an issue for OpenWrt, not an unsupported fork.

brada4, 2025-05-17:
Helps reporting in correct repo where
package is maintained. https://downloads.openwrt.org/snapshots/faillogs/aarch64_cortex-a53/packages/strongswan/

nikolainv225, 2025-05-17:
> Helps reporting in correct repo where package is maintained. https://downloads.openwrt.org/snapshots/faillogs/aarch64_cortex-a53/packages/strongswan/

Thx
[openwrt/packages/issues/26525](https://github.com/openwrt/packages/issues/26525)

robimarko, 2025-05-19:
This has been fixed and strongswan is built again

closed 2025-05-19 by robimarko