openwrt/openwrt #18844: nanopc-T6 emmc card problem

AP1010, 2025-05-19:
### Describe the bug

I own two nanoPC-T6 boards. They're both 16GB RAM models with 256GB eMMC cards. One has the eMMC card identified by parted as Model: MMC A3A564, the other one Model: MMC A3A444. The one with A3A564 works flawlessly in all my applications, the one with A3A444 has problems specifically with OpenWrt. OpenWrt seems unable to read/write the eMMC model A3A444 (works fine with model A3A564) . If I flash Openwrt (24.10.0, 24.10.1, or snapshot) to the emmc, it tries to boot but ends in kernel panic. If I boot OpenWrt from the SD card, it boots OK, but runs into difficulties accessing emmc and throws errors along the lines of:
`[ 2.399727] I/O error, dev mmcblk0, sector 152845 op 0x0:(READ) flags 0x80700 phys_seg 1 prio class 2

[ 2.402253] I/O error, dev mmcblk0, sector 152858 op 0x0:(READ) flags 0x80700 phys_seg 1 prio class 2

[ 2.538843] I/O error, dev mmcblk0, sector 147968 op 0x0:(READ) flags 0x80700 phys_seg 2 prio class 2

[ 2.539887] I/O error, dev mmcblk0, sector 147969 op 0x0:(READ) flags 0x80700 phys_seg 2 prio class 2`
I would suspect a faulty emmc, but...
If I boot FriendlyWrt from SD - it has no problem reading/writing to emmc model A3A444. The same goes for booting Armbian - no issues with emmc A3A444. The problem appears to be OpenWrt-specific.

### OpenWrt version

r28597-0425664679

### OpenWrt release

24.10.1

### OpenWrt target/subtarget

rockchip/armv8

### Device

FriendlyElec NanoPC-T6

### Image kind

Official downloaded image

### Steps to reproduce

Attempt to read/write to emmc card identified as A3A444. Only this particular model gave me trouble.

### Actual behaviour

Unable to perform any I/O operation on emmc. I/O errors thrown, e.g.:
I/O error, dev mmcblk0, sector 152845 op 0x0:(READ) flags 0x80700 phys_seg 1 prio class 2

### Expected behaviour

Ability to perform I/O operations on emmc without errors.

### Additional info

The issue was observed in 24.10.0; 24.10.0-rc7; 24.10.1 and snapshots.
In addition to using official OpenWrt images, I tried building my own tweaking parameters like CQHCI. This had no effect.

### Diffconfig

```text

```

### Terms

- [x] I am reporting an issue for OpenWrt, not an unsupported fork.

AP1010, 2025-05-21:
With the help of FriendlyElec support, I found a work around. If I modify the device tree to disable mmc-hs400, in other words, in  the file:
 ./build_dir/target-aarch64_generic_musl/linux-rockchip_armv8/linux-6.6.89/arch/arm64/boot/dts/rockchip/rk3588-nanopc-t6.dtsi
replace 
`
 mmc-hs400-1_8v;

 mmc-hs400-enhanced-strobe;
`
with:

` //mmc-hs400-1_8v;

 //mmc-hs400-enhanced-strobe;

  mmc-hs200-1_8v;
`
This allows me to use the board with OpenWrt, but still doesn't explain why OpenWrt needs this 'fix' while Arabian and FriendlyWrt don't.

dzikus, 2025-06-29:
Same here MMC A3A444. I can confirm this bug with 24.10.1 r28597-0425664679, everything is working stable with mmc-hs200-1_8v; dtb patch.

FriendlyWRT is working without problems (many versions).

closed 2025-07-21 by openwrt-bot

dzikus, 2025-07-22:
24.10 still affected, PR #19277 not yet merged

commit c4586cea (2025-07-21): rockchip: fix eMMC corruption on NanoPC-T6 with A3A444 chips

Some NanoPC-T6 boards with A3A444 eMMC chips experience I/O errors and
corruption when using HS400 mode. Downgrade to HS200 mode to ensure
stable operation.

Fixes: #18844

Link: https://github.com/openwrt/openwrt/pull/19398
[Add patch header]