openwrt/openwrt #18845: sqm-scripts - 1.6.0-1 needs dependency on iptables-nft and ip6tables-nft

sansrenom, 2025-05-19:
### Describe the bug

Installed luci-app-sqm, which pulled in sqm-scripts,  BUT sqm would not start

turns out that iptables-nft and ip6tables-nft are required for sqm to start, but were not installed on my system

I installed iptables-nft and ip6tables-nft manually and sqm starts and runs great now

Proposed solution: sqm should automatically install iptables-nft and ip6tables-nft 

### OpenWrt version

r24106-10cc5fcd00

### OpenWrt release

23.05.5

### OpenWrt target/subtarget

ramips/mt7621

### Device

Netgear WAC104

### Image kind

Official downloaded image

### Diffconfig

```text

```

### Terms

- [x] I am reporting an issue for OpenWrt, not an unsupported fork.

brada4, 2025-05-19:
According to package Makefile maintainer is @tohojo and @robimarko did last change.
Nope, it requires one of 2 iptables, old iptables-legacy is still supported and usable.

sansrenom, 2025-05-19:
The file /etc/sqm/sqm.conf  is installed by package sqm-scripts:

cat /etc/sqm/sqm.conf 
SQM_LIB_DIR=/usr/lib/sqm
SQM_STATE_DIR=/var/run/sqm
SQM_QDISC_STATE_DIR=${SQM_STATE_DIR}/available_qdiscs
SQM_CHECK_QDISCS="fq_codel codel pie sfq cake"
SQM_SYSLOG=1
IP6TABLES_BINARY=$(command -v ip6tables-nft)
IPTABLES_BINARY=$(command -v iptables-nft)

 sqm won't start out-of-the-box without the *-nft version of iptables, OR it sounds like I *could* have edited this file to use iptables-legacy, but that required edit isn't in any documentation or install guides that I read. It took me more than an hour to debug the non-starting issue :)

My system currently has:

opkg list-installed | grep tables
ip6tables-nft - 1.8.8-2
iptables-mod-ipopt - 1.8.8-2
iptables-nft - 1.8.8-2
iptables-zz-legacy - 1.8.8-2
kmod-ip6tables - 5.15.167-1
libxtables12 - 1.8.8-2
nftables-json - 1.0.8-1
xtables-legacy - 1.8.8-2
xtables-nft - 1.8.8-2

brada4, 2025-05-20:
Probsbly yo& can team up here and make a pull request or two ;-)
https://github.com/openwrt/packages/issues/26254

tohojo, 2025-05-20:
@sansrenom what is the content of your /etc/config/sqm that didn't work without iptables-nft?

sansrenom, 2025-05-20:
@tohojo   in file /etc/config/sqm, the following lines cause sqm startup to return "file not found" or similar error...

IP6TABLES_BINARY=$(command -v ip6tables-nft)
IPTABLES_BINARY=$(command -v iptables-nft)

tohojo, 2025-05-20:
sansrenom ***@***.***> writes:

> sansrenom left a comment (openwrt/openwrt#18845)
>
> @tohojo   in file /etc/config/sqm, the following lines cause sqm startup to return "file not found" or similar error...
>
> IP6TABLES_BINARY=$(command -v ip6tables-nft)
> IPTABLES_BINARY=$(command -v iptables-nft)

Those commands should fail silently, and things should keep working.
Please share the full error output you're seeing, and the contents of
/etc/config/sqm that causes this.