openwrt/openwrt #18854: realtek: RTL839x based switch stalls/reboot

plappermaul, 2025-05-20:
### Describe the bug

This is a collection about unidentified reboots on the RTL839x target with current master

### OpenWrt version

r28597-0425664679

### OpenWrt release

24.10.1

### OpenWrt target/subtarget

realtek

### Device

RTL839x based HP 1920

### Image kind

Official downloaded image

### Actual behaviour

Swtich reboots unexpecetdly without further notice

### Expected behaviour

Switch runs 

### Diffconfig

```text

```

### Terms

- [x] I am reporting an issue for OpenWrt, not an unsupported fork.

plappermaul, 2025-05-20:
@jwh @howels

To avoid that you capture every other PR from me with stall/reboot issues I opened this one only for you.

Before going into a long discussion start with the following test:
 
- install/compile iperf3 into the image
- start "iperf3 -s" on the switch
- start "iperf3 -c <SwitchIP> -P 8 -t 180" on another network device

Check if switch survives these 3 minutes of hammering the CPU

jwh, 2025-05-20:
Hey, thanks

Annoyingly no, it only manages about 100mbps due to the weak CPU - I expect the problem occurs if it just sees too many incoming frames that it can't take quickly enough, which in my case was a broadcast storm. Let me see if I can recreate it without looping it and killing all of my test switches...

plappermaul, 2025-05-20:
Please, just do the above test.

howels, 2025-05-20:
My switch also completes the iperf test, albeit with awful performance. For reference I was also triggering this in a high CPU situation - I think I was trying to pull a lot of prometheus metrics while I have a bunch of sessions open to the web UI and it froze and rebooted. That was the time I was actively using it, but it was also restarting with nothing else running when left to itself. I had disabled Prometheus node exporter so the switch was running `realtek-poe` and `lldpd` only.

jwh, 2025-05-20:
Simply sending it more UDP than it can handle also triggers it, tcp obviously backs off so it probably won't do it:

~150mbps of unicast udp is enough:

```
[83207.551199] rtl838x-eth 1b00a300.ethernet eth0: Ring contention: r: 0, last a2800348, cur a28000e8
[83231.164181] rtl838x-eth 1b00a300.ethernet eth0: Ring contention: r: 0, last a2800114, cur a2800170
[83231.195052] rtl838x-eth 1b00a300.ethernet eth0: Ring contention: r: 0, last a2800348, cur a28000d8
[83231.352936] rtl838x-eth 1b00a300.ethernet eth0: Ring contention: r: 0, last a280032c, cur a2800440
[83276.507274] rtl838x-eth 1b00a300.ethernet eth0: Ring contention: r: 0, last a2800134, cur a28001c8
[83276.618468] rtl838x-eth 1b00a300.ethernet eth0: Ring contention: r: 0, last a2800158, cur a28003d8
[83276.673281] rtl838x-eth 1b00a300.ethernet eth0: Ring contention: r: 0, last a2800300, cur a2800368
[83276.703976] rtl838x-eth 1b00a300.ethernet eth0: Ring contention: r: 0, last a28000a8, cur a2800350
[83276.883547] rtl838x-eth 1b00a300.ethernet eth0: Ring contention: r: 0, last a280006c, cur a28003b8
[83276.924614] rtl838x-eth 1b00a300.ethernet eth0: Ring contention: r: 0, last a280044c, cur a28001f0
```

now it has stopped receiving, it looks like this happens (ringbuffer full?) and then it never recovers

jwh, 2025-05-20:
@howels I haven't had any spontaneous reboots that I can tell, it just stops receiving until I reboot - what else are you running? I am also running lldpd and prometheus (lua exporter), mini_snmpd, chrony and thats pretty much it I think

plappermaul, 2025-05-20:
Ok. So the receive path of the CPU port stalls. That is a point to start with.

howels, 2025-05-20:
> [@howels](https://github.com/howels) I haven't had any spontaneous reboots that I can tell, it just stops receiving until I reboot - what else are you running? I am also running lldpd and prometheus (lua exporter), mini_snmpd, chrony and thats pretty much it I think

That's pretty much my build - here are the packages from my config:

```
CONFIG_PACKAGE_apk-mbedtls=y
CONFIG_PACKAGE_attendedsysupgrade-common=y
CONFIG_PACKAGE_base-files=y
CONFIG_PACKAGE_busybox=y
CONFIG_PACKAGE_ca-bundle=y
CONFIG_PACKAGE_dropbear=y
CONFIG_PACKAGE_firewall4=y
CONFIG_PACKAGE_fstools=y
CONFIG_PACKAGE_fwtool=y
CONFIG_PACKAGE_getrandom=y
CONFIG_PACKAGE_jsonfilter=y
CONFIG_PACKAGE_libc=y
CONFIG_PACKAGE_libgcc=y
CONFIG_PACKAGE_logd=y
CONFIG_PACKAGE_mtd=y
CONFIG_PACKAGE_netifd=y
CONFIG_PACKAGE_openwrt-keyring=y
CONFIG_PACKAGE_procd=y
CONFIG_PACKAGE_procd-seccomp=y
CONFIG_PACKAGE_procd-ujail=y
CONFIG_PACKAGE_rpcd=y
CONFIG_PACKAGE_rpcd-mod-file=y
CONFIG_PACKAGE_rpcd-mod-iwinfo=y
CONFIG_PACKAGE_rpcd-mod-rpcsys=y
CONFIG_PACKAGE_rpcd-mod-ucode=y
CONFIG_PACKAGE_ubox=y
CONFIG_PACKAGE_ubus=y
CONFIG_PACKAGE_ubusd=y
CONFIG_PACKAGE_uci=y
CONFIG_PACKAGE_urandom-seed=y
CONFIG_PACKAGE_urngd=y
CONFIG_PACKAGE_usign=y
CONFIG_PACKAGE_libiwinfo-data=y
CONFIG_PACKAGE_kmod-crypto-crc32c=y
CONFIG_PACKAGE_kmod-crypto-hash=y
CONFIG_PACKAGE_kmod-gpio-button-hotplug=y
CONFIG_PACKAGE_kmod-lib-crc32c=y
CONFIG_PACKAGE_kmod-nf-conntrack=y
CONFIG_PACKAGE_kmod-nf-conntrack6=y
CONFIG_PACKAGE_kmod-nf-flow=y
CONFIG_PACKAGE_kmod-nf-log=y
CONFIG_PACKAGE_kmod-nf-log6=y
CONFIG_PACKAGE_kmod-nf-nat=y
CONFIG_PACKAGE_kmod-nf-reject=y
CONFIG_PACKAGE_kmod-nf-reject6=y
CONFIG_PACKAGE_kmod-nfnetlink=y
CONFIG_PACKAGE_kmod-nft-core=y
CONFIG_PACKAGE_kmod-nft-fib=y
CONFIG_PACKAGE_kmod-nft-nat=y
CONFIG_PACKAGE_kmod-nft-offload=y
CONFIG_PACKAGE_lua=y
CONFIG_PACKAGE_luasocket=y
CONFIG_PACKAGE_ucode=y
CONFIG_PACKAGE_ucode-mod-fs=y
CONFIG_PACKAGE_ucode-mod-math=y
CONFIG_PACKAGE_ucode-mod-ubus=y
CONFIG_PACKAGE_ucode-mod-uci=y
CONFIG_PACKAGE_ucode-mod-uloop=y
CONFIG_PACKAGE_libsysfs=y
CONFIG_PACKAGE_libmbedtls=y
CONFIG_PACKAGE_jansson=y
CONFIG_PACKAGE_libblobmsg-json=y
CONFIG_PACKAGE_libcap=y
CONFIG_PACKAGE_libevent2=y
CONFIG_PACKAGE_libiwinfo=y
CONFIG_PACKAGE_libjson-c=y
CONFIG_PACKAGE_liblua=y
CONFIG_PACKAGE_liblucihttp=y
CONFIG_PACKAGE_liblucihttp-ucode=y
CONFIG_PACKAGE_libmnl=y
CONFIG_PACKAGE_libncurses=y
CONFIG_PACKAGE_libnftnl=y
CONFIG_PACKAGE_libnl-tiny=y
CONFIG_PACKAGE_libsensors=y
CONFIG_PACKAGE_libubox=y
CONFIG_PACKAGE_libubus=y
CONFIG_PACKAGE_libubus-lua=y
CONFIG_PACKAGE_libuci=y
CONFIG_PACKAGE_libuclient=y
CONFIG_PACKAGE_libucode=y
CONFIG_PACKAGE_libudebug=y
CONFIG_PACKAGE_libustream-mbedtls=y
CONFIG_PACKAGE_rpcd-mod-luci=y
CONFIG_PACKAGE_rpcd-mod-rrdns=y
CONFIG_PACKAGE_terminfo=y
CONFIG_PACKAGE_zlib=y
CONFIG_PACKAGE_luci-light=y
CONFIG_PACKAGE_luci-ssl=y
CONFIG_PACKAGE_luci-base=y
CONFIG_PACKAGE_luci-mod-admin-full=y
CONFIG_PACKAGE_luci-mod-network=y
CONFIG_PACKAGE_luci-mod-status=y
CONFIG_PACKAGE_luci-mod-system=y
CONFIG_PACKAGE_luci-app-firewall=y
CONFIG_PACKAGE_luci-app-lldpd=y
CONFIG_PACKAGE_luci-app-package-manager=y
CONFIG_PACKAGE_luci-theme-bootstrap=y
CONFIG_PACKAGE_luci-proto-ipv6=y
CONFIG_PACKAGE_luci-proto-ppp=y
CONFIG_PACKAGE_luci-lib-nixio=y
CONFIG_PACKAGE_nftables-json=y
CONFIG_PACKAGE_knot-resolver_dnstap=y
CONFIG_PACKAGE_lldpd=y
CONFIG_PACKAGE_cgi-io=y
CONFIG_PACKAGE_uhttpd=y
CONFIG_PACKAGE_uhttpd-mod-lua=y
CONFIG_PACKAGE_uhttpd-mod-ubus=y
CONFIG_PACKAGE_ethtool=y
CONFIG_PACKAGE_odhcp6c=y
CONFIG_PACKAGE_realtek-poe=y
CONFIG_PACKAGE_uclient-fetch=y
CONFIG_PACKAGE_uboot-envtools=y
CONFIG_PACKAGE_nano=y
CONFIG_PACKAGE_px5g-mbedtls=y
CONFIG_PACKAGE_sysfsutils=y
CONFIG_PACKAGE_jshn=y
CONFIG_PACKAGE_libjson-script=y
CONFIG_PACKAGE_lm-sensors=y
CONFIG_PACKAGE_owut=y
CONFIG_PACKAGE_prometheus-node-exporter-lua=y
CONFIG_PACKAGE_prometheus-node-exporter-lua-hwmon=y
CONFIG_PACKAGE_prometheus-node-exporter-lua-netstat=y
CONFIG_PACKAGE_prometheus-node-exporter-lua-openwrt=y
CONFIG_PACKAGE_prometheus-node-exporter-lua-realtek-poe=y
CONFIG_PACKAGE_ucode-mod-html=y
CONFIG_PACKAGE_ucode-mod-uclient=y
```

jwh, 2025-05-20:
I don't think there is even anything in there that would cause this and its specifically triggered by high load anyway - at least it seems its only the CPU port that breaks

plappermaul, 2025-05-20:
Maybe try https://github.com/openwrt/openwrt/pull/18855

howels, 2025-05-22:
> Invalid Target/Subtarget reported. `realtek` Is this from a supported device?

@plappermaul please add the correct target name to avoid an invalid tag.

BrainSlayer, 2025-05-23:
> Ok. So the receive path of the CPU port stalls. That is a point to start with.

i can reproduce it on 839x and 9313 btw

howels, 2025-05-23:
> > I will keep these running and see what happens, I've got 2xJG928A and 1xJG927A and so far I haven't got the uptime for the JG928A consistently much more than a week. Guess there's some sequence of events that occur, it's rare but super-annoying.
> 
> there is a easy sequence. just do a udp based iperf3 tests with 1 gbit speed and 4 connections. it immediatly happens.

Different arguments than the above test? Mine didn't reboot during iperf3
https://github.com/openwrt/openwrt/issues/18854#issuecomment-2895360704

howels, 2025-05-24:
> > > > I will keep these running and see what happens, I've got 2xJG928A and 1xJG927A and so far I haven't got the uptime for the JG928A consistently much more than a week. Guess there's some sequence of events that occur, it's rare but super-annoying.
> > > 
> > > 
> > > there is a easy sequence. just do a udp based iperf3 tests with 1 gbit speed and 4 connections. it immediatly happens.
> > 
> > 
> > Different arguments than the above test? Mine didn't reboot during that type of iperf3 test [#18854 (comment)](https://github.com/openwrt/openwrt/issues/18854#issuecomment-2895360704)
> 
> no? but you said its super rare. but its not. its easy to trigger and leads to reboots caused by message flood which will block processes and leads to triggering of the watchdog. if you are not making use of the watchdog it will likelly not happen. anyway i'm currenly working on that driver while developing the rtl931x support and will try to fix that issue.

I'm gonna test a bunch of iperf3 variations and see if I can replicate the reboot. So far it's been happening when I wasn't actively logged in.

BrainSlayer, 2025-05-24:
the problem is that this is what i found out that the serial console is flooded by messages which brings the cpu to the limit which triggers the watchdog at the end since all stalls.. solution. the add rate limiting to the warning output. i did this already in my tree + some buffer changes and napi threading. your iperf3 client argument is missing the udp bandwidth btw. thisis no stress test then. add -b 1G

howels, 2025-05-24:
> the problem is that this is what i found out that the serial console is flooded by messages which brings the cpu to the limit which triggers the watchdog at the end since all stalls.. solution. the [...]

You are correct on that, I remember iperf3 would use all available bandwidth when I used it before but perhaps the UDP option changes that behaviour. Unfortunately both JG928A and JG927A complete the 3-minute iperf3 test with the `-b 1G` option set: `iperf3 -c <switch address> --udp -b 1G -t 180 -P4`
```
- - - - - - - - - - - - - - - - - - - - - - - - -
[ ID] Interval           Transfer     Bitrate         Jitter    Lost/Total Datagrams
[SUM]  0.0-180.0 sec  1745 datagrams received out-of-order
[  5]   0.00-180.02 sec   125 MBytes  5.84 Mbits/sec  0.334 ms  3788944/3879639 (98%)  receiver
[SUM]  0.0-180.0 sec  1882 datagrams received out-of-order
[  6]   0.00-180.02 sec   125 MBytes  5.84 Mbits/sec  0.414 ms  3882860/3973651 (98%)  receiver
[SUM]  0.0-180.0 sec  1874 datagrams received out-of-order
[  9]   0.00-180.02 sec   125 MBytes  5.84 Mbits/sec  0.338 ms  3785755/3876476 (98%)  receiver
[SUM]  0.0-180.0 sec  1847 datagrams received out-of-order
[ 11]   0.00-180.02 sec   125 MBytes  5.83 Mbits/sec  0.205 ms  3885324/3975853 (98%)  receiver
[SUM]   0.00-180.02 sec   501 MBytes  23.3 Mbits/sec  0.323 ms  15342883/15705619 (98%)  receiver
```
and on the other:
```
[ ID] Interval           Transfer     Bitrate         Jitter    Lost/Total Datagrams
[SUM]  0.0-180.0 sec  1745 datagrams received out-of-order
[  5]   0.00-180.02 sec   125 MBytes  5.84 Mbits/sec  0.334 ms  3788944/3879639 (98%)  receiver
[SUM]  0.0-180.0 sec  1882 datagrams received out-of-order
[  6]   0.00-180.02 sec   125 MBytes  5.84 Mbits/sec  0.414 ms  3882860/3973651 (98%)  receiver
[SUM]  0.0-180.0 sec  1874 datagrams received out-of-order
[  9]   0.00-180.02 sec   125 MBytes  5.84 Mbits/sec  0.338 ms  3785755/3876476 (98%)  receiver
[SUM]  0.0-180.0 sec  1847 datagrams received out-of-order
[ 11]   0.00-180.02 sec   125 MBytes  5.83 Mbits/sec  0.205 ms  3885324/3975853 (98%)  receiver
[SUM]   0.00-180.02 sec   501 MBytes  23.3 Mbits/sec  0.323 ms  15342883/15705619 (98%)  receiver
```
I do see a ton of console messages with Ring contention but these don't seem to force a reboot:
```
[370556.300932] rtl838x-eth 1b00a300.ethernet eth0: Ring contention: r: 0, last a2800130, cur a2800330
[370556.392583] rtl838x-eth 1b00a300.ethernet eth0: Ring contention: r: 0, last a280046c, cur a2800250
[370556.442032] rtl838x-eth 1b00a300.ethernet eth0: Ring contention: r: 0, last a28000b8, cur a28003c4
```
Going to pull from master and try to test again later today.

plappermaul, 2025-05-24:
For the moment I'm happy that the rx patch (incl message rate limit) improved the situation. Maybe we can narrow down the switch dying after X hours.

BrainSlayer, 2025-05-24:
i made 2 patches which made it better. first the rate limit and second i increased the buffer size (see the 300->400 change) according to some old commit messages i found in the previous project this change needs a kernel config adjust.
CONFIG_ARCH_FORCE_MAX_ORDER must be changed from 11 to 13. otherwise allocations might fail. in addition i add threaded napi 
all togther this shows much less of such messages if i flood the device. it reduced from thousands to just a few

plappermaul, 2025-05-24:
Additionally there is some old 2022 stuff available at https://github.com/openwrt/openwrt/pull/11207  For now this is post 6.12 material.

BrainSlayer, 2025-05-24:
> Additionally there is some old 2022 stuff available at [#11207](https://github.com/openwrt/openwrt/pull/11207) For now this is post 6.12 material.

thx for the hint. i already ported it now and will run some tests

howels, 2025-05-25:
Noticed that although the `iperf3` tests did not cause a reboot, on my unpatched JG928A switch the SSH sessions have been locking up after every few seconds. Nothing in the logs but the symptoms are hard to ignore.

jwh, 2025-05-26:
I don't know if its just something I haven't seen before, or if its poe related (which it might be, realtek-poe is a bit sketchy sometimes) but I've started seeing this and I'm not 100% which PR it started after:

```
[18441.709764] rcu: Stack dump where RCU GP kthread last ran:
[18441.727892] Sending NMI from CPU 1 to CPUs 0:
[18441.742300] NMI backtrace for cpu 0 skipped: idling at 0x8080e994
[18415.251700] rcu: INFO: rcu_sched detected stalls on CPUs/tasks:
[18415.271350] rcu:     0-...!: (0 ticks this GP) idle=d740/0/0x0 softirq=147726/147726 fqs=0 (false positive?)
[18415.303046] rcu:     (detected by 1, t=6004 jiffies, g=230925, q=5994 ncpus=2)
[18415.326095] Sending NMI from CPU 1 to CPUs 0:
[18415.340540] NMI backtrace for cpu 0
[18415.352102] CPU: 0 PID: 0 Comm: swapper/0 Tainted: G           O       6.6.90 #0
[18415.376584] $ 0   : 00000000 00000001 00000001 809d0f00
[18415.393924] $ 4   : 00000000 81003f00 81003f00 00000000
[18415.411242] $ 8   : 00000000 0000ffff 00000000 00000000
[18415.428561] $12   : a2880000 e19227e1 6808e000 00000001
[18415.445878] $16   : fffbdf50 81003f00 00000001 81ca4000
[18415.463197] $20   : 80b10000 81003f84 81003f90 80910000
[18415.480514] $24   : d7b20000 1ff26d83
[18415.497833] $28   : 808f8000 8180de10 81004080 8060e028
[18415.515151] Hi    : 00000000
[18415.524647] Lo    : 9f52fcd3
[18415.534142] epc   : 805bda30 0x805bda30
[18415.546831] ra    : 8060e028 0x8060e028
[18415.559519] Status: 11007f03 KERNEL EXL IE
[18415.573382] Cause : 50808100 (ExcCode 00)
[18415.586647] PrId  : 00019555 (MIPS 34Kc)
[18415.599624] CPU: 0 PID: 0 Comm: swapper/0 Tainted: G           O       6.6.90 #0
[18415.624110] Stack : 00000000 00000000 81000010 801867f0 00000000 00000000 00000000 00000000
[18415.651862]         00000000 00000000 00000000 00000000 00000000 00000001 8180db10 8183b840
[18415.679592]         8180dba8 00000000 00000000 8180da60 00000038 8080e064 ffffffea 00006276
[18415.707348]         8180da24 00000276 00000000 80916bc0 8082b538 8180daf0 00000000 80910000
[18415.735103]         8180dd58 00000020 81000010 8010326c 00000018 804b946c 00000000 80ae0000
[18415.762834]         ...
```

Eventually it'll stop responding and some time later the watchdog will reset it, @howels is this what you are/were seeing?

jwh, 2025-05-26:
> Still have no operational RTL839x at hand. So I can only ask you to bisect this. If it is kernel related then there are only 20 patches from the last month. My favourite candidate would be the rx() patch. Although it does good to package flooding as you experienced.

If you want to reach out to me privately (irc, matrix), I can send a switch to you

plappermaul, 2025-07-10:
Give this a try. https://github.com/openwrt/openwrt/pull/19365

Approve it makes the world better for you.

plappermaul, 2025-08-05:
Tests where successful, This will be completely fixed after inclusion of https://github.com/openwrt/openwrt/pull/19468

tim-seoss, 2025-09-01:
These are now all backported to 24.10.  Thanks to @Djfe
Close this issue now?