openwrt/openwrt #18887: WPA3-SAE shows CCMP and not GCMP

sgobiraj, 2025-05-23:
### Describe the bug

Luci is showing that the cipher on WPA3-SAE is using CCMP. Doesn't the WPA3 spec say it should be using GCMP? There is also no option to change the mode like there is when you set it to use WPA2-PSK. When you set it to WPA2-PSK there is a drop down available to set the cipher (CCMP, GCMP, etc).

### OpenWrt version

r28597-0425664679

### OpenWrt release

24.10.1

### OpenWrt target/subtarget

mediatek/filogic

### Device

ASUS TUF-AX4200

### Image kind

Official downloaded image

### Steps to reproduce

[image]

### Diffconfig

```text

```

### Terms

- [x] I am reporting an issue for OpenWrt, not an unsupported fork.

sgobiraj, 2025-05-23:
Can also confirm that the Windows device connected shows it's using CCMP.

    Name                   : Wi-Fi
    Description            : Intel(R) Wi-Fi 6E AX210 160MHz
    GUID                   : a0668011-6e65-4a14-88f3-667a215dbd2b
    Interface type         : Primary
    State                  : connected
    Network type           : Infrastructure
    Radio type             : 802.11ax
    Authentication         : WPA3-Personal  (H2E)
    Cipher                 : CCMP
    Connection mode        : Profile
    Band                   : 5 GHz
    Channel                : 36
    Receive rate (Mbps)    : 2162
    Transmit rate (Mbps)   : 1729
    Signal                 : 86%

    Hosted network status  : Not available

brada4, 2025-05-24:
wifi spec does not not back your claim about HE / AX mode.
https://www.wi-fi.org/system/files/WPA3%20and%20Wi-Fi%20Enhanced%20Open%20Deployment%20Guide_v1.0.pdf