openwrt/openwrt #18894: ZBT-WE826: LED rt2800soc-phy0::radio does not light up on OpenWrt 24.10.1 but worked on 22.03.1

netserver, 2025-05-23:
### Describe the bug

cat /sys/class/leds/rt2800soc-phy0::radio/device/uevent
DRIVER=rt2800_wmac
OF_NAME=wmac
OF_FULLNAME=/wmac@10180000
OF_COMPATIBLE_0=ralink,rt7620-wmac
OF_COMPATIBLE_1=ralink,rt2880-wmac
OF_COMPATIBLE_N=2
MODALIAS=of:NwmacT(null)Cralink,rt7620-wmacCralink,rt2880-wmac

cat /sys/class/leds/rt2800soc-phy0::radio/trigger
timer heartbeat default-on netdev mmc0 phy0rx phy0tx phy0assoc phy0radio [phy0tpt]

dmesg | grep -i led
[    0.000000] printk: bootconsole [early0] enabled
[    0.676483] mt7620-pci: probe of 10140000.pcie failed with error -1
[    0.853538] Serial: 8250/16550 driver, 16 ports, IRQ sharing enabled
[    0.867034] printk: console [ttyS0] disabled
[    0.881599] printk: console [ttyS0] enabled
[    0.890057] printk: bootconsole [early0] disabled
[    1.122573] mtk_soc_eth 10100000.ethernet: mdio-bus disabled
[    6.178153] phy phy-usbphy.0: remote usb device wakeup disabled
[   18.508812] jitterentropy: Initialization failed with host not compliant with requirements: 9
[   20.632129] jitterentropy: Initialization failed with host not compliant with requirements: 9
[   76.805022] br-lan: port 1(eth0.1) entered disabled state
root@6800:~# dmesg | grep -i gpio
[    6.146044] gpio_button_hotplug: loading out-of-tree module taints kernel.

FIRMWARE FROM https://mirror-03.infra.openwrt.org/releases/  
openwrt-24.10.1-ramips-mt7620-zbtlink_zbt-we826-16m-squashfs-sysupgrade.bin
openwrt-24.10.0-ramips-mt7620-zbtlink_zbt-we826-16m-squashfs-sysupgrade.bin
openwrt-23.05.0-ramips-mt7620-zbtlink_zbt-we826-16m-squashfs-sysupgrade.bin
LAN  LEDS WORK OK
USB  LED WORK OK
WIFI LED DOESNT WORK 

openwrt-22.03.1-ramips-mt7620-zbtlink_zbt-we826-16m-squashfs-sysupgrade
openwrt-19.07.4-ramips-mt7620-zbt-we826-16M-squashfs-sysupgrade.bin
LAN  LED WORK OK
USB  LED WORK OK
WIFI LED WORK OK

### OpenWrt version

r28597-0425664679

### OpenWrt release

 24.10.1

### OpenWrt target/subtarget

 ramips/mt7620

### Device

Zbtlink ZBT-WE826 (16M)

### Image kind

Official downloaded image

### Actual behaviour

### Diffconfig

```text

```

### Terms

- [x] I am reporting an issue for OpenWrt, not an unsupported fork.

namiltd, 2025-05-23:
cat /sys/kernel/debug/gpio
ls /sys/class/leds

netserver, 2025-05-23:
many thanls namiltd 

**cat /sys/kernel/debug/gpio**
OpenWrt 24.10.1, r28597-0425664679
 -----------------------------------------------------
root@6800:~# cat /sys/kernel/debug/gpio
gpiochip0: GPIOs 512-535, parent: platform/10000600.gpio, 10000600.gpio:
 gpio-513 (                    |reset               ) in  hi IRQ ACTIVE LOW

gpiochip1: GPIOs 536-551, parent: platform/10000638.gpio, 10000638.gpio:
 gpio-550 (                    |green:power         ) out lo
 gpio-551 (                    |green:usb           ) out lo

gpiochip2: GPIOs 552-552, parent: platform/10000688.gpio, 10000688.gpio:
 gpio-552 (                    |green:wifi          ) out hi ACTIVE LOW

**root@6800:~# ls /sys/class/leds**
green:power              green:wifi               rt2800soc-phy0::quality
green:usb                rt2800soc-phy0::assoc    rt2800soc-phy0::radio

namiltd, 2025-05-23:
dmesg | grep rt2880_gpio

netserver, 2025-05-23:
i solved this issue with this, now wifi led work on 24.10.1   . many thanks namiltd 

uci add system led
uci set system.@led[-1].name='green:wifi'
uci set system.@led[-1].sysfs='green:wifi'
uci set system.@led[-1].trigger='phy0tpt'
uci commit system
/etc/init.d/system reload

 -----------------------------------------------------
 OpenWrt 24.10.1, r28597-0425664679
 -----------------------------------------------------
root@6800:~# dmesg | grep rt2880_gpio
root@6800:~#
root@6800:~#

namiltd, 2025-05-24:
This should be changed in the sources though.

target/linix/ramips/dts/mt7620a_zbtlink_zbt-we826.dtsi:
```
 		air {
 			label = "green:wifi";
 			gpios = <&gpio3 0 GPIO_ACTIVE_LOW>;
+			linux,default-trigger = "phy0tpt";
 		};
```
target/linix/ramips/mt7620/base-files/etc/board.d/01_leds:
```
 dlink,dwr-116-a1|\
 head-weblink,hdrm200|\
 ohyeah,oy-0001|\
 planex,mzk-ex300np|\
-zbtlink,zbt-we826-16m|\
-zbtlink,zbt-we826-32m|\
 zbtlink,zbt-wr8305rt|\
 zyxel,keenetic-lite-iii-a|\
 zyxel,keenetic-omni|\
 zyxel,keenetic-omni-ii|\
 zyxel,keenetic-viva)
 	ucidef_set_led_netdev "wifi_led" "wifi" "green:wifi" "wlan0"
 	;;
```