openwrt/openwrt #18901: GRO fraglist/SQM/Packet Steering

keyodi, 2025-05-24:
### Describe the bug

Enabling fraglist GRO results in performance degradation with SQM and Packet Steering set to Enabled.

### OpenWrt version

r28597-0425664679

### OpenWrt release

24.10.1

### OpenWrt target/subtarget

ramips/mt7621

### Device

Linksys EA6350 v4

### Image kind

Official downloaded image

### Steps to reproduce

Install SQM and configure based on your bandwidth. Queuing discipline used is fq_codel/simplest_tbf.qos.

### Actual behaviour

Internet bandwidth: 350/350
SQM config: 340/320: fq_codel/simplest_tbf.qos

Bandwidth test

**SQM:**
Setting Packet Steering to Enable w/ SQM ~220/230
Setting Packet Steering to Enable (all CPUs) w/ SQM:  ~270/280

**Packet Steering:**
Setting Packet Steering to Enable w/o SQM ~220/260
Setting Packet Steering to Enable (all CPUs) w/o SQM:  350/350

### Expected behaviour

SQM:
Setting Packet Steering to Enable w/ SQM: 320/310
Setting Packet Steering to Enable (all CPUs) w/ SQM: 320/310

Packet Steering:
Setting Packet Steering to Enable w/o SQM 350/350
Setting Packet Steering to Enable (all CPUs) w/o SQM:  350/350

### Additional info

Disabling fraglist GRO results in the expected behavior. These were my results with the applied patch:

SQM:
Setting Packet Steering to Enable w/ SQM: 250/260 --Still an issue here
Setting Packet Steering to Enable (all CPUs) w/ SQM: 320/310 --Good

Packet Steering:
Setting Packet Steering to Enable w/o SQM: 350/350 -- Good
Setting Packet Steering to Enable (all CPUs) w/o SQM:  350/350 --Good

[600-net-disable-fraglist-GRO-by-default.patch](https://github.com/user-attachments/files/20428477/600-net-disable-fraglist-GRO-by-default.patch)

### Diffconfig

```text

```

### Terms

- [x] I am reporting an issue for OpenWrt, not an unsupported fork.

brada4, 2025-05-25:
24.10-SNAPSHOT has 1 fix for gro-list - can you try that?

shdf2, 2025-05-28:
Hello, we are impacted on Filogic as well. With igmpproxy and iptv multicast streams, there's a problem with pixelization and skipping images. The problem is solved by runing **ethtool -K eth1 gro off**

brada4, 2025-05-28:
F$aglist alone can be controlled viabethtool

keyodi, 2025-06-20:
Issue will be resolved with this merge: https://github.com/openwrt/openwrt/pull/19190

closed 2025-06-20 by keyodi

keyodi, 2025-06-21:
> No it's not. Unless I'm missunderstanding the PR, it excludes ramips-mt7621 from the change. It keeps fraglist enabled for mt7621 but disables it for other plattforms like mt7628 and mt7620

Misread the logic. Reopened.

neheb, 2026-10-08:
```
Review of #18901: fraglist GRO hurts SQM/packet steering on mt7621

Scope. The original report is 24.10.1 (kernel 6.6) on an mt7621 (Linksys EA6350 v4). Throughput drops when patch 600 turns on fraglist GRO by default. Turning off only rx-gro-list, while leaving normal GRO on, brings it back to 23.05 levels.

Two of the comments describe a different bug. shdf2's BPI-R3 IPTV pixelation through igmpproxy is the ipmr multicast drop from #25713. pesa1234's MT6000 report on 6.12 is probably the same thing. They should be redirected to #25713 so the two problems stay separate.

Why 24.10 is slower

On mt7621, normal TCP GRO works well because the merged packet goes back out through hardware TSO: the Ethernet controller splits it, not the CPU. Fraglist GRO marks the merged packet differently (SKB_GSO_FRAGLIST), and mtk_eth_soc on 6.6 can't transmit that type. The kernel therefore splits every forwarded GRO packet in software and fixes up the NAT checksums one segment at a time. On an 880 MHz MIPS that is strictly worse than GRO+TSO, which fits keyodi's result.

pending-6.12 and pending-6.18 include 732-05-net-ethernet-mtk_eth_soc-add-support-for-sending-fra.patch, added by nbd in July 2025, which lets the driver transmit fraglist packets. 24.10 doesn't have it: it carries patch 600 and patch 681, but not 732-05.

What is probably still broken on 25.12 and main: SQM download

SQM shapes download traffic on an ifb device. IFB_FEATURES uses NETIF_F_GSO_SOFTWARE. Patch 681-net-remove-NETIF_F_GSO_FRAGLIST-from-NETIF_F_GSO_SOFTWARE removes fraglist from that set and adds it back only for dummy, loopback, macvlan and a few other devices, not ifb. So every fraglist GRO packet is split when it leaves the ifb queue and goes back through routing, NAT and conntrack one packet at a time. The download side loses the whole GRO benefit and still pays the GRO cost. Normal GRO packets pass through ifb whole.

Candidate fix: switch ifb to NETIF_F_GSO_SOFTWARE_ALL in patch 681. ifb doesn't modify packets; it only passes them back to the stack or to a device. So it should fall within 681's "safe to add" rule. It still needs testing.

What doesn't fit

The "Packet Steering: Enable + SQM → 220 down" result persists even with fraglist turned off. That's a separate packet-steering CPU-placement question: the default mode doesn't spread the ifb/SQM work across CPUs. It doesn't belong in this issue.

Suggested reply / actions

1. Ask keyodi to retest on 25.12, which has 732-05, with SQM off and on. I'd expect SQM off to be fixed and SQM download to still be slow.
2. If SQM download is still slow, patch ifb in 681 as above.
3. For 24.10, either backport 732-05 to pending-6.6, or document the workaround: ethtool -K <wan/lan ports> rx-gro-list off. It's probably not worth a backport if 24.10 is close to EOL.
4. Point the IPTV commenters to #25713.
```

brada4, 2026-10-08:
@neheb
 > NAT and conntrack one packet at a time

Thats fals conntrack detects invalid skb size when gro stuffs 2 packets in skb. Just search issues for raspberry and dns

khanjui, 2026-10-12:
@neheb I can confirm the fact that rx-gro-list causes lower speeds with or without sqm and with or without software flow offloading. At least on my connection which is 1 gbit pppoe over vlan on AX300T (mt7981). Latest snapshot.

@JuliusBairaktaris Seeing your https://github.com/openwrt/openwrt/pull/25780 maybe you would like to have a deeper look into this issue and others caused by rx-gro-list.

It seems like no one is going to merge this any time soon: https://github.com/openwrt/openwrt/pull/19190

JuliusBairaktaris, 2026-10-12:
I definitely wouldn't go with disabling gro out of the box.

@khanjui not to hijack this thread but can you test:   https://github.com/openwrt/openwrt/pull/25679 it needs some with mediatek