openwrt/openwrt #18915: `bss_transition` option is not supported for wireless interface

constgen, 2025-05-25:
### Describe the bug

This part of documentation https://openwrt.org/docs/guide-user/network/wifi/basic#bss_transition_management_frames_options_80211v notice that `bss_transition` replaced `ieee80211v` as an option. But on practice I see that the old `ieee80211v` still works and `bss_transition` breaks the initialization of WiFi radio

Reproduced on both: Oficcial Image and Custom Build Image (just more pacakages)

### OpenWrt version

r28597-0425664679

### OpenWrt release

24.10.1

### OpenWrt target/subtarget

ramips/mt7621

### Device

D-Link DIR-2150 R1

### Image kind

Self-built image

### Steps to reproduce

> [!NOTE]
> WiFi password is masked in the following configs and logs 

| Current working | Fails with an error |
|--------|--------|
| [image] | [image] |

### Actual behaviour

> [!NOTE]
> WiFi password is masked in the following logs 

```log
Sun May 25 14:52:42 2025 daemon.err hostapd: Line 53: unknown configuration item 'bss_transition'
Sun May 25 14:52:42 2025 daemon.err hostapd: 1 errors found in configuration file '<inline>'
Sun May 25 14:52:42 2025 daemon.err hostapd: Failed to set up interface with data: driver=nl80211 logger_syslog=127 logger_syslog_level=2 logger_stdout=127 logger_stdout_level=2 country_code=DE ieee80211d=1 ieee80211h=1 hw_mode=a beacon_int=100 stationary_ap=1 chanlist=100 tx_queue_data2_burst=2.0 #num_global_macaddr=1 #macaddr_base= ieee80211n=1 ht_coex=0 ht_capab=[HT40+][LDPC][SHORT-GI-20][SHORT-GI-40][TX-STBC][RX-STBC1] ieee80211ac=1 vht_oper_chwidth=2 vht_oper_centr_freq_seg0_idx=114 vht_capab=[RXLDPC][SHORT-GI-80][SHORT-GI-160][TX-STBC-2BY1][RX-ANTENNA-PATTERN][TX-ANTENNA-PATTERN][RX-STBC-1][VHT160-80PLUS80][MAX-A-MPDU-LEN-EXP7] channel=100  interface=phy1-ap0 bssid=78:98:e8:66:d5:44 ctrl_interface=/var/run/hostapd ap_isolate=1 bss_load_update_period=60 chan_util_avg_period=600 disassoc_low_ack=1 skip_inactivity_poll=0 preamble=1 wmm_enabled=1 ignore_broadcast_ssid=0 uapsd_advertisement_enabled=1 utf8_ssid=1 multi_ap=0 ocv=0 sae_require_mfp=1 sae_pwe=2 wpa_passphrase=******** wpa_psk_file=/var/run/hostapd-phy1-ap0.psk
Sun May 25 14:52:42 2025 daemon.notice hostapd: hostapd.add_iface failed for phy phy1 ifname=phy1-ap0
```

### Expected behaviour

`ieee80211v` should faile with an error, but `bss_transition` should work

### Additional info

This is necessaery for correct setup of Usteer, which requires declaration of `bss_transition` too in its documentation https://openwrt.org/docs/guide-user/network/wifi/usteer#configure_80211k_and_80211v_on_all_ap-nodes . But using `ieee80211v` seems to work too

[image]

### Diffconfig

```text
make: *** Нет правила для сборки цели «scripts/config/conf».  Останов.
```

### Terms

- [x] I am reporting an issue for OpenWrt, not an unsupported fork.

danpawlik, 2025-05-25:
But...there are two different parameters. Please share: 
```
apk list -I |grep wpad
```

brada4, 2025-05-26:
Install wpad-mbedtls in place of wpad-basic-mbedtls to get k+v support for steering.

brada4, 2025-05-26:
No idea, that doc is complete mess, just replace wpad and add usteer in firmware-selector, no config edit ever needed.

constgen, 2025-05-26:
Ok. This works now. Required a reboot of a router, by the way.

And now how can we contribute to this page https://openwrt.org/docs/guide-user/network/wifi/usteer to update the requied preconditions for BSS?

closed 2025-05-26 by constgen