openwrt/openwrt #18917: Website 404 error

alex4926, 2025-05-25:
### Describe the bug

When I try to download an update for my router I get an 404 error.

### OpenWrt version

r24106-10cc5fcd00

### OpenWrt release

23.05.5

### OpenWrt target/subtarget

mediatek/mt7622

### Device

Xiaomi Redmi Router AX6S

### Image kind

Official downloaded image

### Steps to reproduce

1. Open this website. - (https://openwrt.org/toh/xiaomi/ax3200)
2. Under the installation section click "Sysupgrade image" redirect link to download the image
3. It tries to open up this website (https://downloads.openwrt.org/releases/24.10.0/targets/mediatek/mt7622/openwrt-24.10.0-mediatek-mt7622-xiaomi_redmi-router-ax6s-squashfs-sysupgrade.itb) but it says 404 Not Found, nginx

[image]

### Actual behaviour

It gives me a 404 error

### Expected behaviour

It should start downloading the update image

### Additional info

This started happening when this update got released. Even Attended Sysupgrade menu on LuCI doesn't work it just refreshes and doesn't update

### Diffconfig

```text

```

### Terms

- [x] I am reporting an issue for OpenWrt, not an unsupported fork.

namiltd, 2025-05-25:
Links updated.

wozza164, 2025-06-07:
Issue still seems to exist, I have the same behaviour and cannot download the upgrade image.