openwrt/openwrt #18922: ipq40xx: AES-GCM and AES-XTS does not work with Qualcomm Crypto Engine (QCE)

LGA1150, 2025-05-26:
### Describe the bug

AES-GCM and AES-XTS does not work with Qualcomm Crypto Engine (QCE)

### OpenWrt version

r29772-1df56d02cf

### OpenWrt release

SNAPSHOT

### OpenWrt target/subtarget

ipq40xx/generic

### Device

cm520-79f

### Image kind

Self-built image

### Steps to reproduce

Build an image with self tests on:

`make kernel_menuconfig` -> `Cryptographic API` -> Turn off `Disable run-time self tests`

And try to set up a VPN connection that uses in-kernel crypto API, such as IPSec and OpenVPN DCO.

### Actual behaviour

1. AES-XTS self test fail during boot:
```
[    1.435555] alg: skcipher: xts-aes-qce setkey failed on test vector 0; expected_error=0, actual_error=-126, flags=0x1
[    1.435643] alg: self-tests for xts(aes) using xts-aes-qce failed (rc=-126)
[    1.435656] ------------[ cut here ]------------
[    1.451936] WARNING: CPU: 1 PID: 114 at crypto/testmgr.c:5936 alg_test+0x4d4/0x534
[    1.456817] alg: self-tests for xts(aes) using xts-aes-qce failed (rc=-126)
[    1.456828] Modules linked in:
[    1.471032] CPU: 1 PID: 114 Comm: cryptomgr_test Not tainted 6.6.92 #0
[    1.474159] Hardware name: Generic DT based system
[    1.480677]  unwind_backtrace from show_stack+0x10/0x14
[    1.485443]  show_stack from dump_stack_lvl+0x40/0x4c
[    1.490567]  dump_stack_lvl from __warn+0x94/0xbc
[    1.495772]  __warn from warn_slowpath_fmt+0xf8/0x15c
[    1.500460]  warn_slowpath_fmt from alg_test+0x4d4/0x534
[    1.505497]  alg_test from cryptomgr_test+0x18/0x38
[    1.510876]  cryptomgr_test from kthread+0xd8/0xf4
[    1.515478]  kthread from ret_from_fork+0x14/0x28
[    1.520338] Exception stack(0xc226ffb0 to 0xc226fff8)
[    1.525118] ffa0:                                     00000000 00000000 00000000 00000000
[    1.530158] ffc0: 00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000
[    1.538318] ffe0: 00000000 00000000 00000000 00000000 00000013 00000000
[    1.546486] ---[ end trace 0000000000000000 ]---
```

2. Try connecting the VPN, and get AES-GCM test fail:
```
[  178.248590] alg: aead: gcm_base(ctr-aes-qce,cryptd(ghash-ce-sync)) encryption test failed (wrong result) on test vector 0, cfg="in-place (two sglists)"
[  178.248709] alg: self-tests for gcm(aes) using gcm_base(ctr-aes-qce,cryptd(ghash-ce-sync)) failed (rc=-22)
[  178.248726] ------------[ cut here ]------------
[  178.270726] WARNING: CPU: 2 PID: 4584 at crypto/testmgr.c:5936 alg_test+0x4d4/0x534
[  178.275566] alg: self-tests for gcm(aes) using gcm_base(ctr-aes-qce,cryptd(ghash-ce-sync)) failed (rc=-22)
[  178.275585] Modules linked in: ksmbd pppoe ppp_async nft_fib_inet nf_flow_table_inet ath10k_pci(O) ath10k_core(O) ath(O) wireguard pppox ppp_generic nft_xfrm nft_tproxy nft_socket nft_reject_ipv6 nft_reject_ipv4 nft_reject_inet nft_reject nft_redir nft_quota nft_queue nft_numgen nft_nat nft_masq nft_log nft_limit nft_hash nft_flow_offload nft_fib_ipv6 nft_fib_ipv4 nft_fib nft_ct nft_chain_nat nf_tables nf_nat nf_flow_table nf_conntrack_netlink nf_conntrack mac80211(O) libchacha20poly1305 curve25519_neon cfg80211(O) slhc poly1305_arm nfnetlink_queue nfnetlink nf_tproxy_ipv6 nf_tproxy_ipv4 nf_socket_ipv6 nf_socket_ipv4 nf_reject_ipv6 nf_reject_ipv4 nf_log_syslog nf_defrag_ipv6 nf_defrag_ipv4 mdio_netlink(O) libcurve25519_generic libcrc32c hwmon fou6 fou crc_ccitt compat(O) chacha_neon ntfs3 sch_tbf sch_ingress sch_htb sch_hfsc em_u32 cls_u32 cls_route cls_matchall cls_fw cls_flow cls_basic act_skbedit act_mirred act_gact ledtrig_usbport nls_ucs2_utils cifs_arc4 asn1_decoder ip6_gre ip_gre gre sit ipcomp6 xfrm6_tunnel
[  178.283521]  esp6 ah6 xfrm4_tunnel ipcomp esp4 ah4 ipip ip6_tunnel oid_registry netlink_diag tunnel6 tunnel4 ip_tunnel tun ovpn(O) udp_tunnel ip6_udp_tunnel xfrm_user xfrm_ipcomp af_key xfrm_algo nls_utf8 nls_iso8859_1 nls_cp950 nls_cp936 nls_cp437 sha512_arm sha1_arm_neon sha1_arm md5 ghash_arm_ce echainiv cmac chacha20poly1305 authencesn authenc leds_gpio xhci_plat_hcd xhci_pci xhci_hcd dwc3 dwc3_qcom ramoops reed_solomon pstore gpio_button_hotplug(O) vfat fat ext4 mbcache jbd2 exfat crc32c_generic [last unloaded: ksmbd]
[  178.406745] CPU: 2 PID: 4584 Comm: cryptomgr_test Tainted: G        W  O       6.6.92 #0
[  178.428884] Hardware name: Generic DT based system
[  178.436953]  unwind_backtrace from show_stack+0x10/0x14
[  178.441556]  show_stack from dump_stack_lvl+0x40/0x4c
[  178.446675]  dump_stack_lvl from __warn+0x94/0xbc
[  178.451881]  __warn from warn_slowpath_fmt+0xf8/0x15c
[  178.456569]  warn_slowpath_fmt from alg_test+0x4d4/0x534
[  178.461608]  alg_test from cryptomgr_test+0x18/0x38
[  178.466986]  cryptomgr_test from kthread+0xd8/0xf4
[  178.471583]  kthread from ret_from_fork+0x14/0x28
[  178.476445] Exception stack(0xc3333fb0 to 0xc3333ff8)
[  178.481225] 3fa0:                                     00000000 00000000 00000000 00000000
[  178.486266] 3fc0: 00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000
[  178.494425] 3fe0: 00000000 00000000 00000000 00000000 00000013 00000000
[  178.502706] ---[ end trace 0000000000000000 ]---
```
And the VPN does not work either.

### Expected behaviour

There should not be any fail logs, and VPN should work.

### Diffconfig

```text

```

### Terms

- [x] I am reporting an issue for OpenWrt, not an unsupported fork.

LGA1150, 2025-05-26:
@cotequeiroz @thara-gopinath

cuztomas, 2025-07-04:
Not sure if we can call this a bug, QCA IPQ4019, does not have hardware support for those, only software based ones. Only the later Wi-Fi-6 and Wi-Fi-7 QCA SoC have the hardware based acceleration for that