openwrt/openwrt #18929: Openssl aes-256-gcm performance degradation

jjm2473, 2025-05-27:
### Describe the bug

Openssl aes-256-gcm performance degradation compared to OpenWRT 22.03.7

### OpenWrt version

r28597-0425664679

### OpenWrt release

24.10.1

### OpenWrt target/subtarget

x86/64

### Device

innotek GmbH VirtualBox

### Image kind

Official downloaded image

### Steps to reproduce

1. `opkg update ; opkg install openssl-util`
2. `openssl speed -elapsed -evp aes-256-gcm`
3. Check the result, compared to OpenWRT 22.03.7

### Actual behaviour

Performance degradation

### Expected behaviour

Performance at least remains consistent

### Additional info

Test on 24.10.1
[image]

Test on 22.03.7
[image]

### Diffconfig

```text

```

### Terms

- [x] I am reporting an issue for OpenWrt, not an unsupported fork.

danpawlik, 2025-05-27:
Could you make also test on openssl 3.5 which can be found in snapshot?
IMHO it is more topic for openssl mailing list than here

jjm2473, 2025-05-27:
@danpawlik 

Openssl 3.5.0 test speed is abnormal, either.

[image]

```
root@OpenWrt:~# openssl version
OpenSSL 3.5.0 8 Apr 2025 (Library: OpenSSL 3.5.0 8 Apr 2025)
root@OpenWrt:~# openssl speed -elapsed -evp aes-256-gcm
You have chosen to measure elapsed time instead of user CPU time.
Doing AES-256-GCM ops for 3s on 16 size blocks: 7328156 AES-256-GCM ops in 3.01s
Doing AES-256-GCM ops for 3s on 64 size blocks: 7185166 AES-256-GCM ops in 3.01s
Doing AES-256-GCM ops for 3s on 256 size blocks: 6349361 AES-256-GCM ops in 3.10s
Doing AES-256-GCM ops for 3s on 1024 size blocks: 3537206 AES-256-GCM ops in 3.03s
Doing AES-256-GCM ops for 3s on 8192 size blocks: 768846 AES-256-GCM ops in 3.00s
Doing AES-256-GCM ops for 3s on 16384 size blocks: 415843 AES-256-GCM ops in 3.02s
version: 3.5.0
built on: Sun May 25 18:39:13 2025 UTC
options: bn(64,64)
compiler: x86_64-openwrt-linux-musl-gcc -fPIC -pthread -m64 -Wa,--noexecstack -Wall -O3 -pipe -fno-caller-saves -fno-plt -fhonour-copts -ffunction-sections -fdata-sections -Wformat -Werror=format-security -fstack-protector -D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,relro -O3 -DPIC -fpic -pipe -fno-caller-saves -fno-plt -fhonour-copts -ffunction-sections -fdata-sections -Wformat -Werror=format-security -fstack-protector -O3 -fpic -fuse-ld=bfd -znow -zrelro -DOPENSSL_USE_NODELETE -DL_ENDIAN -DOPENSSL_PIC -DOPENSSL_BUILDING_OPENSSL -DNDEBUG -D_FORTIFY_SOURCE=1 -DPIC
CPUINFO: OPENSSL_ia32cap=0xfefa3203478bffff:0x0000000000002529:0x0000000010000400:0x0000000000000000:0x0000000000000000
The 'numbers' are in 1000s of bytes per second processed.
type             16 bytes     64 bytes    256 bytes   1024 bytes   8192 bytes  16384 bytes
AES-256-GCM      38953.65k   152774.29k   524334.33k  1195412.19k  2099462.14k  2256017.12k
root@OpenWrt:~# cat /etc/openwrt_release
DISTRIB_ID='OpenWrt'
DISTRIB_RELEASE='SNAPSHOT'
DISTRIB_REVISION='r29805-b69e6c35c5'
DISTRIB_TARGET='x86/64'
DISTRIB_ARCH='x86_64'
DISTRIB_DESCRIPTION='OpenWrt SNAPSHOT r29805-b69e6c35c5'
DISTRIB_TAINTS=''
root@OpenWrt:~# cat /tmp/sysinfo/model
innotek GmbH VirtualBox
```

Maybe it's a bug of openssl, if we can reproduce it on other OS.

@neheb Not a HW accel issue. Using the same libopenssl 3.0.16, the openssl 3.0.16 test speed is abnormal, while the openssl 3.0.15 test speed is normal (see above).

jjm2473, 2025-05-27:
It can be confirmed that the problem is related to the executable file of openssl 3.0.16+, and should not affect other programs that use libopenssl

jjm2473, 2025-05-27:
The data in https://github.com/openssl/openssl/issues/20391 shows that 3.0.8 performs better than 1.1.1q when the block size is 16 bytes. The performance difference is not much in other block size. https://github.com/openssl/openssl/issues/27513  mentioned that there is a performance degradation between 3.3.1 and 3.4.1, but I noticed that there is a performance degradation between 3.0.15 and 3.0.16, which may be because OpenSSL maintains multiple minor versions at the same time. According to the [release tags](https://github.com/openssl/openssl/tags), this problem should be introduced between Sep 3, 2024 and Feb 11, 2025

brada4, 2025-05-28:
Any indications about firmware and kernel mitigation levels and cpu id in general?

brada4, 2025-05-28:
Thus no defrct, just comparing apples to oranges?

jjm2473, 2025-09-25:
This issue appears to be a test procedure modification and is expected.

closed 2025-09-25 by jjm2473