openwrt/openwrt #18943: RPi4 incorrectly parsed text input from USB keyboard

gavdotcom, 2025-05-28:
### Describe the bug

Text written to Busybox (v1.36.1) gets all chopped up.  Via SSH it's fine. Via directly attached USB keyboard, it's nonsensical. 

For example I type:
root@openwrt:~# pwd
-ash w: not found
root@openwrt:~#
-ash: pd: not found
root@openwrt:~#

The reason why I'm on the console is that I locked myself out of the network so am fixing things up, but cannot do that due to the keyboard commands not being processed correctly.

I am onto my third RPi SBC with this issue. I had the same issue on a RPi3B and I thought the hardware was faulty as I tried several keyboards, so tossed the board. I put the SD card into a RPi3B+ and had the same issue.

I did a fresh install onto a new RPi4B with new SD card and am getting the same result.

The common thread was that all of the images I used were all 24.10.1. Either for:
Raspberry Pi 3A+/3B/3B+/CM3/Zero2/Zero2W (64bit)   or
Raspberry Pi 4B/400/CM4 (64bit)

Don't know how to progress with this. Hopefully I'm not the only one with this issue. What would be causing the software to not process CLI commands correctly? I have not seen of issue before. I have been using OpenWRT since 2007 running "White Russian" on a WRT54G but this is my first time logging a fault.

### OpenWrt version

24.10.1

### OpenWrt release

r28597-0425664679

### OpenWrt target/subtarget

bcm27xx/bcm2711

### Device

Raspberry Pi4B

### Image kind

Official downloaded image

### Steps to reproduce

Boot up
With USB keyboard and HDMI monitor attached - type in commands
Observe the commands written correctly on screen (although sometimes a bit laggy)
Hit Enter
See commands truncated / garbled when not processed correctly by the shell

### Actual behaviour

Garbled text, with certain characters held back in memory as if they were written on the next command line

### Expected behaviour

For text to not get garbled as described.

### Additional info

N/A

### Diffconfig

```text
N/A - downloaded image.
```

### Terms

- [x] I am reporting an issue for OpenWrt, not an unsupported fork.

gavdotcom, 2025-05-28:
> Invalid Release reported. r28597-0425664679
> Is this from a clean repository?

"Is this from a clean repository?" No idea what you mean its official
sources.

I can't run your bug logging commands because the command line does not
work. Read fault description.

milari, 2025-08-15:
Hi.
it affects me too, I'm using: 

VERSION_ID="24.10.2"
HOME_URL="https://openwrt.org/"
BUG_URL="https://bugs.openwrt.org/"
SUPPORT_URL="https://forum.openwrt.org/"
BUILD_ID="r28739-d9340319c6"
OPENWRT_BOARD="bcm27xx/bcm2709"
OPENWRT_ARCH="arm_cortex-a7_neon-vfpv4"
OPENWRT_TAINTS=""
OPENWRT_DEVICE_MANUFACTURER="OpenWrt"
OPENWRT_DEVICE_MANUFACTURER_URL="https://openwrt.org/"
OPENWRT_DEVICE_PRODUCT="Generic"
OPENWRT_DEVICE_REVISION="v0"
OPENWRT_RELEASE="OpenWrt 24.10.2 r28739-d9340319c6"
OPENWRT_BUILD_DATE="1750711236"

same hw, but unaffected onto 23.05 or 22.03.
Thanks.

milari, 2025-08-16:
Hi.
I've checked too.
If upgrading from a previous release to 24.10 the bug happens.
If installing from scratch, it doesn't.
I hope it can help.
Cheers...

challs, 2026-01-08:
I have run into the same problem when trying to upgrade a RPi4 to 24.10.4.

The problem occurs after only installing all previously installed packages: If I restore the configuration from the backup of the old device without installing any packages, the device is still usable. In my case, that meant that I was missing network firmware, so I had found myself locked out from the network.

I am able to use a serial console connected via GPIO to enter failsafe mode and enter commands via the serial connection.